The-OpenROAD-Project/OpenROAD-flow-scripts #1821: Error in build using docker from sources

Mudbhatkal, 2024-02-29:
### Subject

[Build] for compilation-related issues.

### Describe the bug

#8 2610.9 Consolidate compiler generated dependencies of target gpl_py
#8 2610.9 [ 95%] Building CXX object src/gpl/CMakeFiles/gpl_py.dir/CMakeFiles/gpl_py.dir/replace-pyPYTHON_wrap.cxx.o
#8 2615.3 [ 95%] Building CXX object src/gpl/CMakeFiles/gpl_py.dir/gpl_py-pythonInitVar.cc.o
#8 2615.3 [ 95%] Linking CXX static library _gpl_py.a
#8 2615.6 [ 95%] Built target gpl_py
#8 2615.6 [ 95%] Generating cts-tclInitVar.cc
#8 2615.7 [ 95%] Swig compile TritonCTS.i for tcl
#8 2615.9 Consolidate compiler generated dependencies of target cts
#8 2615.9 [ 95%] Building CXX object src/cts/src/CMakeFiles/cts.dir/CMakeFiles/cts.dir/TritonCTSTCL_wrap.cxx.o
#8 2618.3 gmake[1]: *** [CMakeFiles/Makefile2:5490: src/drt/CMakeFiles/drt.dir/all] Error 2
#8 2618.3 gmake[1]: *** Waiting for unfinished jobs....
#8 2618.3 [ 95%] Building CXX object src/cts/src/CMakeFiles/cts.dir/cts-tclInitVar.cc.o
#8 2618.4 [ 95%] Building CXX object src/cts/src/CMakeFiles/cts.dir/MakeTritoncts.cpp.o
#8 2621.2 [ 95%] Building CXX object src/cts/src/CMakeFiles/cts.dir/CtsGraphics.cpp.o
#8 2625.1 [ 95%] Building CXX object src/rmp/src/CMakeFiles/rmp.dir/rmp-tclInitVar.cc.o
#8 2625.2 [ 95%] Building CXX object src/rmp/src/CMakeFiles/rmp.dir/blifParser.cpp.o
#8 2635.2 [ 95%] Building CXX object src/rmp/src/CMakeFiles/rmp.dir/blif.cpp.o
#8 2637.6 [ 95%] Linking CXX static library gpl.a
#8 2638.0 [ 95%] Built target gpl
#8 2638.0 [ 95%] Building CXX object src/rmp/src/CMakeFiles/rmp.dir/Restructure.cpp.o
#8 2639.8 [ 95%] Linking CXX static library cts.a
#8 2640.1 [ 95%] Built target cts
#8 2640.1 [ 95%] Building CXX object src/rmp/src/CMakeFiles/rmp.dir/MakeRestructure.cpp.o
#8 2653.8 [ 95%] Linking CXX static library rmp.a
#8 2654.0 [ 95%] Built target rmp
#8 2654.0 gmake: *** [Makefile:146: all] Error 2
#8 ERROR: process "/bin/sh -c ./build_openroad.sh --no_init --local --threads ${numThreads}" did not complete successfully: exit code: 2
------
 > [4/4] RUN ./build_openroad.sh --no_init --local --threads 4:
2635.2 [ 95%] Building CXX object src/rmp/src/CMakeFiles/rmp.dir/blif.cpp.o
2637.6 [ 95%] Linking CXX static library gpl.a
2638.0 [ 95%] Built target gpl
2638.0 [ 95%] Building CXX object src/rmp/src/CMakeFiles/rmp.dir/Restructure.cpp.o
2639.8 [ 95%] Linking CXX static library cts.a
2640.1 [ 95%] Built target cts
2640.1 [ 95%] Building CXX object src/rmp/src/CMakeFiles/rmp.dir/MakeRestructure.cpp.o
2653.8 [ 95%] Linking CXX static library rmp.a
2654.0 [ 95%] Built target rmp
2654.0 gmake: *** [Makefile:146: all] Error 2
------
Dockerfile.builder:14
--------------------
  12 |     WORKDIR /OpenROAD-flow-scripts
  13 |     
  14 | >>> RUN ./build_openroad.sh --no_init --local --threads ${numThreads}
  15 |     
--------------------
ERROR: failed to solve: process "/bin/sh -c ./build_openroad.sh --no_init --local --threads ${numThreads}" did not complete successfully: exit code: 2

./build_openroad.sh command fails. Docker setup is fine . Able to run helloworld and docker desktop.

### Expected Behavior

./build_openroad.sh should run without error

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 6.5.0-21-generic
os: Ubuntu 22.04.2 LTS (Jammy Jellyfish)
./etc/Env.sh: line 57: cmake: command not found
```

### To Reproduce

git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts
cd OpenROAD-flow-scripts
./build_openroad.sh

Mudbhatkal, 2024-02-29:
[Uploading build_openroad.log…]()

maliberty, 2024-02-29:
> Uploading build_openroad.log…

That just links back to this PR

maliberty, 2024-02-29:
This isn't a good sign:
````
[WARNING] Your current OpenROAD version is outdated.
...
./etc/Env.sh: line 57: cmake: command not found
```

Mudbhatkal, 2024-03-01:
[build_openroad.log](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/14459037/build_openroad.log)

Mudbhatkal, 2024-03-01:
git pull says already upto date.

vijayank88, 2024-03-01:
@Mudbhatkal 
Can you try `./build_openroad.sh --clean --threads 1`

Mudbhatkal, 2024-03-04:
> @Mudbhatkal Can you try `./build_openroad.sh --clean --threads 1`

This worked ! Thank you

closed 2024-03-04 by Mudbhatkal