The-OpenROAD-Project/OpenROAD-flow-scripts #1824: Unable to start docker container

Mudbhatkal, 2024-03-06:
### Subject

[Build] for compilation-related issues.

### Describe the bug

~/Documents/OpenROAD-flow-scripts$ docker run -it -u $(id -u ${USER}):$(id -g ${USER}) -v $(pwd)/flow/platforms:/OpenROAD-flow-scripts/flow/platforms:ro flow-scripts:latest
Unable to find image 'flow-scripts:latest' locally
docker: Error response from daemon: pull access denied for flow-scripts, repository does not exist or may require 'docker login': denied: requested access to the resource is denied.
See 'docker run --help'.

### Expected Behavior

Container should start

### Environment

```shell
kernel: Linux 6.5.0-21-generic
os: Ubuntu 22.04.2 LTS (Jammy Jellyfish)
./etc/Env.sh: line 57: cmake: command not found
```

### To Reproduce

git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts
cd OpenROAD-flow-scripts
./build_openroad.sh --clean --threads 1
docker run --rm -it -u $(id -u ${USER}):$(id -g ${USER}) -v $(pwd)/flow:/OpenROAD-flow-scripts/flow openroad/flow-ubuntu22-builder (as per docs)
also tried 
docker run -it -u $(id -u ${USER}):$(id -g ${USER}) -v $(pwd)/flow/platforms:/OpenROAD-flow-scripts/flow/platforms:ro flow-scripts:latest
also tried docker login and signed in with username before running docker run command
also tried
docker pull openroad/flow-ubuntu22.04-builder:latest  (image name on docker desktop)

### Relevant log output

```shell
Unable to find image 'flow-scripts:latest' locally
docker: Error response from daemon: pull access denied for flow-scripts, repository does not exist or may require 'docker login': denied: requested access to the resource is denied.
See 'docker run --help'.
docker pull openroad/flow-ubuntu22.04-builder:latest  output : 
Error response from daemon: pull access denied for openroad/flow-ubuntu22.04-builder, repository does not exist or may require 'docker login': denied: requested access to the resource is denied
```

Mudbhatkal, 2024-03-06:
ugh, just figured out that docker run --rm -it -u $(id -u ${USER}):$(id -g ${USER}) -v $(pwd)/flow:/OpenROAD-flow-scripts/flow openroad/flow-ubuntu22.04-builder works.
New to Docker and docs weren't too clear.

closed 2024-03-06 by Mudbhatkal

maliberty, 2024-03-06:
https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/flow/util/docker_shell is the easiest.  Please feel free to make a doc PR to make things clearer.