The-OpenROAD-Project/OpenROAD-flow-scripts #1835: installation problem

ripudamank2, 2024-03-12:
### Subject

[Build] for compilation-related issues.

### Describe the bug

Cloning into 'yosys'...
remote: Enumerating objects: 2467, done.
remote: Counting objects: 100% (2467/2467), done.
remote: Compressing objects: 100% (2134/2134), done.
remote: Total 2467 (delta 384), reused 1405 (delta 223), pack-reused 0
Receiving objects: 100% (2467/2467), 2.89 MiB | 1.37 MiB/s, done.
Resolving deltas: 100% (384/384), done.
Note: switching to '2584903a0605cc7ffdfc1996254ccc1f548403f2'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

[  0%] Building kernel/version_2584903a0.cc
[  0%] Building kernel/driver.o
[  0%] Building techlibs/common/simlib_help.inc
[  0%] Building techlibs/common/simcells_help.inc
[  1%] Building kernel/rtlil.o
[  1%] Building kernel/log.o
[  1%] Building kernel/calc.o
In file included from kernel/driver.cc:20:
./kernel/yosys.h:42:10: fatal error: 'map' file not found
#include <map>
         ^~~~~
[  2%] Building kernel/yosys.o
In file included from kernel/rtlil.cc:20:
./kernel/yosys.h:42:10: fatal error: 'map' file not found
#include <map>
         ^~~~~
[  2%] Building kernel/binding.o
In file included from kernel/log.cc:20:
./kernel/yosys.h:42:10: fatal error: 'map' file not found
#include <map>
         ^~~~~
[  2%] Building kernel/cellaigs.o
In file included from kernel/calc.cc:24:
./kernel/yosys.h:42:10: fatal error: 'map' file not found
#include <map>
         ^~~~~
[  3%] Building kernel/celledges.o
In file included from kernel/yosys.cc:20:
./kernel/yosys.h:42:10: fatal error: 'map' file not found
#include <map>
         ^~~~~
[  3%] Building kernel/satgen.o
[  3%] Building kernel/qcsat.o
In file included from kernel/binding.cc:20:
In file included from kernel/binding.h:23:
In file included from ./kernel/rtlil.h:20:
./kernel/yosys.h:42:10: fatal error: 'map' file not found
#include <map>
         ^~~~~
In file included from kernel/cellaigs.cc:20:
In file included from ./kernel/cellaigs.h:23:
./kernel/yosys.h:42:10: fatal error: 'map' file not found
#include <map>
         ^~~~~
[  4%] Building kernel/mem.o
In file included from kernel/celledges.cc:20:
In file included from ./kernel/celledges.h:23:
./kernel/yosys.h:42:10: fatal error: 'map' file not found
#include <map>
         ^~~~~
In file included from kernel/satgen.cc:20:
In file included from ./kernel/satgen.h:23:
In file included from ./kernel/rtlil.h:20:
./kernel/yosys.h:42:10: fatal error: 'map' file not found
#include <map>
         ^~~~~
In file included from kernel/qcsat.cc:20:
In file included from ./kernel/qcsat.h:23:
In file included from ./kernel/satgen.h:23:
In file included from ./kernel/rtlil.h:20:
./kernel/yosys.h:42:10: fatal error: 'map' file not found
#include <map>
         ^~~~~
In file included from kernel/mem.cc:20:
In file included from ./kernel/mem.h:23:
./kernel/yosys.h:42:10: fatal error: 'map' file not found
#include <map>
         ^~~~~
[  4%] Building kernel/ffmerge.o
In file included from kernel/ffmerge.cc:20:
In file included from ./kernel/ffmerge.h:23:
In file included from ./kernel/ffinit.h:23:
./kernel/yosys.h:42:10: fatal error: 'map' file not found
#include <map>
         ^~~~~
1 error generated.
make: *** [Makefile:757: kernel/log.o] Error 1
make: *** Waiting for unfinished jobs....
1 error generated.
1 error generated.
1 error generated.
make: *** [Makefile:757: kernel/driver.o] Error 1
make: *** [Makefile:757: kernel/cellaigs.o] Error 1
make: *** [Makefile:757: kernel/binding.o] Error 1
1 error generated.
make: *** [Makefile:757: kernel/calc.o] Error 1
1 error generated.
make: *** [Makefile:757: kernel/celledges.o] Error 1
1 error generated.
make: *** [Makefile:757: kernel/mem.o] Error 1
1 error generated.
1 error generated.
make: *** [Makefile:757: kernel/qcsat.o] Error 1
make: *** [Makefile:757: kernel/yosys.o] Error 1
1 error generated.
make: *** [Makefile:757: kernel/ffmerge.o] Error 1
1 error generated.
make: *** [Makefile:757: kernel/satgen.o] Error 1
1 error generated.
make: *** [Makefile:757: kernel/rtlil.o] Error 1

### Expected Behavior

when i am trying to install OpenROAD i am getting above error in yoosys 

### Environment

```shell
https://openroad-flow-scripts.readthedocs.io/en/latest/user/BuildLocally.html
```

### To Reproduce

https://openroad-flow-scripts.readthedocs.io/en/latest/user/BuildLocally.html

maliberty, 2024-03-12:
What OS?  What compiler?   The missing 'map' suggests you don't have a correct compiler install.

ripudamank2, 2024-03-13:
OS : 6.5.0-25-generic #25~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC  x86_64 x86_64 x86_64 GNU/Linux.
1. How to find compiler  ?
2. /Documents/OpenROAD-flow-scripts$ sudo ./setup.sh   is giving error . Then this script or env for ORFS need to be updated

maliberty, 2024-03-13:
If you haven't run setup.sh successfully then compiling is moot.  What is the error?

ripudamank2, 2024-03-13:
i just did this 
git clone  https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts
cd OpenROAD-flow-scripts
sudo ./setup.sh

and it given me error when yosys was getting install

maliberty, 2024-03-13:
Was the error during setup.sh or build.sh?  The above errors look like they would have happened during build.sh.

ripudamank2, 2024-03-13:
I didnt run build.sh yet.   I just run setup.sh

maliberty, 2024-03-13:
I don't think setup.sh would generate the errors you show.  Please provide a complete log.

ripudamank2, 2024-03-13:
[logs.txt](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/14594222/logs.txt)
please check the log file

maliberty, 2024-03-13:
what does `g++ version` show?

ripudamank2, 2024-03-13:
g++ --version
g++ (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

maliberty, 2024-03-13:
I have nearly the same version `g++ (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0` and get no compile errors.  There is something wrong with your compiler.  I suggest you reinstall it.

ripudamank2, 2024-03-14:
i tried removing the g++ and re-installing it . 
But didnt work out

maliberty, 2024-03-14:
Can you compile just `#include <map>` ?  If not you have issues that are unrelated to OR.

maliberty, 2024-03-15:
Create foo.cc with that line as it sole content and run `g++ -c foo.cc`.

ripudamank2, 2024-03-15:
it compiled  without any error. 
when i compile while adding this #include <map> only.

ripudamank2, 2024-03-15:
i found the problem its the problem in the Makefile of yosys tool . 
Makefile:219:LD = clang++
Makefile:313:LD = clang++
It causing the problem of map file . 
Do you know how i can fix ?
if i run ./setup file it will always download yosys from git and remove my changes in the Makefile.

ripudamank2, 2024-03-15:
i tried doing that but same problem

maliberty, 2024-03-15:
Then you have a problem with your compiler install that has nothing to do with yosys

closed 2024-03-16 by maliberty

ordicker, 2024-12-31:
I got the same problem on ubuntu 22.04.
`sudo apt install g++-12`
solved it for me.