The-OpenROAD-Project/OpenROAD-flow-scripts #1839: Unexpected bug found in running a test circuit

dalyles, 2024-03-13:
### Subject

[Design] for example design issues (i.e. the design does not pass functional validation, etc.)

### Describe the bug

Hi everybody,
@nbp-lbl and I are trying to run a simple test circuit but we're coming across this error:
```
[ERROR PDN-0185] Insufficient width (2.85 um) to add straps on layer metal4 in grid "grid" with total strap width 28.48 um and offset 2.0 um.
Error: grid_strategy-M1-M4-M7.tcl, 19 PDN-0185
Command exited with non-zero status 1
```

We're not so sure how to diagnose this issue.
We are running a circuit called <code>test_module</code> in the <code>nangate45</code> platform.
Input files are provided below.

### Expected Behavior

A complete (not necessarily clean) run of ORFS for <code>test_module</code>

### Environment

```shell
.
```

### To Reproduce

To reproduce this bug,  one will need the following files

<code>test_module.v</code>
```
module test_module (
        clk,
        rst,
        in_left,
        in_top,
        in_diag,
        match,
        out_right,
        out_bottom,
        out_diag
);
        input wire clk;
        input wire rst;
        input wire in_left;
        input wire in_top;
        input wire in_diag;
        input wire match;
        output wire out_right;
        output wire out_bottom;
        output wire out_diag;
        wire or_wire;
        reg out_int;
        assign or_wire = (in_left | in_top) | in_diag;
        always @(posedge clk or posedge rst) begin
                if (rst)
                        out_int <= 1'b0;
                else
                        out_int <= or_wire;
        end
        assign out_bottom = out_int;
        assign out_right = out_int;
        assign out_diag = match & out_int;
endmodule
```

<code>config.mk</code>
```
export DESIGN_NAME = test_module
export PLATFORM    = nangate45

export VERILOG_FILES = ./designs/src/$(DESIGN_NAME)/test_module.v
export SDC_FILE      = ./designs/$(PLATFORM)/$(DESIGN_NAME)/constraint.sdc
export ABC_AREA      = 1

# Adders degrade test_module
export ADDER_MAP_FILE :=

export CORE_UTILIZATION ?= 55
export PLACE_DENSITY_LB_ADDON = 0.20
export TNS_END_PERCENT        = 100
export REMOVE_CELLS_FOR_EQY   = TAPCELL*
```

<code>constraint.sdc</code>
```
current_design test_module
  
set clk_name  core_clock
set clk_port_name clk
set clk_period 0.46
set clk_io_pct 0.2

set clk_port [get_ports $clk_port_name]

create_clock -name $clk_name -period $clk_period $clk_port

set non_clock_inputs [lsearch -inline -all -not -exact [all_inputs] $clk_port]

set_input_delay  [expr $clk_period * $clk_io_pct] -clock $clk_name $non_clock_inputs
set_output_delay [expr $clk_period * $clk_io_pct] -clock $clk_name [all_outputs]
```

### Relevant log output

```shell
==========================================================================
floorplan final report_design_area
--------------------------------------------------------------------------
Design area 10 u^2 253% utilization.
Elapsed time: 0:00.23[h:]min:sec. CPU time: user 0.20 sys 0.02 (99%). Peak memory: 94772KB.
(trap 'mv ./logs/nangate45/test_module/base/2_2_floorplan_io.tmp.log ./logs/nangate45/test_module/base/2_2_floorplan_io.log' EXIT; /usr/bin/time -f 'Elapsed time: %E[h:]min:sec. CPU time: user %U sys %S (%P). Peak memory: %MKB.' /home/dlyles/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad -exit -no_init  /home/dlyles/OpenROAD-flow-scripts/flow/scripts/io_placement_random.tcl -metrics ./logs/nangate45/test_module/base/2_2_floorplan_io.json) 2>&1 | tee ./logs/nangate45/test_module/base/2_2_floorplan_io.tmp.log
OpenROAD v2.0-12466-gcc041c5df 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
Found 0 macro blocks.
Using 2 tracks default min distance between IO pins.
[INFO PPL-0007] Random pin placement.
Elapsed time: 0:00.22[h:]min:sec. CPU time: user 0.20 sys 0.01 (99%). Peak memory: 92760KB.
(trap 'mv ./logs/nangate45/test_module/base/2_3_floorplan_tdms.tmp.log ./logs/nangate45/test_module/base/2_3_floorplan_tdms.log' EXIT; /usr/bin/time -f 'Elapsed time: %E[h:]min:sec. CPU time: user %U sys %S (%P). Peak memory: %MKB.' /home/dlyles/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad -exit -no_init  /home/dlyles/OpenROAD-flow-scripts/flow/scripts/tdms_place.tcl -metrics ./logs/nangate45/test_module/base/2_3_floorplan_tdms.json) 2>&1 | tee ./logs/nangate45/test_module/base/2_3_floorplan_tdms.tmp.log
OpenROAD v2.0-12466-gcc041c5df 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
No macros found: Skipping global_placement
Elapsed time: 0:00.22[h:]min:sec. CPU time: user 0.20 sys 0.01 (99%). Peak memory: 92408KB.
(trap 'mv ./logs/nangate45/test_module/base/2_4_floorplan_macro.tmp.log ./logs/nangate45/test_module/base/2_4_floorplan_macro.log' EXIT; /usr/bin/time -f 'Elapsed time: %E[h:]min:sec. CPU time: user %U sys %S (%P). Peak memory: %MKB.' /home/dlyles/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad -exit -no_init  /home/dlyles/OpenROAD-flow-scripts/flow/scripts/macro_place.tcl -metrics ./logs/nangate45/test_module/base/2_4_floorplan_macro.json) 2>&1 | tee ./logs/nangate45/test_module/base/2_4_floorplan_macro.tmp.log
OpenROAD v2.0-12466-gcc041c5df 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ORD-0030] Using 112 thread(s).
No macros found: Skipping macro_placement
Elapsed time: 0:00.22[h:]min:sec. CPU time: user 0.17 sys 0.05 (103%). Peak memory: 96772KB.
(trap 'mv ./logs/nangate45/test_module/base/2_5_floorplan_tapcell.tmp.log ./logs/nangate45/test_module/base/2_5_floorplan_tapcell.log' EXIT; /usr/bin/time -f 'Elapsed time: %E[h:]min:sec. CPU time: user %U sys %S (%P). Peak memory: %MKB.' /home/dlyles/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad -exit -no_init  /home/dlyles/OpenROAD-flow-scripts/flow/scripts/tapcell.tcl -metrics ./logs/nangate45/test_module/base/2_5_floorplan_tapcell.json) 2>&1 | tee ./logs/nangate45/test_module/base/2_5_floorplan_tapcell.tmp.log
OpenROAD v2.0-12466-gcc041c5df 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO TAP-0004] Inserted 2 endcaps.
[INFO TAP-0005] Inserted 0 tapcells.
Elapsed time: 0:00.23[h:]min:sec. CPU time: user 0.21 sys 0.02 (99%). Peak memory: 92840KB.
(trap 'mv ./logs/nangate45/test_module/base/2_6_floorplan_pdn.tmp.log ./logs/nangate45/test_module/base/2_6_floorplan_pdn.log' EXIT; /usr/bin/time -f 'Elapsed time: %E[h:]min:sec. CPU time: user %U sys %S (%P). Peak memory: %MKB.' /home/dlyles/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad -exit -no_init  /home/dlyles/OpenROAD-flow-scripts/flow/scripts/pdn.tcl -metrics ./logs/nangate45/test_module/base/2_6_floorplan_pdn.json) 2>&1 | tee ./logs/nangate45/test_module/base/2_6_floorplan_pdn.tmp.log
OpenROAD v2.0-12466-gcc041c5df 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[ERROR PDN-0185] Insufficient width (2.85 um) to add straps on layer metal4 in grid "grid" with total strap width 28.48 um and offset 2.0 um.
Error: grid_strategy-M1-M4-M7.tcl, 19 PDN-0185
Command exited with non-zero status 1
Elapsed time: 0:00.42[h:]min:sec. CPU time: user 0.39 sys 0.02 (98%). Peak memory: 94584KB.
make[1]: *** [Makefile:626: do-2_6_floorplan_pdn] Error 1
make: *** [Makefile:626: results/nangate45/test_module/base/2_6_floorplan_pdn.odb] Error 2
```

maliberty, 2024-03-13:
This design is so trivial that I expect the area isn't big enough to put a power grid.  You could switch to a fixed area floorplan or use a very low utilization to increase the area.

dalyles, 2024-03-13:
Hi @maliberty 

Thanks for your feedback. I tried reruning it with a core utilizaiton parameter value of 5 but it unfortunately yielded similar results:

```
[ERROR PDN-0185] Insufficient width (14.06 um) to add straps on layer metal4 in grid "grid" with total strap width 28.48 um and offset 2.0 um.
Error: grid_strategy-M1-M4-M7.tcl, 19 PDN-0185
```

maliberty, 2024-03-13:
```
#export CORE_UTILIZATION ?= 55
#export PLACE_DENSITY_LB_ADDON = 0.20
export DIE_AREA    = 0 0 80 80
export CORE_AREA   = 10 10 70 70
```
works
[image]

dalyles, 2024-03-13:
I also attempted to define the area by setting this up:  <code>export CORE_AREA = (0 0 1000 1000)</code> and removing <code>CORE_UTILIZATION</code>

```
==========================================================================
Floorplan check_setup
--------------------------------------------------------------------------
number instances in verilog is 8
Error: floorplan.tcl, 61 can't read "::env(DIE_AREA)": no such variable
Command exited with non-zero status 1
Elapsed time: 0:00.22[h:]min:sec. CPU time: user 0.19 sys 0.03 (100%). Peak memory: 93700KB.
make[1]: *** [Makefile:596: do-2_1_floorplan] Error 1
make: *** [Makefile:596: results/nangate45/test_module/base/2_1_floorplan.odb] Error 2
```

dalyles, 2024-03-13:
Hi @maliberty 
Thanks very much for providing a solution.
Was it a hard requirement to comment out the two parameters above?
From my understanding we need to initalize DIE_AREA and CORE_AREA
in lius of CORE_UTILIZATION for circuits like this.

@nbp-lbl do you have any comments on this?

Thanks!

dalyles, 2024-03-13:
This is very valuable insight. Thanks again.

maliberty, 2024-03-16:
Glad to help.  This seems resolved so I'm closing it.

closed 2024-03-16 by maliberty