The-OpenROAD-Project/OpenROAD-flow-scripts #1855: Remove inactionable `[WARNING DRT-0145] New SPACINGTABLE PARALLELRUNLENGTH overrides old SPACING rule.`

oharboe, 2024-03-19:
### Description

untar [detail-route-inactionable-warning.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/14646602/detail-route-inactionable-warning.tar.gz)

```
$ NUM_CORES=16 ./run-me-mock-array-asap7-base.sh 
OpenROAD v2.0-12608-g50753a4c5 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[WARNING STA-0450] virtual clock clock_vir can not be propagated.
[INFO ORD-0030] Using 16 thread(s).
detailed_route -output_drc ./reports/asap7/mock-array/base/5_route_drc.rpt -output_maze ./results/asap7/mock-array/base/maze.log -bottom_routing_layer M2 -top_routing_layer M7 -save_guide_updates -verbose 1 -droute_end_iter 15
[INFO DRT-0149] Reading tech and libs.
[WARNING DRT-0145] New SPACINGTABLE PARALLELRUNLENGTH overrides old SPACING rule.
[WARNING DRT-0145] New SPACINGTABLE PARALLELRUNLENGTH overrides old SPACING rule.
[WARNING DRT-0145] New SPACINGTABLE PARALLELRUNLENGTH overrides old SPACING rule.
[WARNING DRT-0145] New SPACINGTABLE PARALLELRUNLENGTH overrides old SPACING rule.
[WARNING DRT-0323] Via(s) in pin VDD of Element will be ignored
[WARNING DRT-0323] Via(s) in pin VSS of Element will be ignored

Units:                1000
```

### Suggested Solution

Fix the underlying problem or silence(make debugPrint?) inactionable warning

### Additional Context

@maliberty What about the `[WARNING DRT-0323] Via(s) in pin VDD of Element will be ignored`, that looks like a separate problem. I can file a github issue be filed. Where does this problem originate? What information is needed?

maliberty, 2024-03-19:
Fixed by https://github.com/The-OpenROAD-Project/OpenROAD/pull/4813

closed 2024-03-19 by maliberty

oharboe, 2024-03-19:
@maliberty what about [WARNING DRT-0323] Via(s) in pin VDD of Element will be ignored

maliberty, 2024-03-19:
Let's make it a separate issue.  @osamahammad21 can comment in the new issue about the impact of this warning (does it imply drc checking issues?)

oharboe, 2024-03-19:
> Let's make it a separate issue. @osamahammad21 can comment in the new issue about the impact of this warning (does it imply drc checking issues?)

Broke out issue for the second warning: https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/issues/1862