The-OpenROAD-Project/OpenROAD-flow-scripts #1862: Fix the cause of `[WARNING DRT-0323] Via(s) in pin VDD of Element will be ignored` or remove the bogus warning

oharboe, 2024-03-19:
### Subject

[Stage]: Detail Router.

### Describe the bug

untar [detail-route-inactionable-warning.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/14646602/detail-route-inactionable-warning.tar.gz)

```
$ NUM_CORES=16 ./run-me-mock-array-asap7-base.sh 
OpenROAD v2.0-12608-g50753a4c5 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[WARNING STA-0450] virtual clock clock_vir can not be propagated.
[INFO ORD-0030] Using 16 thread(s).
detailed_route -output_drc ./reports/asap7/mock-array/base/5_route_drc.rpt -output_maze ./results/asap7/mock-array/base/maze.log -bottom_routing_layer M2 -top_routing_layer M7 -save_guide_updates -verbose 1 -droute_end_iter 15
[INFO DRT-0149] Reading tech and libs.
[irrelevant warnings for this issue deleted]
[WARNING DRT-0323] Via(s) in pin VDD of Element will be ignored
[WARNING DRT-0323] Via(s) in pin VSS of Element will be ignored

Units:                1000
```

### Expected Behavior

No warning.

### Environment

```shell
OpenROAD v2.0-12608-g50753a4c5
```

### To Reproduce

See above

maliberty, 2024-03-19:
I'd like to understand the implications before removing the warning.  @osamahammad21 please comment

oharboe, 2024-03-19:
> I'd like to understand the implications before removing the warning. @osamahammad21 please comment

I imagine that the root cause could require a different test-case earlier in the flow.

The above test case is from `make DESIGN_CONFIG=designs/asap7/mock-array/config.mk detail_route_issue`.

maliberty, 2024-03-19:
Its clear that they come from the pdn of Element and are exported to the pin in write_lef_abstract.  I'm more concerned if we could miss something like a cut spacing constraint.

osamahammad21, 2024-03-24:
I found that we do skip all vias defined in the macro pin and we only consider rects on the routing layers. I have no idea why tho. Should we consider them?

maliberty, 2024-03-28:
It seems like we should consider them.

closed 2024-03-28 by tspyrou