The-OpenROAD-Project/OpenROAD-flow-scripts #1868: Make `./build_openroad.sh --local --latest` less fickle

oharboe, 2024-03-21:
### Description

Oftentimes I get an error:

```
$ ./build_openroad.sh --local --latest
[INFO FLW-0027] Saving logs to build_openroad.log
[INFO FLW-0028] ./build_openroad.sh --local --latest
[INFO FLW-0002] Updating git submodules.
Submodule path 'tools/OpenROAD': checked out '50753a4c53db3cd42009d5117fe51b73ebfe8104'
[INFO FLW-0004] Updating OpenROAD app to the HEAD  of origin/master.
Previous HEAD position was 50753a4c5 Merge pull request #4798 from eder-matheus/drt_false_warning
HEAD is now at c2b1d565b Merge pull request #4824 from The-OpenROAD-Project-staging/odb-boost-adapt
M	test/sky130hs/sky130hs.vars
From https://github.com/The-OpenROAD-Project/OpenROAD
 * branch                master     -> FETCH_HEAD
Already up to date.
[INFO FLW-0001] Using local build method. This will create binaries at 'tools/install' unless overwritten.
[INFO FLW-0017] Compiling Yosys.
make: Entering directory '/home/oyvind/OpenROAD-flow-scripts/tools/yosys'
mkdir -p /home/oyvind/OpenROAD-flow-scripts/tools/install/yosys/bin
cp yosys yosys-config yosys-abc yosys-filterlib yosys-smtbmc yosys-witness /home/oyvind/OpenROAD-flow-scripts/tools/install/yosys/bin
strip -S /home/oyvind/OpenROAD-flow-scripts/tools/install/yosys/bin/yosys
strip /home/oyvind/OpenROAD-flow-scripts/tools/install/yosys/bin/yosys-abc
strip /home/oyvind/OpenROAD-flow-scripts/tools/install/yosys/bin/yosys-filterlib
mkdir -p /home/oyvind/OpenROAD-flow-scripts/tools/install/yosys/share/yosys
cp -r share/. /home/oyvind/OpenROAD-flow-scripts/tools/install/yosys/share/yosys/.
make: Leaving directory '/home/oyvind/OpenROAD-flow-scripts/tools/yosys'
[INFO FLW-0018] Compiling OpenROAD.
-- OpenROAD version: v2.0-12680-gc2b1d565b
-- System name: Linux
-- Compiler: GNU 13.2.0
-- Build type: Debug
-- Install prefix: /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
CMake Error: Error: generator : Ninja
Does not match the generator used previously: Unix Makefiles
Either remove the CMakeCache.txt file and CMakeFiles directory or choose a different binary directory.

CMake Error at /home/oyvind/OpenROAD-flow-scripts/dependencies/share/cmake-3.24/Modules/FetchContent.cmake:1592 (message):
  CMake step for googletest failed: 1
Call Stack (most recent call first):
  /home/oyvind/OpenROAD-flow-scripts/dependencies/share/cmake-3.24/Modules/FetchContent.cmake:1744:EVAL:2 (__FetchContent_directPopulate)
  /home/oyvind/OpenROAD-flow-scripts/dependencies/share/cmake-3.24/Modules/FetchContent.cmake:1744 (cmake_language)
  /home/oyvind/OpenROAD-flow-scripts/dependencies/share/cmake-3.24/Modules/FetchContent.cmake:1958 (FetchContent_Populate)
  third-party/gtest/CMakeLists.txt:8 (FetchContent_MakeAvailable)
```

-- Configuring incomplete, errors occurred!
See also "/home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/build/CMakeFiles/CMakeOutput.log".
See also "/home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/build/CMakeFiles/CMakeError.log".
```

Fix, after which `./build_openroad.sh --local --latest` workfs fine

```
find . -regex .*CMakeCache.* -exec rm {} \;
```

### Suggested Solution

Make ./build_openroad.sh more robust.

maliberty, 2024-03-28:
I've never seen either error. 

```
CMake Error: Error: generator : Ninja
Does not match the generator used previously: Unix Makefiles
Either remove the CMakeCache.txt file and CMakeFiles directory or choose a different binary directory.
```

It sounds like a cmake restriction as it is already telling you what to do.  Are you intending to switch between make & ninja?

```
CMake Error at /home/oyvind/OpenROAD-flow-scripts/dependencies/share/cmake-3.24/Modules/FetchContent.cmake:1592 (message):
  CMake step for googletest failed: 1
```

I'm guessing it comes during a download of googletest from the web if not installed.  Most likely it is some sort of network bandwidth issue.  I'm not sure if we could setup some sort of retry.  I'm not sure how to reproduce this issue.

maliberty, 2024-05-21:
No updates, seems like a cmake issue rather than an OR one from what I can tell.

closed 2024-05-21 by maliberty

oharboe, 2024-05-22:
@maliberty We made an effort to reproduce, but were unable to. I suspect some sort of race condition.