The-OpenROAD-Project/OpenROAD-flow-scripts #1873: OpenROAD GUI not working on ubuntu 20.04

dhdhlee, 2024-03-22:
### Subject

[Stage]: Other. Please describe below.

### Describe the bug

I'm doing the same thing as before, but I'm having trouble running the GUI. I can't find a solution
The state of not moving on from the loading spef.

name@DESKTOP-J0G4UE8:~/openroad/flow$ make gui_final
ODB_FILE=./results/nangate45/gcd/base/6_final.odb /home/dhlee/openroad/tools/install/OpenROAD/bin/openroad -gui  /home/dhlee/openroad/flow/scripts/gui.tcl
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-dhlee'
This plugin does not support propagateSizeHints()
OpenROAD v2.0-12608-g50753a4c5
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
Loading spef

### Expected Behavior

GUI turns on normally

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 5.10.16.3-microsoft-standard-WSL2
os: Ubuntu 20.04.6 LTS (Focal Fossa)
cmake version 3.24.2
-- The CXX compiler identification is GNU 9.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-12608-g50753a4c5
-- System name: Linux
-- Compiler: GNU 9.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 9.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/bin/python3.8 (found version "3.8.10") found components: Interpreter
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
[... 98 lines cut ...]
-- Found OpenMP: TRUE (found version "4.5")
-- GPU is not enabled
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- GUI is enabled
-- Charts widget is enabled
-- Found Boost: /home/dhlee/openroad/dependencies/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS)
-- Found Boost: /home/dhlee/openroad/dependencies/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found suitable version "1.80.0", minimum required is "1.78")
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- Found Boost: /home/dhlee/openroad/dependencies/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization system thread
-- Found Boost: /home/dhlee/openroad/dependencies/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0")
-- Found Eigen3: /home/dhlee/openroad/dependencies/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.0")
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.uQnVFCeKjJ
```

### To Reproduce

a

maliberty, 2024-03-22:
How large is the design?  Perhaps it is still loading.  No test case is attached

dhdhlee, 2024-03-22:
> How large is the design? Perhaps it is still loading. No test case is attached

Design is gcd from the openroad default example

maliberty, 2024-03-22:
Does `openroad -gui` work?

dhdhlee, 2024-03-22:
It doesn't work
I am working with UBUNTU20.04.6 on WSL on my local PC, do I need to do anything further?

aa@aaa:~/openroad/flow$ openroad -gui
qt.qpa.xcb: could not connect to display
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "/usr/lib/qt5/plugins/platforms" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.

Signal 6 received
Stack trace:
 0# 0x0000562FA0CB8878 in openroad
 1# 0x00007FC713280090 in /lib/x86_64-linux-gnu/libc.so.6
 2# gsignal in /lib/x86_64-linux-gnu/libc.so.6
 3# abort in /lib/x86_64-linux-gnu/libc.so.6
 4# 0x00007FC713D58AAD in /lib/x86_64-linux-gnu/libQt5Core.so.5
 5# QGuiApplicationPrivate::createPlatformIntegration() in /lib/x86_64-linux-gnu/libQt5Gui.so.5
 6# QGuiApplicationPrivate::createEventDispatcher() in /lib/x86_64-linux-gnu/libQt5Gui.so.5
 7# QCoreApplicationPrivate::init() in /lib/x86_64-linux-gnu/libQt5Core.so.5
 8# QGuiApplicationPrivate::init() in /lib/x86_64-linux-gnu/libQt5Gui.so.5
 9# QApplicationPrivate::init() in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
10# gui::startGui(int&, char**, Tcl_Interp*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool) in openroad
11# ord::tclAppInit(Tcl_Interp*) in openroad
12# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
13# main in openroad
14# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
15# _start in openroad
Aborted

> Does `openroad -gui` work?

dhdhlee, 2024-03-22:
Problem solved.
I think there was a problem with the settings on x server
Thank you

maliberty, 2024-03-22:
I'm glad you figured it out

closed 2024-03-22 by maliberty

samueltensingh, 2025-01-28:
I have the same problem 

qt.qpa.xcb: could not connect to display
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "/usr/lib/qt5/plugins/platforms" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

vailable platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, wayland-egl, wayland, wayland-xcomposite-egl, wayland-xcomposite-glx, xcb.

Please can you let me know the procedure to fix this. xorg is installed at my end
I am using openroad inside the docker container

quadpixels, 2025-04-27:
I just found that if you're using WSL2 in Windows 11, you need to run the Docker image with `docker run --volume="/tmp/.X11-unix/:/tmp/.X11-unix:rw" -it XXXX` (make `/tmp/.x11-unix` available to the container)

And then run the Openroad GUI program by specifying the `DISPLAY` variable, `DISPLAY=:0.0 openroad -gui `