The-OpenROAD-Project/OpenROAD-flow-scripts #1880: Running with CORNERS="WC BC" causes ibex asap7 to error in read_library.tcl

b62833, 2024-03-23:
### Subject

[Flow] for any util, flow Makefile, or flow script issues.

### Describe the bug

Running

```
make \
  FLOW_VARIANT=ibex_for_CORNERS_bug \
  CORNERS="WC BC" \
  DESIGN_CONFIG=./designs/asap7/ibex/config.mk 
```

errors with 

Error: read_liberty.tcl, 25 can't read "::env(LIB_FILES_WC)": no such variable

Assuming the asap7 libraries are named correctly the bug is in line 13 of 

  flow/scripts/read_library.tcl

It should be

`13     set LIBKEY "[string toupper $corner]_LIB_FILES"`

### Expected Behavior

Flow to run correctly without error.

### Environment

```shell
WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 6.5.0-25-generic
os: Ubuntu 23.10 (Mantic Minotaur)
cmake version 3.24.2
-- The CXX compiler identification is GNU 13.2.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-12331-g6011bbe77
-- System name: Linux
-- Compiler: GNU 13.2.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 13.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/bin/python3.11 (found version "3.11.6") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
[... 98 lines cut ...]
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- Found Eigen3: /usr/local/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.1") 
-- GUI is enabled
-- Charts widget is not enabled
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization 
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS) 
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found suitable version "1.80.0", minimum required is "1.78")  
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization system thread 
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.KYcwgWbKri
```

### To Reproduce

make \\
  FLOW_VARIANT=ibex_for_CORNERS_bug \\
  CORNERS="WC BC" \\
  DESIGN_CONFIG=./designs/asap7/ibex/config.mk

maliberty, 2024-03-24:
This is a bug and I'll fix it shortly.

maliberty, 2024-03-25:
Fixed

closed 2024-03-25 by maliberty