The-OpenROAD-Project/OpenROAD-flow-scripts #1905: Feature request: check if numbers of warnings increasemin CI

oharboe, 2024-04-02:
> @maliberty I seem to recall you added number of warnings and errors to the `make metadata` information?
> 
> Does this mean that if there is a regression(increase) in number of warnings that this will now be caught by CI and that a choice can be made whether or not to accept the new warnings?

It means only that it is recorded currently.  We could need to set limits to check against.  It seems unrelated to this PR though

_Originally posted by @maliberty in https://github.com/The-OpenROAD-Project/OpenROAD/issues/4889#issuecomment-2033002652_

jhkim-pii, 2025-10-15:
@oharboe 
Having checkers for the warning count would be helpful. 
But the problem is that there are a lot of warnings now.

For example,
```
/workspace/review/OpenROAD-flow-scripts/flow $ rg -c WARN logs/nangate45/mempool_group/base/*.log
logs/nangate45/mempool_group/base/5_2_route.log:3
logs/nangate45/mempool_group/base/5_3_fillcell.log:2
logs/nangate45/mempool_group/base/2_2_floorplan_macro.log:1
logs/nangate45/mempool_group/base/2_3_floorplan_tapcell.log:1
logs/nangate45/mempool_group/base/2_4_floorplan_pdn.log:271
logs/nangate45/mempool_group/base/6_1_fill.log:1
logs/nangate45/mempool_group/base/3_1_place_gp_skip_io.log:1
logs/nangate45/mempool_group/base/3_2_place_iop.log:1
logs/nangate45/mempool_group/base/5_1_grt.log:2009
logs/nangate45/mempool_group/base/3_5_place_dp.log:1
logs/nangate45/mempool_group/base/6_report.log:2011
logs/nangate45/mempool_group/base/3_4_place_resized.log:80
logs/nangate45/mempool_group/base/4_1_cts.log:2
logs/nangate45/mempool_group/base/2_1_floorplan.log:1274
logs/nangate45/mempool_group/base/3_3_place_gp.log:1
```

2_4_floorplan_pdn.log
```
270 [WARNING PDN-1041] Ignoring non-block cell for grid (CORE_macro_grid_2): TBUF_X1 
271 [WARNING PDN-1041] Ignoring non-block cell for grid (CORE_macro_grid_2): TBUF_X16
272 [WARNING PDN-1041] Ignoring non-block cell for grid (CORE_macro_grid_2): TBUF_X2
273 [WARNING PDN-1041] Ignoring non-block cell for grid (CORE_macro_grid_2): TBUF_X4
274 [WARNING PDN-1041] Ignoring non-block cell for grid (CORE_macro_grid_2): TBUF_X8
275 [WARNING PDN-1041] Ignoring non-block cell for grid (CORE_macro_grid_2): TINV_X1
```

2_1_floorplan.log
```
56273 [WARNING RSZ-0075] makeBufferedNet failed for driver gen_tiles\[0\].i_tile.i_tile/gen_cores\[1\].gen_mempool_cc.riscv_core.i_snitch/_15914_/ZN
56274 [WARNING RSZ-0075] makeBufferedNet failed for driver gen_tiles\[0\].i_tile.i_tile/gen_cores\[1\].gen_mempool_cc.riscv_core.i_snitch/adder_result\\ALU_33_0_33_0_33_unused_CO_X_HAN_CARLSON/_311__1/S
56275 [WARNING RSZ-0075] makeBufferedNet failed for driver gen_tiles\[0\].i_tile.i_tile/gen_cores\[1\].gen_mempool_cc.riscv_core.i_snitch/_10117_/_563__1/ZN
56276 [WARNING RSZ-0075] makeBufferedNet failed for driver gen_tiles\[0\].i_tile.i_tile/gen_cores\[1\].gen_mempool_cc.riscv_core.i_snitch/_15958_/ZN
56277 [WARNING RSZ-0075] makeBufferedNet failed for driver gen_tiles\[0\].i_tile.i_tile/i_local_req_interco/_16533_/ZN
```

A small OpenROAD behavior change can result in more number of warnings, which makes CI fail.
I worry about frequent CI fails due to trivial warning count increases.

So.. as a first step, how about adding `*__warnings__type_count` and their checkers?
`*__warnings__type_count` is less sensitive than `*__warnings__count`.

```
WARN-001: blahblah
WARN-001: rghtihj
WARN-001: rigjarg
WARN-002: rigjrg
-> 
warnings__count = 4
warnings__type_count = 2
```

@precisionmoon FYI.

oharboe, 2025-10-15:
@jhkim-pii Good points! Perhaps this feature would do the most good as an "observe and report" feature. At least initially until we have experience.

oharboe, 2025-10-15:
rules.json could store a dictionary with warning/info id and count for more accurate and actionable reporting.

jhkim-pii, 2025-10-15:
OK. Then let me implement these things.
1. Add a new `warnings__type_count` metric and its checker
2. Add new `*__warnings__count:*` metrics and their checkers w/ proper margins (0 margin will cause too many CI failures).

oharboe, 2025-10-15:
@jhkim-pii I believe https://github.com/The-OpenROAD-Project/OpenROAD/issues/8658 is a specific example of a problematic type of warning spam w.r.t. automatic checks for changes in number of warnings...

I think initially an ability to tell if number of warnings increased and decreased and which warnings would be a good tool to educate us on any problematic warnings that have to be dealt with before automatic checks in CI for increase/decrease in warnings can be enforced...

rovinski, 2025-10-18:
> In metrics 2.1 I think you would use a modifier like floorplan__warnings__count:IFP-002 rather than a dictionary.

One concern with this approach is that message IDs change over time, and some get reused. If you want to look at a historical view, there would likely be quite a bit of noise.

maliberty, 2025-10-18:
I view it as a way to tamp down noisy warnings rather than a profile I would track over long periods.

oharboe, 2025-10-22:
Try it out and loosen/tighten as needed?

jhkim-pii, 2025-10-23:
> Try it out and loosen/tighten as needed?

Yeah. We will adjust it if any issue arises.