The-OpenROAD-Project/OpenROAD-flow-scripts #1935: Consistent treatment of flag variables

povik, 2024-04-11:
### Description

There are a number of variables to turn things on/off, as documented on the "flow variables" page: https://openroad-flow-scripts.readthedocs.io/en/latest/user/FlowVariables.html

As far as I can tell those variables don't have consistent interpretation, for some variables they must be set to 1 for the documented effect (e.g. `SKIP_CTS_REPAIR_TIMING`), for other variables the existence of the variable is all that matters, the value is ignored (e.g. `EQUIVALENCE_CHECK`).

### Suggested Solution

The obvious suggestion is to have a helper to check flags which would be used throughout the Tcl code; I have no idea if there's a fitting place to define one.

maliberty, 2024-04-11:
I agree things have gotten somewhat inconsistent.

I believe all steps start with load.tcl so that could be a place to put the utility (I see append_env_var is there which seems like a utility).  You could also create a utility.tcl and source it from load.tcl to make a cleaner separation. 

Are you interested in resolving this?

povik, 2024-04-12:
> Are you interested in resolving this?

Yes! If I get to it I will make the change.

povik, 2024-09-17:
@maliberty Not really, no

socclosure, 2024-09-22:
@maliberty @povik 
I am proposing to add a small makefile file target to print value and name of all user's configurable variables which can be used in all subsequent TCL code of any other targets. 

For reference check "print_all_env_cfg" in Makefile in the "ISSUE_1935" branch of my local git Repo at
https://github.com/socclosure/OpenROAD-flow-scripts/blob/ISSUE_1935/flow/Makefile

Let me know your feedback,

socclosure, 2024-09-27:
@maliberty @povik  Any feedback on my proposal on code implementation of this task?

maliberty, 2024-10-07:
There is a lot of progress on this recently in 
[image]

I'm not sure how printing the variables ensures consistent treatment of them.

socclosure, 2024-10-08:
As per my understanding the user wants to know the value & existence of
each configuration variable which can be set in multiple files.
So, I thought printing the name and value of each configuration variable
can provide the information that the user is looking for.

But it seems I might be missing something.
Can you please elaborate on the requirements of this issue?

-REgards
Jignesh
Can you elaborate

socclosure, 2024-10-14:
@maliberty @povik Any feedback on my question or proposal?

oharboe, 2024-10-14:
I would say this is solved, the first sweep is done.

Next up it would be valuable to have  well thought out and tested PRs to address specific inconsistencies, one per PR.

For instance, I think it would be good to have a default value instead if negative names. SKIP_... values should be negated and a default of 1 should be added to variables.yaml. This is at the nitpicking level though.

closed 2024-10-14 by maliberty

socclosure, 2024-10-14:
Thanks @maliberty and @oharboe for your feedback and closing this issue.