The-OpenROAD-Project/OpenROAD-flow-scripts #1938: Standard cell placement under power straps with same metal as standard cell pins causing obstructions for route.

nchiolino, 2024-04-15:
### Subject

[Stage]: Detail Placement.

### Describe the bug

I am working with a less mature process that has only three metal layers. The standard cell pins use both Metal-1 and Metal-2 for escapement to routing. I am using Metal-2 for the vertical power strapping to the standard cell horizontal rows. Detail placement is placing standard cells under the Metal-2 power straps which is resulting in the Metal-2 standard cell pins to short to power or ground as well as being obstructed during detailed route. 

### Expected Behavior

Shouldn't the detailed placement tool know that the standard cell pins contain Metal-2 and not to short them to the Metal-2 power and ground straps during placement? If not, how do I tell the detail placement tool to not place any standard cells under the Metal-2 power and ground straps?

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.

It is recommened to pull the latest changes.

If problem persists, file a github issue with the re-producible test case.

kernel: Linux 6.5.0-26-generic

os: Ubuntu 22.04.4 LTS (Jammy Jellyfish)

cmake version 3.24.2

-- The CXX compiler identification is GNU 11.4.0

-- Detecting CXX compiler ABI info

-- Detecting CXX compiler ABI info - done

-- Check for working CXX compiler: /usr/bin/c++ - skipped

-- Detecting CXX compile features

-- Detecting CXX compile features - done

-- OpenROAD version: v2.0-10925-gb7c9b7e1c

-- System name: Linux

-- Compiler: GNU 11.4.0

-- Build type: RELEASE

-- Install prefix: /usr/local

-- C++ Standard: 17

-- C++ Standard Required: ON

-- C++ Extensions: OFF

-- The C compiler identification is GNU 11.4.0

-- Detecting C compiler ABI info

-- Detecting C compiler ABI info - done

-- Check for working C compiler: /usr/bin/cc - skipped

-- Detecting C compile features

-- Detecting C compile features - done

-- Found Python: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter 

-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
[... 234 lines cut ...]

-- Tcl Extended disabled

-- Python3 enabled

-- Configuring done

-- Generating done

-- Build files have been written to: /tmp/tmp.CtPBzhKa85
```

### To Reproduce

I am implementing a proprietary technology so unfortunately, this is not reproduceable by the community.

### Screenshots

As seen in this image, Metal-2 vertical straps (RED) are running over Metal-2 standard cell pins (RED). This results in a pin short to either power or ground and also obstructs the detailed router and will not complete the flow. Unobstructed standard cells can be seen on the left side of the image.
[image]

### Additional Context

My plan is to leave large columns of unobstructed area for routing of M1-M3 over standard cells. Vertical straps of Metal-2 will interrupt placement cells allowing only Metal-3 to route east/west to hop over Metal-2 power/ground straps. Floorplan can be seen in the attached image.
[image]

maliberty, 2024-04-15:
In pdks with more layers you rarely see this so the placer doesn't handle it currently.  I suggest putting placement blockages under your stripes to avoid any cells going there as a workaround.

nchiolino, 2024-04-15:
Matt, thank you so very much for the super fast response.
Understood. That sounds like a great work around, how would I put placement blockage under the stripes? I apologize if that is somewhat straight forward. Do I handle that in the PDN.tcl file?

nchiolino, 2024-04-15:
Ok. So if I am understanding this correctly. This script is run every time right now for macros. So would I insert what I need into this script or create a script of my own? Does this script get called in the Makefile?

maliberty, 2024-04-15:
There is a hook in pdn for POST_PDN_TCL (https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/4ac2d011db0a3516f79ee8910eb5fb74e663ef74/flow/scripts/pdn.tcl#L7) so you could create a script and set the variable to pick it up.

nchiolino, 2024-04-15:
Excellent! I will give that a shot. 
Thank you very much for the support.

nchiolino, 2024-04-15:
Before this issue is closed, would you like for me to feedback to you if it worked? I will leave that decision up to you.

maliberty, 2024-04-15:
Its always nice to hear about success 😄

nchiolino, 2024-04-15:
The issue that I am running into right now is I dont know how to get the shape of the POWER and GROUND nets. In the macro version it sets a box after it finds the instances using 
     `foreach inst [$block getInsts] {
      if {[[$inst getMaster] getType] == "BLOCK"} {
      set box [$inst getBBox]`
But when I go to find nets using $block getNets it does not return the shape of the net. I am assuming. How do I get a box of the nets? Is there a getNetShape or something?

maliberty, 2024-04-15:
You'll need to use dbNet::getSWires and then dbSWire::getWires to get the boxes that make up the net.

nchiolino, 2024-04-17:
I apologize for bothering you again on this but I think I need some assistance in understanding what these odb objects are doing. I tried modifying the placement_blockage.tcl script to simply place a box of placement blockage of specific x and y size it seems to not be working. I ran the flow past placement and when viewing the layout in the gui I am not seeing the blockage. I have placement blockage view turned on as well. Here is the script.
```
proc block_stripes {} {
  set block [ord::get_db_block]
  set b [odb::dbBlockage_create $block 200 60 400 200]
 $b setSoft
 puts $b
}
```

maliberty, 2024-04-17:
I think you are making such a small blockage it is hard to see.  I tried `set b [odb::dbBlockage_create $block 1200 60 4000 2000]`
[image]

nchiolino, 2024-04-17:
Ah yes. I made a bad assumption of the order of magnitude of the box. I see it on my end now. Excellent! Thank you very much.

nchiolino, 2024-04-17:
Thank you for helping me with that. It looks like it is sort of working but cells are still placing over that area. Is there a way to set a hard keep out? Sorry for all the questions and thank you for the support. 
[image]

maliberty, 2024-04-18:
Blockages are hard by default

maliberty, 2024-04-18:
I can't actually see the placement blockage in your image - would you turn off the stripe and turn on the blockage

nchiolino, 2024-04-18:
Oh sorry. Is this image better? Again, it seems to be placing standard cells over the blockage still.
[image]

maliberty, 2024-04-18:
At what step in the flow are you creating the blockage?

nchiolino, 2024-04-18:
The code that is generating that blockage is being implemented in the POST_PDN.tcl file. So it is executed during the PDN step. Should the blockage exist before the PDN script? I notice that it looks for macros before this step, is that critical?

maliberty, 2024-04-18:
That should work.  Is this something where you can provide a non-proprietary test case?

nchiolino, 2024-04-18:
Yes, I did some work to repeat this "issue" with the sky130hd tech. To repeat what I have, I modified the the following:
/flow/designs/sky130hd/gcd/config.mk
```
export DESIGN_NAME = gcd
export PLATFORM    = sky130hd
export VERILOG_FILES = ./designs/src/$(DESIGN_NICKNAME)/gcd.v
export SDC_FILE      = ./designs/$(PLATFORM)/$(DESIGN_NICKNAME)/constraint.sdc
# Adders degrade GCD
export ADDER_MAP_FILE :=
export CORE_UTILIZATION = 20
export CORE_ASPECT_RATIO = 0.3
export CORE_MARGIN       = 2
export PLACE_DENSITY     = 0.65
#export TNS_END_PERCENT = 100
#export DIE_AREA    = 0.0 0.0 86.36 88.88
#export CORE_AREA   = 5.00 5.00 81.36 83.88
# Use custom power grid with core rings offset from the pads
export PDN_TCL = ./designs/$(PLATFORM)/gcd/pdn.tcl
#export EQUIVALENCE_CHECK    =   1
export REMOVE_CELLS_FOR_EQY = sky130_fd_sc_hd__tapvpwrvgnd*
```
I added the line
`export POST_PDN_TCL ?= $(PLATFORM_DIR)/post_pdn.tcl`
to /flow/platforms/sky130hd/config.mk
The custom pdn.tcl file that I added here: /flow/designs/sky130hd/gcd/pdn.tcl
```
####################################
# global connections
####################################
add_global_connection -net VDD -pin_pattern {^VDD$} -power
add_global_connection -net VDD -pin_pattern {^VDDPE$}
add_global_connection -net VDD -pin_pattern {^VDDCE$}
add_global_connection -net VDD -pin_pattern {VPWR}
add_global_connection -net VDD -pin_pattern {vdd}
add_global_connection -net VSS -pin_pattern {^VSS$} -ground
add_global_connection -net VSS -pin_pattern {^VSSE$}
add_global_connection -net VSS -pin_pattern {VGND}
####################################
# voltage domains
####################################
set_voltage_domain -power VDD -ground VSS
####################################
# standard cell grid
####################################
define_pdn_grid -name "Core"
#add_pdn_ring -grid {grid} -layers {met4 met5} -widths {3 3} -spacings {1.6 1.6} -pad_offsets {10 10} -connect_to_pads -starts_with POWER
add_pdn_ring -grid "Core" -layers {met4 met5} -widths 3.0 -spacings 1.6 -core_offsets 10
add_pdn_stripe -layer {met1} -width 0.49  -pitch 5.44  -offset 0.00 -followpins
#add_pdn_stripe -layer {met4} -width 0.96 -spacing 5.00 -pitch 56.00 -offset 2.00 -extend_to_core_ring
add_pdn_stripe -layer {met4} -width 2.96 -spacing 1.00 -pitch 56.00 -offset 2.00 -extend_to_core_ring
add_pdn_stripe -layer {met5} -width 1.60 -pitch 56.00 -offset 2.00 -extend_to_core_ring
add_pdn_connect -layers {met1 met4}
add_pdn_connect -layers {met4 met5}
pdngen
####################################
```
The next file that was added generates the blockage. It lives here:
/flow/platforms/sky130hd/post_pdn.tcl
```
proc block_stripes {} {
  set block [ord::get_db_block]
      set b [odb::dbBlockage_create $block 32000 14000 165000 40000]
      $b setSoft
}
```
The function block_stripes has to then get called in the pdn.tcl script here: 
/OpenROAD-flow-scripts/flow/scripts/pdn.tcl
```
source $::env(SCRIPTS_DIR)/load.tcl
load_design 2_5_floorplan_tapcell.odb 1_synth.sdc "Starting PDN generation"

source $::env(PDN_TCL)
pdngen

if { [info exists ::env(POST_PDN_TCL)] && [file exists $::env(POST_PDN_TCL)] } {
  source $::env(POST_PDN_TCL)
}

block_stripes

# Check all supply nets
set block [ord::get_db_block]
foreach net [$block getNets] {
    set type [$net getSigType]
    if {$type == "POWER" || $type == "GROUND"} {
# Temporarily disable due to CI issues
#        puts "Check supply: [$net getName]"
#        check_power_grid -net [$net getName]
    }
}

if {![info exists save_checkpoint] || $save_checkpoint} {
  if {[info exists ::env(GALLERY_REPORT)]  && $::env(GALLERY_REPORT) != 0} {
    write_def $::env(RESULTS_DIR)/2_floorplan.def
  }
  write_db $::env(RESULTS_DIR)/2_6_floorplan_pdn.odb
}
```
All other files associated with the sky130hd has not been modified from when I grabbed it from the repo.
Once this is all setup, I ran
`make place`
and it generated the image attached. You can see that a standard cell has been placed inside the blockage.
[image]
Let me know your thoughts. 
Thank you

nchiolino, 2024-04-23:
Sorry to request more help on this issue but as it turns out, I am terrible at TCL, definitely not my strong suit. That in combination with little to no documentation on the db variable, I think I could use a hand on this script. I have test all the sections of this script apart from the collection of shapes. The set shapset, clip result, and output work. The only portion that I cannot correctly script is the collect shapes portion (lines 12-17). 

```
proc block_stripes {} {
  set tech [ord::get_db_tech]
  set units [$tech getDbUnitsPerMicron]
  set block [ord::get_db_block]

  #
  # Collect up all the stripes as boxes
  #
  #set shapes {}
  #    lappend shapes [odb::newSetFromRect 873399 0 903399 306200]

  set shapes [dbNet::getSWires]
  foreach wires [$wires dbSWire::getWires] {
       set box [$wires getBBox]
       lappend shapes [odb::newSetFromRect [$box xMin] [$box yMin] [$box xMax] [$box yMax]]
    }
   }
  #
  # Set shapeset variable 
  #
  set shapeSet [odb::orSets $shapes]

  #
  # Clip result to the core area
  #
  set core [$block getCoreArea]
  set xl [$core xMin]
  set yl [$core yMin]
  set xh [$core xMax]
  set yh [$core yMax]
  set core_rect [odb::newSetFromRect $xl $yl $xh $yh]
  set shapeSet [odb::andSet $shapeSet $core_rect]

  #
  # Output the blockages
  #
  set rects [odb::getRectangles $shapeSet]
  foreach rect $rects {
      set b [odb::dbBlockage_create $block \
                 [$rect xMin] [$rect yMin] [$rect xMax] [$rect yMax]]
      $b setSoft
   }
}
```

maliberty, 2024-04-23:
I looked at the detailed placer and I think it is not handling such blockages (the global placer does).  A workaround would be to generate the blockage creation into a script and then run it before initialize_flooplan so that the rows are cut by these blockages.

nchiolino, 2024-05-06:
@maliberty , thank you very much for looking into that. That makes sense to have to generate the shape before it creates the site rows. I apologize for the delayed response, I have been working to get other parts of the flow operating with this technology. I am now back onto this issue and I very much need this to work in order to realize a fully routed design unfortunately. 
I started by inserting the block_stripes proc in the floorplan.tcl script below, at this point and unfortunately, it looks like its working too well. It is now cutting the horizontal power strapes and the pdn generation is attempting to connect the met1 strapes with another pair of met2 vertical wires. Any advice on what to do next? 
```
# Initialize floorplan using CORE_UTILIZATION
# ----------------------------------------------------------------------------
} elseif {[info exists ::env(CORE_UTILIZATION)] && $::env(CORE_UTILIZATION) != "" } {
  set aspect_ratio 1.0
  if {[info exists ::env(CORE_ASPECT_RATIO)] && $::env(CORE_ASPECT_RATIO) != ""} {
    set aspect_ratio $::env(CORE_ASPECT_RATIO)
  }

  set core_margin 1.0
  if {[info exists ::env(CORE_MARGIN)] && $::env(CORE_MARGIN) != ""} {
    set core_margin $::env(CORE_MARGIN)
  }

proc block_stripes {} {
  set block [ord::get_db_block]
      set b [odb::dbBlockage_create $block 838399 39100 903399 268899]
      $b setSoft
}
block_stripes

  initialize_floorplan -utilization $::env(CORE_UTILIZATION) \
                       -aspect_ratio $aspect_ratio \
                       -core_space $core_margin \
                       -site $::env(PLACE_SITE)

# Initialize floorplan using DIE_AREA/CORE_AREA
```
[image]
[image]

nchiolino, 2024-05-06:
@maliberty , I had another idea that I am trying to setup. Is the flow capable of placing a MACRO with no pins and no liberty information in the core? I think that if I can create a BLOCK of met2 obstruction or via obstruction right under the power straps, that the DRC rules will avoid placing a standard cell in that area. Is that equivalent to the blockage we have already tried? I am trying to use the macro place command but cant seem to get it to work.
```
place_macro
    -macro_name macro_name
    -location {x y}
    [-orientation orientation]
```
Any thoughts on that idea?

rovinski, 2024-05-06:
> placing a MACRO with no pins and no liberty information in the core?

Yes

There is the `cut_rows` command which breaks up rows to remove certain cell sites, and this could be run after power planning is done so that it won't impact the power grid. Unfortunately, it's only set up right now to remove cell sites underneath macros.

If you could create fake macros (they don't even need obstruction layers) over the areas you want to prevent placement, I believe you could run `cut_rows` to remove the cell sites, and then delete the macros.

I was thinking briefly about how hard it would be to simply add in a manual `cut_rows` function to remove sites in a user-specified rectangle, but I think that ultimately making the detailed placer respect placement blockages is a better solution.

maliberty, 2024-05-06:
It is somewhat equivalent as the macro will just cause the rows to be cut the same as the blockage did.  However the OBS will induce spacing requirements that I don't think you want.

maliberty, 2024-05-06:
@rovinski note that if dpl obeys the blockage the flow will have to be changed as well.  Today gpl will respect blockages around macros but rsz might put some buffers in an otherwise blocked channel between macros.  Currently dpl will legalize them there but if we obeyed the blockage they would be moved far away.  We would have to remove the macro blockages after gpl.  We would need to distinguish mpl created blockages from user ones.

rovinski, 2024-05-06:
Hmm, is there a notion of a soft blockage vs. hard blockage? Or just one kind of blockage? I didn't consider that, I was assuming that (no cell site) ≅ (placement blockage). In that case, a manual cut_row function might be useful.

rovinski, 2024-05-06:
LEF/DEF Ref seems to have soft blockages: https://coriolis.lip6.fr/doc/lefdef/lefdefref/DEFSyntax.html#Blockages

> `SOFT`
> Indicates that the initial placement should not use the area, but later phases, such as timing optimization or clock tree synthesis, can use the blockage area. This can be used to preserve certain areas (such as small channels between blocks) for buffer insertion after the initial placement.

And it seems like it's already in ODB: https://github.com/The-OpenROAD-Project/OpenROAD/blob/6fc686431fbb0dadf024120f912b28d982b899d5/src/odb/include/odb/db.h#L4162-L4165

So it seems like DPL just needs to check if the blockage is soft or not before considering it to be valid.

maliberty, 2024-05-06:
The flow is gpl, resize, dpl so dpl would need to distinguish the source of the instance when considering the blockage.  Possible but trickier.

rovinski, 2024-05-07:
I read it as:
gpl: do not place in soft or hard blockages
rsz: do not place in hard blockages
dpl: do not place in hard blockages

I think that it doesn't necessarily matter what the source of the instance is, just about gpl vs dpl.
If you want to make it more specific to consider the source, then that's an okay interpretation too. But like you said, trickier.

nchiolino, 2024-05-07:
Excellent. Thank you both very much for all the feedback. Unfortunately I am not quite sure how to proceed, am I creating a dummy macro and running the cut_rows command? I only found one example of cut_rows being used in the ihp-sg13g2 config file. I did find the mp_test1.tcl example under the mpl2 tools example. I was able to run this example then simply run cut_rows with success. So I understand the flow. However, I cannot figure out how to add an empty macro to my flow design, run cut_rows, then delete the macro later. Any more advice? Thank you very very much again.

maliberty, 2024-05-07:
I think the fastest path for now is to add the blockages.  It will be some time before dpl can be enhanced.

nchiolino, 2024-05-07:
So to be clear, I should add the blockage with a fake macro, then run cut_rows ? How do I delete the macro before place?

maliberty, 2024-05-07:
I don't think you need a fake macro is odb::dbBlockage_create is working for you.

nchiolino, 2024-05-07:
Unfortunately it is not. If I use odb::dbBlockage_create anywhere after initialize_floorplan it looks like standard cells are not avoiding the placement blockage as seen in the first image below:
[image]
I tried inserting the blockage statement in gpl and dpl scripts with the same result. 
If I use the blockage statement before initialize_floorplan, it seems to work too well as seen in the second image below:
[image]
Does that make it more clear? I can provide more information if you would like.
Thank you

maliberty, 2024-05-07:
My thought was to run through pdn generation and write out a script to generate the blockages.  Then re-run using that script at initialize_floorplan to cut the rows.

We do need to do some work on improving our blockage support but I think that will work for the short term.

nchiolino, 2024-05-08:
Understood, I will give that a try. I am not quite sure how to do that exactly. Should I:

- make floorplan (find the size and location of the blockage I need from the met2 strapes)
- insert blockage statement before initialize_floorplan
- re-run make floorplan.

Will that not produce the same result as image 2 in my last comment? 
I really do apologize for all the questions.

maliberty, 2024-05-08:
I'm not sure what you mean by "it seems to work too well".  Where you have blockages it looks to have cut the rows.

nchiolino, 2024-05-08:
Ah yes, I do apologize for being vague on this thread in my later posts, that is my fault. Here is a detailed breakdown of how I am running the flow now, and the result I am getting. I hope this helps clarify some things. 

First: I run the flow through PDN to gather vertical met2 power rail location and size for the blockage creation. I have also run the flow through placement this time to show that cells are being placed behind the power rails. So the command is: make place and the result of that is shown in figure 1.
[image]
Second: I insert the blockage code into floorplan.tcl here:
```
# Initialize floorplan using CORE_UTILIZATION
# ----------------------------------------------------------------------------
} elseif {[info exists ::env(CORE_UTILIZATION)] && $::env(CORE_UTILIZATION) != "" } {
  set aspect_ratio 1.0
  if {[info exists ::env(CORE_ASPECT_RATIO)] && $::env(CORE_ASPECT_RATIO) != ""} {
    set aspect_ratio $::env(CORE_ASPECT_RATIO)
  }
  set core_margin 1.0
  if {[info exists ::env(CORE_MARGIN)] && $::env(CORE_MARGIN) != ""} {
    set core_margin $::env(CORE_MARGIN)
  }

proc block_stripes {} {
  set block [ord::get_db_block]
      set b [odb::dbBlockage_create $block 838399 39100 903399 268899]
      $b setSoft
}
block_stripes
  initialize_floorplan -utilization $::env(CORE_UTILIZATION) \
                       -aspect_ratio $aspect_ratio \
                       -core_space $core_margin \
                       -site $::env(PLACE_SITE)
```
Next, run make nuke, to clear out all previous files. Then re-run the flow through route, so I simply run make. The result of that can be seen in figure 2 below:
[image]
At first glance it appears to have worked but upon closer inspection it suffers from two major issues. One, it is not running all met1 horizontal power straps due to cells not being placed there. The only reason there is met1 horizontal straps is becuase I am running -extend_to_core_ring. This can be seen in figure 3 below:
[image]
Problem number two is that detail route is failing to find all the pins to route to. I am assuming becuase there are cells on the each side of row blockage that it cant get to? I have pasted a couple of screen shots from the drt error below.
[image]
[image]
Please let me know if I am running this flow the way you are describing or if you have any more questions in general.
Thanks!

maliberty, 2024-05-08:
Would you show a screenshot of one of the failed instances, say output3?

nchiolino, 2024-05-08:
Actually, I think that the power and ground for those bottom cells is the issue. It does not seem to be connected to power and ground but I dont know how to fix the pdn issues.

maliberty, 2024-05-08:
The detailed router doesn't care about power connection.  I don't think the problem relates to pdn but it just an issue with accessing the cell.  You could verify that by not running pdn at all and see if it is the same.

nchiolino, 2024-05-08:
Understood. How do I run the flow without PDN? I used to be able to do this by commenting out the PDN file in the config file but I get errors when I do this now.

maliberty, 2024-05-08:
Just comment out `pdngen` in scripts/pdn.tcl

nchiolino, 2024-05-09:
Interesting. I ran the flow with no PDN or blockage comments and am still getting these DRT errors. Here are a couple of screen shots.
[image]
[image]
They seem to be for different standard cells this time but still the same result overall.

nchiolino, 2024-05-09:
I apologize, looks to be a false alarm! During all the changes I was making, I forgot that I was using a version of the LEF standard cells with too much OBS in them. I have a new version of the standard cells that work better with the DRT. It looks like its starting to look good now for routing. I will continue through with this flow and report back how far I got.
Thanks.

nchiolino, 2024-05-09:
This is so very close to being 100% working! The only remaining issue is getting that bottom row of power to fill in somehow. Is there a way to run maybe a second grid?
[image]
[image]

maliberty, 2024-05-09:
@gadfort do you have any thoughts on this bottom row?

nchiolino, 2024-05-09:
Just to provide some more clarification. Here is what I am seeing in regards to PDN with the blockage.
[image]
A majority of the designs that I have done in the past, I have used the -followpins command for the first layer standard cell row power grid. The only way the blockage is working is if I use the -extend_to_core_ring command, but it is not populating that bottom row, and now I am discovering that its also not connecting some met1 rows over the blockage.
Is there a way to re-run pdn for just the core, after place?

gadfort, 2024-05-09:
I don't see any reason why the met1 followpins would skip the bottom row, I would need a better understanding of how the floorplan was generated. If you run PDN after you have cut the rows the met1 (which follows the rows) will also get cut. (pdn does not look at standard cells for obstructions, so delaying it would also not help).

nchiolino, 2024-05-09:
@gadfort , to me it seems like -followpins is treating each "broken up" chunk of the core as separate core areas that need to be connected with met2 (vertical). So its adding those random met2 vertical straps to connect each core segment. When I do -extend_to_core_ring its treating the core ring as the vertical connections and not inserting the random met2 vertical strips. So it looks like the closest to what I need will be to use -extend_to_core_ring and just post process that last met1 row in GDS. 
I also tried playing with the other options like -number_of_straps and that seems to do nothing. Any thoughts on why that would be?

gadfort, 2024-05-09:
I think the correct order of operations for this would be:
1. init floorplan
2. generate power grid
3. add blockages
4. cut rows
5. optional remove blockages

This should allow you to use the followpins correctly since all you are attempting to do is hack around detail placements lack of support for the blockage.

nchiolino, 2024-05-09:
I cannot seem to cut the rows at any step after init floor plan unless someone can show me a detailed step by step to do that. I have honestly tried to insert the blockage at every step before place and route and it only removes the rows before init floorplan. I think I am stuck with repairing the horizontal met1 rows after it produces GDS. Not a terrible compromise given the alternative.

maliberty, 2024-05-09:
I was hoping this wouldn't be that hard to work around.  I'll see about handling blockages in legalization next week.

nchiolino, 2024-05-09:
No worries at all. I understand that I am using it in a very non-traditional way. It's honestly not in a terrible place given all those constraints. It's producing a fully routed chip which is fantastic for this technology. I really appreciate you looking into it and keep me posted if you find a solution that I can incorporate into this flow.

maliberty, 2024-05-14:
https://github.com/The-OpenROAD-Project/OpenROAD/pull/5104 will give hard blockage support in placement legalization which would make a simpler flow possible.

nchiolino, 2024-05-14:
Excellent! Thank you very much. I have two questions with this:

1. I am building the tool locally, how do you suggest I import these changes? Do I just simply download the new dpl script and replace the one I have?
2. How does the blockage work now? Do I keep the blockage in the same place as I have it now or do I need to move it into dpl or post_pdn?

Sorry again for all the questions but I cannot thank you enough for the support.

maliberty, 2024-05-14:
You can just create the blockages after pdn and then should be obeyed by later placement.

build.sh --latest should give you the more up to date OR version

nchiolino, 2024-05-14:
Very awesome. Thank you so much. I will keep you posted on if it I could get it to work on my end. 
Thank you

[40 more replies]

closed 2024-05-22 by maliberty