The-OpenROAD-Project/OpenROAD-flow-scripts #1949: VIA Routing over VIA Obstruction

nchiolino, 2024-04-19:
### Subject

[Stage]: Detail Router.

### Describe the bug

I am working with a more primitive technology and this technology does not allow any VIA's (the first via layer) to be placed inside the standard cells. To prevent this, the standard cells have a large VIA OBS rectangles inside of them. However, when implementing the spacing rule in the technology LEF file for VIA spacing within that layer, the detail router does not complete. When I remove that spacing, it completes but puts VIA's over via obstructions.

### Expected Behavior

When using a spacing requirement in the technology LEF file, I expect the routed VIA's to avoid all VIA obstructions and find alternative paths to route to the pin.

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.

It is recommened to pull the latest changes.

If problem persists, file a github issue with the re-producible test case.

kernel: Linux 6.5.0-27-generic

os: Ubuntu 22.04.4 LTS (Jammy Jellyfish)

cmake version 3.24.2

-- The CXX compiler identification is GNU 11.4.0

-- Detecting CXX compiler ABI info

-- Detecting CXX compiler ABI info - done

-- Check for working CXX compiler: /usr/bin/c++ - skipped

-- Detecting CXX compile features

-- Detecting CXX compile features - done

-- OpenROAD version: v2.0-10925-gb7c9b7e1c

-- System name: Linux

-- Compiler: GNU 11.4.0

-- Build type: RELEASE

-- Install prefix: /usr/local

-- C++ Standard: 17

-- C++ Standard Required: ON

-- C++ Extensions: OFF

-- The C compiler identification is GNU 11.4.0

-- Detecting C compiler ABI info

-- Detecting C compiler ABI info - done

-- Check for working C compiler: /usr/bin/cc - skipped

-- Detecting C compile features

-- Detecting C compile features - done

-- Found Python: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter 

-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
[... 235 lines cut ...]
-- Tcl Extended disabled

-- Python3 enabled

-- Configuring done

-- Generating done

-- Build files have been written to: /tmp/tmp.7V2WVeUNb1
```

### To Reproduce

Unfortunately, this is a proprietary technology and I cannot share any sizes or spacings. 

### Screenshots

The first image is an example of the router placing a via over via obstruction with no spacing rule enabled. The second image is the error that is produced once the via spacing rule is turned on. Specific spacing numbers and sizes have been removed for the sake of the screenshot. I have real numbers in these positions during runtime. 
[image]
[image]

### Additional Context

As a side note, there is a similar rule for via2 that the router is obeying perfectly with no issue during route. 
[image]

maliberty, 2024-04-19:
What do the pins look like on these cells? Generally we access std cell pins from above with a via so this is an unusual case.

maliberty, 2024-04-19:
Is this something where you can share a test case?

nchiolino, 2024-04-19:
Here is a screen shots of the two via obstruction layers in each cell (via1 and metal-1 pin on the left via2 and metal-2 pin on the right). This technology does not support stacked vias and will not allow via1 over contacts.
[image]
I am fully prepared to modify these standard cell obstructions to work better with the tool so to be honest I'm just curious as to why the detailed router is not choosing to use the routing lane to avoid the obstruction and approach the pin on metal-1. I tried attaching an image to help explain my question of the routing decision.
[image]

nchiolino, 2024-04-22:
I was able to work around this issue by manipulating the LEF obstructions to a less constrained area. The initial VIA obstruction that was in the LEF cell is too heavy handed for the detailed router. I shrunk the obstructions down to the minimum area while also not blocking the pins. This has resulted in a DRC clean fully routed layout.

closed 2024-04-22 by nchiolino

maliberty, 2024-04-22:
I'm glad you found a solution