The-OpenROAD-Project/OpenROAD-flow-scripts #1953: Bug in ihp-sg13g2 jpeg flow

luarss, 2024-04-20:
### Subject

[Stage]: Synthesis.

### Describe the bug

Flow terminates at yosys synthesis stage.

### Expected Behavior

Should not terminate.

### Environment

```shell
Unknown git commit, this is not a git repository.

Please make sure that you have the latest code changes and add the commit
hash in the description.

kernel: Linux 6.2.0-1014-gcp
os: Ubuntu 22.04.3 LTS (Jammy Jellyfish)
cmake version 3.24.2
CMake Warning at CMakeLists.txt:98 (message):
  OpenROAD git describe failed, using sha1 instead

-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: ee5d3738a40b10e171360db01d3a749c871f52e7
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
[... 582 lines cut ...]
21. Executing OPT pass (performing simple optimizations).
21.1. Executing OPT_EXPR pass (perform const folding).
21.2. Executing OPT_MERGE pass (detect identical cells).
21.3. Executing OPT_MUXTREE pass (detect dead branches in mux trees).
21.4. Executing OPT_REDUCE pass (consolidate $*mux and $reduce_* inputs).
21.5. Executing OPT_MERGE pass (detect identical cells).
21.6. Executing OPT_DFF pass (perform DFF optimizations).
21.7. Executing OPT_CLEAN pass (remove unused cells and wires).
21.8. Executing OPT_EXPR pass (perform const folding).
21.9. Rerunning OPT passes. (Maybe there is more to do..)
21.10. Executing OPT_MUXTREE pass (detect dead branches in mux trees).
21.11. Executing OPT_REDUCE pass (consolidate $*mux and $reduce_* inputs).
21.12. Executing OPT_MERGE pass (detect identical cells).
21.13. Executing OPT_DFF pass (perform DFF optimizations).
21.14. Executing OPT_CLEAN pass (remove unused cells and wires).
21.15. Executing OPT_EXPR pass (perform const folding).
21.16. Finished OPT passes. (There is nothing left to do.)
abc -script /workspace/flow/scripts/abc_speed.script -liberty ./objects/ihp-sg13g2/jpeg/base/lib/sg13g2_stdcell_typ_1p20V_25C.lib -constr ./objects/ihp-sg13g2/jpeg/base/abc.constr -dont_use sg13g2_antennanp -dont_use sg13g2_lgcp_1 -dont_use sg13g2_sighold -dont_use sg13g2_slgcp_1 -dont_use sg13g2_dfrbp_2 -D 8000
22. Executing ABC pass (technology mapping using ABC).
22.1. Extracting gate netlist of module `\jpeg_encoder' to `<abc-temp-dir>/input.blif'..
22.1.1. Executing ABC.
ERROR: ABC: execution of command ""/workspace/tools/install/yosys/bin/yosys-abc" -s -f /tmp/yosys-abc-oTnAlj/abc.script 2>&1" failed: return code 134.
Command exited with non-zero status 1
Elapsed time: 3:38.97[h:]min:sec. CPU time: user 217.81 sys 1.05 (99%). Peak memory: 723988KB.
```

maliberty, 2024-04-20:
It appears this design is not running in the CI.

luarss, 2024-04-21:
Should we remove the `autotuner.json` file for this design then? Since the default flow is not working.

maliberty, 2024-04-21:
Probably we should just get the base case working or remove the whole design.  Can the autotuner find a working configuration?

maliberty, 2024-04-22:
@povik this appears to be an yosys-abc crash.  Should that be reported to yosyshq or directly to abc?

povik, 2024-04-22:
Hah, just the other way I was looking at the buffering code and noting those assertions doesn't make such sense (the units there are such that 1 shouldn't be a special value), so I suspect the fix is to simply remove those.

povik, 2024-04-22:
Generally I think one has better chances with opening a Yosys issue than reporting it to the upstream abc project. Here though the issue can be pointed out easily so maybe an abc report would work too. We carry some downstream patches for abc issues and the prospect of getting those mainline are uncertain.

maliberty, 2024-04-22:
@povik if you are planning to look at it I could just assign this issue to you or I could transfer it to yosyshq.

povik, 2024-04-22:
It's not something I was planning to look at, but feel free to assign it to me, I just don't know when I will get to it. When I do, I can confirm removing the assertions does the trick (and double-check the assertions are nonsensical).

maliberty, 2024-04-22:
@QuantamHD FYI in case Alan wants to look at this directly.

QuantamHD, 2024-04-22:
Alan won't fix any bugs for you, but if you open PRs we can email him to get him to review.

mi-and-core, 2024-04-23:
I confirm the bug.
Interestingly, abc works as expected when in flow/scripts/abc_speed.script the last lines

```
buffer -c
topo
stime -c
upsize -c
dnsize -c
```

be replaced by:

```
buffer
topo
stime
upsize
dnsize
```

That could be a clue.

mi-and-core, 2024-04-23:
> Should that be reported to yosyshq or directly to abc?

From the above findings ('-c' in abc_speed.script) i guess : abc.

povik, 2025-04-14:
Abc upstream removed the assertions and the change has reached ORFS

closed 2025-04-14 by povik