The-OpenROAD-Project/OpenROAD-flow-scripts #1975: Integration of SRAM macros

mbautista-lab, 2024-04-26:
### Description

How do I integrate the SRAM macros I generated using OpenRAM to the ORFS ?  
Do you have any suggestion in which step of the flow we add this ? I did look into the macros placement but I am not sure how to do it.   or some example I can look into. I found some online but is on Openlane.

maliberty, 2024-04-26:
See https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/flow/designs/sky130hd/microwatt/config.mk where ADDITIONAL_* are set

rovinski, 2024-04-26:
To add some more context, OpenRAM should generate 4 files that you need: The verilog model (or rather, just the module interface), the digital timing model (.lib), the physical abstract (`.lef`) and the physical model (`.gds`). The `.lef` can be generated from the `.gds` if it isn't automatically generated by OpenRAM, although you may have to use a tool like [Magic](https://github.com/RTimothyEdwards/magic) to do so.

Once you have these files, you need to instantiate your RAM macro in your RTL, just like you would for any other module. This module will only have an instantiation, not a definition (in other words, do _not_ include the SRAM verilog model in your list of source files).

Next, as @maliberty mentioned above, you need to add the other model files for OpenROAD to use them. Add the .lib file to `ADDITIONAL_LIBS`, the .lef file to `ADDITIONAL_LEFS`, and the .gds to `ADDITIONAL_GDS`.

Once you have that, ORFS should automatically do the rest, including macro placement, macro power grid, etc. I think most OpenRAM macros should work out of the box with ORFS, but if the maximum metal layer for the RAM is not the usual one (usually metal4?) then you might have to alter the macro power grid script, which is located in the $(PLATFORM_DIR) directory (whichever PDK you are using).

oharboe, 2024-04-27:
Mocking SRAMs can be an alternative and give important insight into the design early on even if fakerams are not available or too difficult to implement: https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/1980

mbautista-lab, 2024-04-30:
Thank you @maliberty  
I'm looking into the example you shared, and I will try to run it. 

Thanks, @rovinski, for the additional information. I'm working on it and will try to incorporate my generated SRAM into my design.

I will keep you posted then,

oharboe, 2024-05-01:
@mbautista-lab There is now an example of how to mock sram in OpenROAD-flow-scripts. This can be easier than fakesram to set up initially and also covers SRAM types that don't exist in fakesram: 

Here is the readme:

https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/flow/designs/asap7/riscv32i-mock-sram/README.md

Pull request was just merged has some more information even if the history is a bit messier than the final result, it is interesting and there are some screenshots: https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/1980

mbautista-lab, 2024-05-07:
Thank you, @oharboe. This is helpful. I will check it out.

mbautista-lab, 2024-05-09:
I have follow-up questions. I encountered this error below. I have the generated SRAM from OpenRAM, and I have added the .gds, .lef, and .lib  in my config,mk. There is a compatibility issue with the units. I tried changing the unit in my .lef file, but I don't know if it's the right thing to do.  Also, When I change the unit to 1000 in my .lef file. The flow runs and is finished, but  I do not see the Sram macro in my layout, its missing. 

 

Running floorplan.tcl, stage 2_1_floorplan
[INFO ODB-0227] LEF file: /home/mgbautista/OpenROAD-flow-scripts/flow/platforms/sky130hd/lef/sky130_fd_sc_hd.tlef, created 13 layers, 25 vias
[INFO ODB-0227] LEF file: /home/mgbautista/OpenROAD-flow-scripts/flow/platforms/sky130hd/lef/sky130_fd_sc_hd_merged.lef, created 441 library cells
[WARNING ODB-0205] The LEF UNITS DATABASE MICRON convert factor (2000) is greater than the database units per micron (1000) of the current technology.
[INFO ODB-0227] LEF file: ./designs/sky130hd/noc_buffer_in/lef/sky130_sram_2kbyte_1rw1r_32x512_8.lef, created 1 library cells
[ERROR ODB-0292] LEF data from ./designs/sky130hd/noc_buffer_in/lef/sky130_sram_2kbyte_1rw1r_32x512_8.lef is discarded due to errors
Error: floorplan.tcl, 3 ODB-0292
Command exited with non-zero status 1
Elapsed time: 0:00.35[h:]min:sec. CPU time: user 0.32 sys 0.02 (99%). Peak memory: 102468KB.
make[1]: *** [Makefile:615: do-2_1_floorplan] Error 1
make: *** [Makefile:614: results/sky130hd/noc_buffer_in/base/2_1_floorplan.odb] Error 2

maliberty, 2024-05-09:
When you generate the ram I hope you have an option to set the dbu/micro to the correct value.  Simply changing it in the LEF will result in it being the wrong size. 

Do you have an instance of your RAM in the netlist?

mbautista-lab, 2024-05-09:
Yes I did,   this is how instantiate the ram, 

`ifdef SYNTH

 sky130_sram_2kbyte_1rw1r_32x512_8 sky_sram_macro(
    // Port 0: RW
    .clk0   (wr_clk]),
    .csb0   (wr_en),   //- Chip select
    .web0   (wr_en),   //- Write enable
    .wmask0 (4{wr_en}),
    .addr0  (addra),
    .din0   (din),
    .dout0  (),
    // Port 1: R
    .clk1   (rd_clk),
    .csb1   (rd_en),   //- Chip select
    .addr1  (addrb),
    .dout1  (dout)
);

`else

	reg [WRITE_DATA_WIDTH - 1:0] mem [0:FIFO_WRITE_DEPTH - 1];
	always @(posedge wr_clk)
		if (wr_en)
			mem[addra] <= din;
	always @(posedge rd_clk)
		if (rst)
			dout <= 'h0;
		else if (rd_en)
		dout <= mem[addrb];

`endif

maliberty, 2024-05-09:
Do you see that instance in the .v coming out of yosys?  Do you see it in the final .def from ORFS?

mbautista-lab, 2024-05-09:
I can see the xpm_fifo_async, which the top module that contains the SRAM,  but I cant see the SRAM in the .v and .def file.

maliberty, 2024-05-09:
If it isn't in the yosys output then there is nothing OR can do.  Did you define SYNTH when you ran yosys?

mbautista-lab, 2024-05-09:
I have updated a new .lef with the correct unit, and yes, it's the definition I'm missing.
 Now I can see the macro in my layout. many thanks,

maliberty, 2024-05-09:
np

closed 2024-05-09 by maliberty

TheAnimatrix, 2025-04-28:
I get no access point errors for my openram generated macro on din0[0] for 1r1w sram, i made sure to change lef.py to have 1000 micron units in OpenRAM and im using tech="sky130". 

Do you have any idea what i'm doing wrong ?