The-OpenROAD-Project/OpenROAD-flow-scripts #1992: Error in building locally in WSL docker

onomaruko, 2024-05-07:
### Subject

[Build] for compilation-related issues.

### Describe the bug

I tried to build locally after following step
1. https://openroad-flow-scripts.readthedocs.io/en/latest/user/BuildWithWSL.html#
2.   https://openroad-flow-scripts.readthedocs.io/en/latest/user/BuildLocally.html
  Then build has not succeeded. 

### Expected Behavior

I would like to build in my Widows PC. 
Windows 11 Pro 16.0 GB Intel(R) Core(TM) i7-10750H CPU @ 2.60GHz   2.59 GHz

### Environment

```shell
Mybe no module yet.
```

### To Reproduce

[build_openroad.log](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/15228645/build_openroad.log)
[messages.txt](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/15228721/messages.txt)

### Relevant log output

```shell
Let me attach log file and error message in gpl above.
seems some layout information is missing in my build environment.
```

### Additional Context

Thank you for your supports.

Nobuto Ono in Jedat.

rovinski, 2024-05-07:
> c++: fatal error: Killed signal terminated program cc1plus

You are probably running out of memory from building too many files at the same time. Try
```
./build_openroad.sh --local --threads 4
```
To run only 4 jobs at a time. Your CPU has 12 hardware threads so it's likely trying to build 12 at a time right now.

onomaruko, 2024-05-08:
Austin san,

 

Thank you for the quick follow.

Yes, I will try again.

 

Regards,

N.Ono

onomaruko, 2024-05-08:
Hi Austin san,

 

Thank you I succeeded.

 

Regards,

N.Ono

 

 

 

------------------------------------------------------------------------------

Jedat Inc.

Ono, Nobuto

Senior Manager, Business Development

E-mail ***@***.***

TEL: +81-3-6262-8401

http://www.jedat.co.jp

HSB Teppozu, 1-1-12, Minato, Chuo-ku, Tokyo 104-0043 Japan

--------------------------------------------------------------------------------

closed 2024-05-08 by maliberty