The-OpenROAD-Project/OpenROAD-flow-scripts #1996: make file error

mbautista-lab, 2024-05-08:
### Subject

[Stage]: Synthesis.

### Describe the bug

I did a git pull origin master to get the latest update, and  I get this error when I run make.  How do I fix this? Thanks 

 [FLOW] Extracting clock period from SDC file: ./results/sky130hd/noc_decoder/base/clock_period.txt
[FLOW] Setting clock period to 4
ERROR: Command syntax error: Unknown option or option in arguments.
> synth -top noc_decoder -run :fine -flatten -extra-map /home/mgbautista/OpenROAD-flow-scripts/flow/platforms/common/lcu_kogge_stone.v
>                                            ^
Command exited with non-zero status 1
Elapsed time: 0:00.16[h:]min:sec. CPU time: user 0.15 sys 0.01 (98%). Peak memory: 41884KB.
make[1]: *** [Makefile:499: do-yosys] Error 1
make: *** [Makefile:502: results/sky130hd/noc_decoder/base/1_1_yosys.v] Error 2

### Expected Behavior

.

### Environment

```shell
.
```

### To Reproduce

.

maliberty, 2024-05-08:
The error is coming from yosys.  Have you updated the submodule and run build.sh?  What version of yosys are you seeing when you run?

mbautista-lab, 2024-05-08:
yes, I just ran ./build_openroad.sh --local 
My Yosys version is Yosys 0.13+15

I ran again and still getting the same error

vijayank88, 2024-05-08:
> yes, I just ran ./build_openroad.sh --local My Yosys version is Yosys 0.13+15
> 
> I ran again and still getting the same error

@mbautista-lab 
you can try below commands
```
git checkout master
git pull
git clean -xdf ./tools
sudo ./setup.sh
./build_openroad.sh --local
```

mbautista-lab, 2024-05-08:
Thanks that works

closed 2024-05-08 by maliberty