The-OpenROAD-Project/OpenROAD-flow-scripts #1998: Error in ” make DESIGN_CONFIG=./designs/asap7/aes/config.mk”　cts.tcl

onomaruko, 2024-05-09:
### Subject

[Flow] for any util, flow Makefile, or flow script issues.

### Describe the bug

 run make DESIGN_CONFIG=./designs/asap7/aes/config.mk then stop as 
[INFO RSZ-0045] Inserted 1240 buffers, 1 to split loads.
[INFO RSZ-0041] Resized 272 instances.
[INFO RSZ-0043] Swapped pins on 1151 instances.
[INFO RSZ-0049] Cloned 225 instances.
[WARNING RSZ-0062] Unable to repair all setup violations.
[INFO RSZ-0033] No hold violations found.
Error: cts.tcl, 104 couldn't execute "eqy": no such file or directory
Command exited with non-zero status 1
Elapsed time: 1:38.02[h:]min:sec. CPU time: user 97.80 sys 0.13 (99%). Peak memory: 342840KB.
make[1]: *** [Makefile:732: do-4_1_cts] Error 1
make: *** [Makefile:731: results/asap7/aes/base/4_1_cts.odb] Error 2

### Expected Behavior

[4_1_cts.log](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/15259890/4_1_cts.log)

### Environment

```shell
I run ./etc/Env.sh
then
------------------------------------------------------
 ./etc/Env.sh
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 5.15.146.1-microsoft-standard-WSL2
os: Ubuntu 22.04.4 LTS (Jammy Jellyfish)
./etc/Env.sh: line 57: cmake: command not found
----------------------------------------------------
But I just download a few days ago...
```

### To Reproduce

In WSL and locally build.
Then
make DESIGN_CONFIG=./designs/asap7/aes/config.mk

[build_openroad.log](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/15260063/build_openroad.log)

vijayank88, 2024-05-09:
@onomaruko 
have you ran `sudo ./setup.sh` before build commands?
Maybe you can skip equivalence check by commenting line: https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/3d347b425f4f8ecc1e514e2b6504f96592fabd53/flow/designs/asap7/aes/config.mk#L16

onomaruko, 2024-05-10:
Thank you for the quick follow.

Yes, I used ./setup.sh in build command.

But I understood to forget to run

source env.sh !

before make

 

I succeeded to run ./designs/asap7/aes/config.mk in my 16GB Windows PC.

 

Regards,

N.Ono

closed 2024-05-10 by maliberty