The-OpenROAD-Project/OpenROAD-flow-scripts #2011: consult

chros098, 2024-05-16:
### Subject

[Flow] for any util, flow Makefile, or flow script issues.

### Describe the bug

In the configuration of the platform, only one synthesis buff cell and one CTS buff cell are written. May I ask if I write all the buff cells, will the tool add them based on actual needs, such as the first or last one. In addition, will the spacing between cells be strictly maintained according to the spacing specified in the song value set at CELL-PAD-IN_SITES-DETAIL-PLACEMENT? Thank you.

### Expected Behavior

In the configuration of the platform, only one synthesis buff cell and one CTS buff cell are written. May I ask if I write all the buff cells, will the tool add them based on actual needs, such as the first or last one. In addition, will the spacing between cells be strictly maintained according to the spacing specified in the song value set at CELL-PAD-IN_SITES-DETAIL-PLACEMENT? Thank you.

### Environment

```shell
In the configuration of the platform, only one synthesis buff cell and one CTS buff cell are written. May I ask if I write all the buff cells, will the tool add them based on actual needs, such as the first or last one. In addition, will the spacing between cells be strictly maintained according to the spacing specified in the song value set at CELL-PAD-IN_SITES-DETAIL-PLACEMENT? Thank you.
```

### To Reproduce

In the configuration of the platform, only one synthesis buff cell and one CTS buff cell are written. May I ask if I write all the buff cells, will the tool add them based on actual needs, such as the first or last one. In addition, will the spacing between cells be strictly maintained according to the spacing specified in the song value set at CELL-PAD-IN_SITES-DETAIL-PLACEMENT? Thank you.

maliberty, 2024-05-21:
What is "the platform" you are referring to - a public PDK or some private setup?

In general we will use the available cells and not just one size.  Padding should be obeyed.

maliberty, 2024-05-21:
This seems more like a discussion than an issue as no particular problem needs resolving.

maliberty, 2024-05-22:
There is no config CTS-BUF-CELL - where do you see this in ORFS?

chros098, 2024-05-27:
[image]
Here, flow/platforms/nangate45/config.mk, line 74

chros098, 2024-05-27:
[image]

maliberty, 2024-05-27:
That's just a leftover and isn't connected to anything in scripts/.  I'll remove it.

maliberty, 2024-05-27:
removed in https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/2029

closed 2026-04-16 by github-actions[bot]