The-OpenROAD-Project/OpenROAD-flow-scripts #2027: OOM in Global placement asap7/aes-block

luarss, 2024-05-25:
### Subject

[Stage]: Global Placement.

### Describe the bug

I was running Autotuner with the following parameters for design `aes-block` and it ran out of memory. 

{
  "CELL_PAD_IN_SITES_DETAIL_PLACEMENT": 3,
  "CELL_PAD_IN_SITES_GLOBAL_PLACEMENT": 2,
  "CORE_ASPECT_RATIO": 0.8471446033249062,
  "CORE_MARGIN": 2,
  "CORE_UTILIZATION": 1,
  "CTS_CLUSTER_DIAMETER": 160,
  "CTS_CLUSTER_SIZE": 183,
  "PLACE_DENSITY_LB_ADDON": 0.0800363890832451,
  "_FR_FILE_PATH": "",
  "_FR_GR_OVERFLOW": 1,
  "_FR_LAYER_ADJUST": 0.05555296235777712,
  "_PINS_DISTANCE": 1,
  "_SDC_CLK_PERIOD": 573.0071762034466
}

### Expected Behavior

Should consistently work for both. 

### Environment

```shell
**Note this is a Docker build.**
**OR Commit** a515fc6cc97a7092efd51a28c1414e2fb4e53413

Unknown git commit, this is not a git repository.

Please make sure that you have the latest code changes and add the commit
hash in the description.

kernel: Linux 6.5.0-1020-gcp
os: Ubuntu 22.04.4 LTS (Jammy Jellyfish)
cmake version 3.24.2
CMake Warning at CMakeLists.txt:98 (message):
  OpenROAD git describe failed, using sha1 instead

-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: a515fc6cc97a7092efd51a28c1414e2fb4e53413
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS__-Wall
[... 415 lines cut ...]
  File "/usr/local/lib/python3.10/dist-packages/ray/tune/execution/tune_controller.py", line 728, in step
    if not self._actor_manager.next(timeout=0.1):
  File "/usr/local/lib/python3.10/dist-packages/ray/air/execution/_internal/actor_manager.py", line 224, in next
    self._actor_task_events.resolve_future(future)
  File "/usr/local/lib/python3.10/dist-packages/ray/air/execution/_internal/event_manager.py", line 113, in resolve_future
    on_error(e)
  File "/usr/local/lib/python3.10/dist-packages/ray/air/execution/_internal/actor_manager.py", line 770, in on_error
    self._actor_task_failed(
  File "/usr/local/lib/python3.10/dist-packages/ray/air/execution/_internal/actor_manager.py", line 291, in _actor_task_failed
    raise RuntimeError(
RuntimeError: Caught unexpected exception: Task was killed due to the node running low on memory.
Memory on the node (IP: 172.17.0.2, ID: 08613a1644d280fea0efdb84f99710ee5d2bad64a1e6695d76856684) where the task (actor ID: 9dffce11f3bad85eb563d0f701000000, name=AutoTunerBase.__init__, pid=1104623, memory used=0.10GB) was running was 61.61GB / 62.78GB (0.981373), which exceeds the memory usage threshold of 0.95. Ray killed this worker (ID: 925b600069bbd123f1a9af55f3a3fa75842dca461ab17730c77e0f0f) because it was the most recently scheduled task; to see more information about memory usage on this node, use `ray logs raylet.out -ip 172.17.0.2`. To see the logs of the worker, use `ray logs worker-925b600069bbd123f1a9af55f3a3fa75842dca461ab17730c77e0f0f*out -ip 172.17.0.2. Top 10 memory users:
PID     MEM(GB) COMMAND
1124766 58.88
1103582 0.49    python3 distributed.py --design aes-block --platform asap7 --config ../../../../flow/designs/asap7/a...
1104623 0.10    ray::AutoTunerBase.train
1103614 0.09    /usr/local/lib/python3.10/dist-packages/ray/core/src/ray/gcs/gcs_server --log_dir=/tmp/ray/session_2...
1103685 0.07    /usr/bin/python3 /usr/local/lib/python3.10/dist-packages/ray/dashboard/dashboard.py --host=127.0.0.1...
1103814 0.06    /usr/bin/python3 -u /usr/local/lib/python3.10/dist-packages/ray/dashboard/agent.py --node-ip-address...
1103844 0.04    ray::IDLE
1103845 0.04    ray::IDLE
1103834 0.04    ray::IDLE
1103839 0.04    ray::IDLE
Refer to the documentation on how to address the out of memory issue: https://docs.ray.io/en/latest/ray-core/scheduling/ray-oom-prevention.html. Consider provisioning more memory on this node or reducing task parallelism by requesting more CPUs per task. Set max_restarts and max_task_retries to enable retry when the task crashes due to OOM. To adjust the kill threshold, set the environment variable `RAY_memory_usage_threshold` when starting Ray. To disable worker killing, set the environment variable `RAY_memory_monitor_refresh_ms` to zero.
```

gudeh, 2024-06-12:
Hi @luarss, could you run your issue again with the newer OR version? I believe your issue might have been solved.

luarss, 2024-06-16:
@gudeh Can verify it no longer crashes at GPL stage. But it crashes at GRT stage instead. Will open a separate issue for this.

GRT crash logs:
```
OpenROAD 95fab1bf0354d1b0659023b217a20af990c93c20 
Features included (+) or not (-): +Charts +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ORD-0030] Using 1 thread(s).
global_route -guide_file ./results/asap7/aes-block/test-tune-2024-06-15-14-22-56/variant-AutoTunerBase-0b6d017e-or/route.guide -congestion_report_file ./reports/asap7/aes-block/test-tune-2024-06-15-14-22-56/variant-AutoTunerBase-0b6d017e-or/congestion.rpt -congestion_iterations 30 -congestion_report_iter_step 5 -verbose
[INFO GRT-0020] Min routing layer: M1
[INFO GRT-0021] Max routing layer: Pad
[INFO GRT-0022] Global adjustment: 0%
[INFO GRT-0023] Grid origin: (0, 0)
[INFO GRT-0043] No OR_DEFAULT vias defined.
[INFO GRT-0088] Layer M1      Track-Pitch = 0.0360  line-2-Via Pitch: 0.0360
[INFO GRT-0088] Layer M2      Track-Pitch = 0.0390  line-2-Via Pitch: 0.0360
[INFO GRT-0088] Layer M3      Track-Pitch = 0.0360  line-2-Via Pitch: 0.0360
[INFO GRT-0088] Layer M4      Track-Pitch = 0.0480  line-2-Via Pitch: 0.0480
[INFO GRT-0088] Layer M5      Track-Pitch = 0.0480  line-2-Via Pitch: 0.0480
[INFO GRT-0088] Layer M6      Track-Pitch = 0.0640  line-2-Via Pitch: 0.0640
[INFO GRT-0088] Layer M7      Track-Pitch = 0.0640  line-2-Via Pitch: 0.0640
[INFO GRT-0088] Layer M8      Track-Pitch = 0.0800  line-2-Via Pitch: 0.0800
[INFO GRT-0088] Layer M9      Track-Pitch = 0.0800  line-2-Via Pitch: 0.1100
[INFO GRT-0088] Layer Pad     Track-Pitch = 0.0800  line-2-Via Pitch: 2.0700
[INFO GRT-0019] Found 47 clock nets.
[INFO GRT-0001] Minimum degree: 2
[INFO GRT-0002] Maximum degree: 138
[INFO GRT-0003] Macros: 21
[INFO GRT-0043] No OR_DEFAULT vias defined.
[INFO GRT-0004] Blockages: 63508

[INFO GRT-0053] Routing resources analysis:
          Routing      Original      Derated      Resource
Layer     Direction    Resources     Resources    Reduction (%)
---------------------------------------------------------------
M1         Vertical     728801764       2113687          99.71%
M2         Horizontal   631633408      433238382          31.41%
M3         Vertical     728801764      637370533          12.55%
M4         Horizontal   534452048      468936367          12.26%
M5         Vertical     534459332      408733813          23.52%
M6         Horizontal   388695152      270434008          30.43%
M7         Vertical     388697696      338131332          13.01%
M8         Horizontal   291528936      242900393          16.68%
M9         Vertical     194380928      145774460          25.01%
Pad        Horizontal      75720         75710          0.01%
---------------------------------------------------------------
```

closed 2024-06-16 by luarss