The-OpenROAD-Project/OpenROAD-flow-scripts #2049: Error when running build_openroad.sh

aarontan0219, 2024-06-08:
### Subject

[Build] for compilation-related issues.

### Describe the bug

Keep getting error code when running the code
`./build_openroad.sh --local`

### Expected Behavior

Build Successful.

### Environment

```shell
-bash: ./etc/Env.sh: No such file or directory
```

### To Reproduce

No supportive file

### Relevant log output

```shell
[INFO FLW-0027] Saving logs to build_openroad.log
[INFO FLW-0028] ./build_openroad.sh --local
[INFO FLW-0002] Updating git submodules.
[INFO FLW-0001] Using local build method. This will create binaries at 'tools/install' unless overwritten.
[INFO FLW-0017] Compiling Yosys.
make: Entering directory '/home/aarontan/OpenROAD-flow-scripts/tools/yosys'
mkdir -p /home/aarontan/OpenROAD-flow-scripts/tools/install/yosys/bin
cp yosys yosys-config yosys-abc yosys-filterlib yosys-smtbmc yosys-witness /home/aarontan/OpenROAD-flow-scripts/tools/install/yosys/bin
strip -S /home/aarontan/OpenROAD-flow-scripts/tools/install/yosys/bin/yosys
strip /home/aarontan/OpenROAD-flow-scripts/tools/install/yosys/bin/yosys-abc
strip /home/aarontan/OpenROAD-flow-scripts/tools/install/yosys/bin/yosys-filterlib
mkdir -p /home/aarontan/OpenROAD-flow-scripts/tools/install/yosys/share/yosys
cp -r share/. /home/aarontan/OpenROAD-flow-scripts/tools/install/yosys/share/yosys/.
make: Leaving directory '/home/aarontan/OpenROAD-flow-scripts/tools/yosys'
[INFO FLW-0018] Compiling OpenROAD.
-- OpenROAD version: v2.0-14005-gf37c24a6c
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Install prefix: /home/aarontan/OpenROAD-flow-scripts/tools/install/OpenROAD
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- Performing Test C_COMPILER_SUPPORTS__-std=c++17
-- Performing Test C_COMPILER_SUPPORTS__-std=c++17 - Failed
-- Performing Test CXX_COMPILER_SUPPORTS__-std=c++17
-- Performing Test CXX_COMPILER_SUPPORTS__-std=c++17 - Success
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- TCL readline library: /usr/lib/x86_64-linux-gnu/libtclreadline.so
-- TCL readline header: /usr/include/x86_64-linux-gnu
-- Found SWIG: /home/aarontan/OpenROAD-flow-scripts/dependencies/bin/swig (found suitable version "4.1.0", minimum required is "4.0")
-- Using SWIG >= 4.1.0 -flatstaticmethod flag for python
-- boost: 1.80.0
-- spdlog: 1.8.1
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE)
-- STA version: 2.5.0
-- STA git sha: e01d3f163f483f233db00410b6515a767a6ca03b
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Build CXX_FLAGS: -O3 -DNDEBUG
-- Install prefix: /home/aarontan/OpenROAD-flow-scripts/tools/install/OpenROAD
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
[... 359 lines cut ...]
[ 85%] Building CXX object src/pdn/src/CMakeFiles/pdn.dir/grid.cpp.o
[ 85%] Building CXX object src/ppl/CMakeFiles/ppl.dir/src/Netlist.cpp.o
[ 85%] Building CXX object src/ppl/CMakeFiles/ppl.dir/src/SimulatedAnnealing.cpp.o
[ 85%] Building CXX object src/ppl/CMakeFiles/ppl.dir/src/Slots.cpp.o
[ 85%] Building CXX object src/pdn/src/CMakeFiles/pdn.dir/grid_component.cpp.o
[ 85%] Building CXX object src/pdn/src/CMakeFiles/pdn.dir/rings.cpp.o
[ 85%] Building CXX object src/pdn/src/CMakeFiles/pdn.dir/straps.cpp.o
[ 85%] Linking CXX static library pad.a
[ 85%] Building CXX object src/pdn/src/CMakeFiles/pdn.dir/shape.cpp.o
[ 85%] Building CXX object src/pdn/src/CMakeFiles/pdn.dir/via.cpp.o
[ 85%] Built target pad
[ 85%] Building CXX object src/pdn/src/CMakeFiles/pdn.dir/connect.cpp.o
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [src/pdn/src/CMakeFiles/pdn.dir/build.make:184: src/pdn/src/CMakeFiles/pdn.dir/grid.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
[ 85%] Linking CXX static library fin.a
[ 85%] Built target fin
[ 85%] Linking CXX static library ppl.a
[ 85%] Built target ppl
[ 85%] Linking CXX static library dbSta.a
[ 85%] Built target dbSta
gmake[1]: *** [CMakeFiles/Makefile2:6271: src/pdn/src/CMakeFiles/pdn.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
```

rovinski, 2024-06-08:
You are likely running out of memory. By default the build script uses # jobs = # cores to build in parallel. Try `./build_openroad.sh --local --threads 4` to only build 4 at a time.

aarontan0219, 2024-06-08:
Thanks man! It works now

closed 2024-06-08 by aarontan0219