The-OpenROAD-Project/OpenROAD-flow-scripts #2094: Failure to build ORFS docker image: unknown flag: --progress

oharboe, 2024-06-29:
### Subject

[Documentation] for documentation errors.

### Describe the bug

```
$ git log -1
commit 41e7d95599c733b8c6a4bfba39440daed8cb5923 (HEAD, origin/master, origin/HEAD)
Merge: c70eda40 c0481dd6
Author: Vitor Bandeira <vvbandeira@users.noreply.github.com>
Date:   Fri Jun 28 05:14:41 2024 -0700

    Merge pull request #2088 from The-OpenROAD-Project/update-openroad
    
    Update OpenROAD submodule
$ sudo ./setup.sh 
[deleted]
$ ./build_openroad.sh --latest
[INFO FLW-0027] Saving logs to build_openroad.log
[INFO FLW-0028] ./build_openroad.sh --latest
[INFO FLW-0002] Updating git submodules.
Submodule path 'tools/OpenROAD': checked out 'bf8a6a7a1c86511cbcfd56556743d1af4c0f2ac0'
[INFO FLW-0004] Updating OpenROAD app to the HEAD  of origin/master.
Previous HEAD position was bf8a6a7a1 Merge pull request #5299 from eder-matheus/grt_log
HEAD is now at 6dfa697a1 Merge pull request #5295 from amuritna/doc-suggestions
From https://github.com/The-OpenROAD-Project/OpenROAD
 * branch                master     -> FETCH_HEAD
Already up to date.
[INFO FLW-0000] Using docker build method.[INFO FLW-0020] Building docker image for OpenROAD Flow.
Create docker image openroad/flow-ubuntu22.04-dev:41e7d95599c733b8c6a4bfba39440daed8cb5923 using docker/Dockerfile.dev
DEPRECATED: The legacy builder is deprecated and will be removed in a future release.
            Install the buildx component to build images with BuildKit:
            https://docs.docker.com/go/buildx/

unknown flag: --progress
See 'docker build --help'.
```

### Expected Behavior

Compile and produce docker image without problems

### Environment

```shell
$ tools/OpenROAD/etc/Env.sh 
Git commit: 6dfa697a121926d062c52bdc286d9219d4843b1f
kernel: Linux 6.5.0-41-generic
os: Ubuntu 23.10 (Mantic Minotaur)
cmake version 3.27.4
-- The CXX compiler identification is GNU 13.2.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-14343-g6dfa697a1
-- System name: Linux
-- Compiler: GNU 13.2.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 13.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
[... 114 lines cut ...]
-- boost: 1.74.0
-- Found Python3: /usr/include/python3.11 (found version "3.11.6") found components: Development Development.Module Development.Embed 
-- Found ZLIB: /usr/lib/x86_64-linux-gnu/libz.so (found version "1.2.13")  
-- spdlog: 1.10.0
-- Found BISON: /usr/bin/bison (found version "3.8.2") 
-- Found Doxygen: /usr/bin/doxygen (found version "1.9.4") found components: doxygen dot 
-- STA version: 2.5.0
-- STA git sha: ee21333caa14eac641bcccbf2a36b4f6b4ba7de1
-- System name: Linux
-- Compiler: GNU 13.2.0
-- Build type: RELEASE
-- Build CXX_FLAGS: -O3 -DNDEBUG
-- Install prefix: /usr/local
-- Found FLEX: /usr/bin/flex (found version "2.6.4") 
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- SSTA: 0
-- Found SWIG: /usr/bin/swig4.0 (found suitable version "4.1.0", minimum required is "3.0")  
-- STA executable: /home/oyvind/docker-ORFS/tools/OpenROAD/src/sta/app/sta
-- Configuring incomplete, errors occurred!
```

### To Reproduce

See above

vvbandeira, 2024-06-29:
@oharboe
What version of Docker are you using?

oharboe, 2024-06-29:
```
$ docker --version
Docker version 24.0.5, build 24.0.5-0ubuntu1
```

oharboe, 2024-06-29:
No luck after upgrading:

```
$ docker --version
Docker version 24.0.7, build 24.0.7-0ubuntu2~23.10.1
oyvind@famine:~/docker-ORFS$ ./build_openroad.sh --latest
[INFO FLW-0027] Saving logs to build_openroad.log
[INFO FLW-0028] ./build_openroad.sh --latest
[INFO FLW-0002] Updating git submodules.
Submodule path 'tools/OpenROAD': checked out 'bf8a6a7a1c86511cbcfd56556743d1af4c0f2ac0'
[INFO FLW-0004] Updating OpenROAD app to the HEAD  of origin/master.
Previous HEAD position was bf8a6a7a1 Merge pull request #5299 from eder-matheus/grt_log
HEAD is now at 6dfa697a1 Merge pull request #5295 from amuritna/doc-suggestions
From https://github.com/The-OpenROAD-Project/OpenROAD
 * branch                master     -> FETCH_HEAD
Already up to date.
[INFO FLW-0000] Using docker build method.[INFO FLW-0020] Building docker image for OpenROAD Flow.
Create docker image openroad/flow-ubuntu22.04-dev:41e7d95599c733b8c6a4bfba39440daed8cb5923 using docker/Dockerfile.dev
DEPRECATED: The legacy builder is deprecated and will be removed in a future release.
            Install the buildx component to build images with BuildKit:
            https://docs.docker.com/go/buildx/

unknown flag: --progress
See 'docker build --help'.
```

oharboe, 2024-06-29:
This fixes boards up the broken window so that it can be fixed properly later:

https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/2095

closed 2024-06-30 by vvbandeira