The-OpenROAD-Project/OpenROAD-flow-scripts #2103: XDG_RUNTIME_DIR error

Suru03, 2024-07-03:
### Subject

[Stage]: Other. Please describe below.

### Describe the bug

[image]

### Expected Behavior

After running the make_gui command ,I am supposed to get a nand gate. But because of this issue,I am unable to get it.

### Environment

```shell
the previous command was :source ./env.sh
yosys -help
openroad -help
cd flow
make
```

### To Reproduce

--

vijayank88, 2024-07-03:
Enable/Install XServer will resolve this issue.
Check similar #1873

Suru03, 2024-07-03:
any views on how to enable Install XServer

maliberty, 2024-07-03:
This sounds like a general X setup issue and not really related to OR.

vvbandeira, 2025-08-22:
Close due to inactivity. Please reopen if the problem is still happening on the latest version.

closed 2025-08-22 by vvbandeira