The-OpenROAD-Project/OpenROAD-flow-scripts #2160: Create codespaces does not work

oharboe, 2024-07-19:
### Subject

[Stage]: Other. Please describe below.

### Describe the bug

Create codespace on master. The default should be the ubuntu gui image:

[image]

Fails with:

```
====================================== ERROR ====================================
2024-07-19 05:03:32.771Z: Failed to create container.
=================================================================================
2024-07-19 05:03:32.771Z: Error: Command failed: docker inspect --type image ghcr.io/the-openroad-project/openroad-flow-scripts:lxqt
2024-07-19 05:03:32.774Z: Error code: 1302 (UnifiedContainersErrorFatalCreatingContainer)

====================================== ERROR ====================================
2024-07-19 05:03:32.782Z: Container creation failed.
=================================================================================
```

### Expected Behavior

No failure.

### Environment

```shell
v3.0-941-g3395328e
```

### To Reproduce

See above.

### Relevant log output

```shell
2024-07-19 05:02:53.142Z: Host information
2024-07-19 05:02:53.142Z: ----------------
2024-07-19 05:02:53.142Z: OS: Ubuntu 22.04.4 LTS (stable release)
2024-07-19 05:02:53.142Z: Image details: https://github.com/github/codespaces-host-images/blob/main/README.md
2024-07-19 05:02:53.142Z: ----------------

=================================================================================
2024-07-19 05:02:53.142Z: Configuration starting...
2024-07-19 05:02:53.165Z: Cloning...
2024-07-19 05:02:53.200Z: $ git -C "/var/lib/docker/codespacemount/workspace" clone --branch "master" --depth 1 https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts "/var/lib/docker/codespacemount/workspace/OpenROAD-flow-scripts"
2024-07-19 05:02:53.206Z: Cloning into '/var/lib/docker/codespacemount/workspace/OpenROAD-flow-scripts'...
2024-07-19 05:03:22.745Z: Updating files:  23% (533/2222)
2024-07-19 05:03:22.762Z: Updating files:  24% (534/2222)
2024-07-19 05:03:22.772Z: Updating files:  25% (556/2222)
2024-07-19 05:03:22.776Z: Updating files:  26% (578/2222)
2024-07-19 05:03:22.806Z: Updating files:  27% (600/2222)
2024-07-19 05:03:22.948Z: Updating files:  28% (623/2222)
2024-07-19 05:03:22.954Z: Updating files:  29% (645/2222)
2024-07-19 05:03:22.958Z: Updating files:  30% (667/2222)
2024-07-19 05:03:22.964Z: Updating files:  31% (689/2222)
2024-07-19 05:03:22.968Z: Updating files:  32% (712/2222)
2024-07-19 05:03:23.009Z: Updating files:  33% (734/2222)
2024-07-19 05:03:23.015Z: Updating files:  34% (756/2222)
2024-07-19 05:03:23.019Z: Updating files:  35% (778/2222)
2024-07-19 05:03:23.024Z: Updating files:  36% (800/2222)
2024-07-19 05:03:23.028Z: Updating files:  37% (823/2222)
2024-07-19 05:03:23.032Z: Updating files:  38% (845/2222)
2024-07-19 05:03:23.057Z: Updating files:  39% (867/2222)
2024-07-19 05:03:23.150Z: Updating files:  40% (889/2222)
[... 387 lines cut ...]
[1B2024-07-19 05:04:40.236Z: [1A[2K
bae61b87a7ab: Extracting  41.29MB/42.85MB
[1B2024-07-19 05:04:40.246Z: [1A[2K
bae61b87a7ab: Extracting  42.85MB/42.85MB
[1B2024-07-19 05:04:41.219Z: [1A[2K
bae61b87a7ab: Pull complete 
[1B2024-07-19 05:04:41.257Z: Digest: sha256:0a0177436379ab4eb06842b883a75bfd2a02062ad8f28c38962fa12610c141472024-07-19 05:04:41.257Z: 
2024-07-19 05:04:41.278Z: Status: Downloaded newer image for mcr.microsoft.com/devcontainers/base:alpine
2024-07-19 05:04:41.554Z: Container started2024-07-19 05:04:41.554Z: 
2024-07-19 05:04:41.790Z: Outcome: success User: vscode WorkspaceFolder: /workspaces/OpenROAD-flow-scripts
2024-07-19 05:04:41.800Z: devcontainer process exited with exit code 0

=================================================================================
2024-07-19 05:04:42.808Z: Running blocking commands...
2024-07-19 05:04:42.842Z: $ devcontainer up --id-label Type=codespaces --workspace-folder /var/lib/docker/codespacemount/workspace/OpenROAD-flow-scripts --mount type=bind,source=/.codespaces/agent/mount/cache,target=/vscode --user-data-folder /var/lib/docker/codespacemount/.persistedshare --container-data-folder .vscode-remote/data/Machine --container-system-data-folder /var/vscode-remote --log-level trace --log-format json --update-remote-user-uid-default never --mount-workspace-git-root false --omit-config-remote-env-from-metadata --skip-non-blocking-commands --expect-existing-container --config "/var/lib/docker/codespacemount/workspace/OpenROAD-flow-scripts/.devcontainer/devcontainer.json" --override-config /root/.codespaces/shared/merged_devcontainer.json --default-user-env-probe loginInteractiveShell --container-session-data-folder /workspaces/.codespaces/.persistedshare/devcontainers-cli/cache --secrets-file /root/.codespaces/shared/user-secrets-envs.json
2024-07-19 05:04:43.076Z: @devcontainers/cli 0.56.1. Node.js v18.20.3. linux 6.5.0-1022-azure x64.
2024-07-19 05:04:43.276Z: Outcome: success User: vscode WorkspaceFolder: /workspaces/OpenROAD-flow-scripts
2024-07-19 05:04:43.295Z: devcontainer process exited with exit code 0

=================================================================================
2024-07-19 05:04:43.351Z: Configuring codespace...

=================================================================================
2024-07-19 05:04:43.352Z: Finished configuring codespace.
```

glatosinski, 2024-07-19:
For the Codespaces to work correctly, the following steps are required:

* According to the discussion from https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/1927, an image for Codespaces should be created in Jenkins CI and pushed to the registry ([direct link to discussion](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/1927#discussion_r1672153285))
* After this, the `image` field in https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/d555e36a66e449a0ce6bdf9b2bf61cc1bd79c607/.devcontainer/devcontainer.json#L5 should be changed to the name of the built image

oharboe, 2024-08-20:
@vvbandeira @maliberty Any news?

oharboe, 2024-08-22:
@vvbandeira Please merge #2272 if you are happy with the CI part being done, to hook up the docker image and close this issue.

I'll see about filing some followup issues to improve the docker image later.

vvbandeira, 2024-08-22:
Fixed on #2272

closed 2024-08-22 by vvbandeira