The-OpenROAD-Project/OpenROAD-flow-scripts #2163: Extend power.tcl so that checks the power for acceptable numbers

oharboe, 2024-07-19:
### Description

Define some criterea for "it worked" and update https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/flow/designs/asap7/mock-array/power.tcl with that check(print error and exit 1).

To run:

```
make DESIGN_CONFIG=designs/asap7/mock-array/config.mk final simulate power
```

At the end `make power` outputs something like:

```
Found 1288 unannotated drivers.
Found 7560 partially unannotated drivers.
Group                  Internal  Switching    Leakage      Total
                          Power      Power      Power      Power (Watts)
----------------------------------------------------------------
Sequential             4.14e-01   1.16e-01   6.43e-06   5.30e-01  59.1%
Combinational          1.24e-01   1.00e-01   4.57e-06   2.25e-01  25.0%
Clock                  6.87e-02   7.36e-02   6.51e-07   1.42e-01  15.9%
Macro                  0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
Pad                    0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
----------------------------------------------------------------
Total                  6.07e-01   2.90e-01   1.16e-05   8.97e-01 100.0%
                          67.7%      32.3%       0.0%
Annotated 217546 pin activities.
Group                  Internal  Switching    Leakage      Total
                          Power      Power      Power      Power (Watts)
----------------------------------------------------------------
Sequential             1.70e+00   7.59e-01   6.46e-06   2.46e+00  60.0%
Combinational          8.28e-01   6.65e-01   4.75e-06   1.49e+00  36.5%
Clock                  6.87e-02   7.36e-02   6.51e-07   1.42e-01   3.5%
Macro                  0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
Pad                    0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
----------------------------------------------------------------
Total                  2.59e+00   1.50e+00   1.19e-05   4.09e+00 100.0%
                          63.4%      36.6%       0.0%
```

### Suggested Solution

Print error and exit 1 on unacceptable result in power.tcl

maliberty, 2024-07-19:
I don't see anything either so I'll open a request to sta

socclosure, 2024-07-20:
@tspyrou / @maliberty 

Do you want me to look into this issue for my first contribution to open Road project?

maliberty, 2024-07-20:
@socclosure I don't think this is a good candidate as the problem is in sta itself.   We do label good first issues.  Perhaps https://github.com/The-OpenROAD-Project/OpenROAD/issues/4041 ?

socclosure, 2024-07-20:
@maliberty Make sense. Thanks for information. I will look into Good first issues to start the contribution.

akashlevy, 2024-08-13:
Try `sta::design_power [sta::cmd_corner]`. You can replace `sta::cmd_corner` with `sta::find_corner [corner]` for a particular corner