The-OpenROAD-Project/OpenROAD-flow-scripts #2242: Flow crash when trying to place macro using mpl2 placement.tcl file

nchiolino, 2024-08-13:
### Subject

[Stage]: Floorplaning.

### Describe the bug

Trying to place two macros in specific locations using a placement.tcl file. When the macro placement step occurs, it receives a signal 8 and crashes. 
[image]

### Expected Behavior

I expect mpl2 to place the macros with no errors. I have been able to place the macros using the mpl tool and a .cfg macro placement file, however, the netlist is not correct compared to the synthesized one so I am getting incorrect routing nets for grt. I want to utilize the hierarchical synthesis and macro placement but keep receiving this signal 8 issue.

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.

It is recommened to pull the latest changes.

If problem persists, file a github issue with the re-producible test case.

kernel: Linux 6.5.0-44-generic

os: Ubuntu 22.04.4 LTS (Jammy Jellyfish)

cmake version 3.24.2

-- The CXX compiler identification is GNU 11.4.0

-- Detecting CXX compiler ABI info

-- Detecting CXX compiler ABI info - done

-- Check for working CXX compiler: /usr/bin/c++ - skipped

-- Detecting CXX compile features

-- Detecting CXX compile features - done

-- OpenROAD version: v2.0-13893-gf0cc0ee1a

-- System name: Linux

-- Compiler: GNU 11.4.0

-- Build type: RELEASE

-- Install prefix: /usr/local

-- C++ Standard: 17

-- C++ Standard Required: ON

-- C++ Extensions: OFF

-- The C compiler identification is GNU 11.4.0

-- Detecting C compiler ABI info

-- Detecting C compiler ABI info - done

-- Check for working C compiler: /usr/bin/cc - skipped

-- Detecting C compile features

-- Detecting C compile features - done

-- Found Python: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter 

[... 239 lines cut ...]

-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization system thread 

-- Found Boost: /usr/local/lib/Number of processor cores: 4

cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0")  

-- Found Eigen3: /usr/local/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.1") 

-- TCL readline enabled

-- Tcl Extended disabled

-- Python3 enabled

-- Configuring done

-- Generating done

-- Build files have been written to: /tmp/tmp.Qh4kMbv7LM
```

### To Reproduce

I am running a proprietary pdk and cannot provide a working example at the moment.

maliberty, 2024-08-13:
Are you using MACRO_PLACEMENT_TCL or MACRO_PLACEMENT to place the macros?  If so the macro placer should be skipped.

nchiolino, 2024-08-13:
I am using MACRO_PLACEMENT_TCL. That is what it is crashing on when called. I am assuming.

nchiolino, 2024-08-13:
Its demonstrating to me that it might be a resource problem. I am working on increasing my resources.

nchiolino, 2024-08-14:
@maliberty , when you say that the macro placer can be skipped if I use a placement file with exact locations, do you mean that I don't need to set the RTLMP_FLOW variable?

maliberty, 2024-08-14:
RTLMP_FLOW will be ignored if MACRO_PLACEMENT_TCL is set and the macro placer won't be run either.  It makes me think there  is something wrong in how you are setting it.  You can see the logic in https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/ee899c351818915c3aa721ff4c29fe9f7fce6df0/flow/scripts/macro_place_util.tcl#L47 .  Do you see the "Using manual macro placement file" message?

nchiolino, 2024-08-14:
To provide more clarification, I have had success placing macros setting SYNTH_HIERARCHICAL = 0 and using a macro_placement.cfg file. Since my top level RTL is assembled in a hierarchical fashion AND mpl2 is a better tool, I was hoping to get macro placement using SYNTH_HIERARCHICAL = 1. Also, I was getting some very strange results with PAD net names wanting to route to internal core gates during grt. I assumed it was a failure with not using the hierarchical method. Therefore, I am trying to blaze a trail using the hierarchical synth and macro placement.
For the question you asked just now. Here is a screenshot of my terminal just after floorplan and when it calls the macro placement and it does look that its not even getting to the line of code you mentioned before. I have only seen that line of code when doing the non-hierarchical method mentioned above. Does that help clarify my journey thus far? (note: I have blacked out a couple of lines of code for protection of sensitive information)
[image]

nchiolino, 2024-08-14:
After some code debug with print statements, I was able to find out that the crash is occurring on the first line of my macro placement tcl file. Here is a screenshot of my file. Its very simple and the two memory macros are hierarchecly under the top level soc chip. Is there an issue with my syntax perhaps?
[image]

maliberty, 2024-08-14:
We are getting closer.  Since you can't share you test case, would you build for debug and get exact line info?  Does you design have routing tracks?

nchiolino, 2024-08-14:
Yes, the design has routing tracks and the X Y locations are whole numbers of the site location. During the exercise for building the test case with a "dummy" macro, for some reason, it worked. This is even with the proprietary technology. So now I am wondering if its an issue with the macro itself. I will attempt to insert the memory macro into the test case to isolate that variable. 
Stay tuned.

maliberty, 2024-08-14:
That's a good clue.  Is it possible you macro has pins without any physical geometry?

nchiolino, 2024-08-15:
Due to deadline issues, I am moving back to flat synthesis and utilizing mpl for macro placement. I will have to pick this up if I have spare time.

maliberty, 2024-08-15:
Ok, reopen if you decide to pick it up.

closed 2024-08-15 by maliberty