The-OpenROAD-Project/OpenROAD-flow-scripts #2258: Missing PDN vias

nchiolino, 2024-08-16:
### Subject

[Stage]: Floorplaning.

### Describe the bug

During PDN I am getting warnings that vias could not be inserted on some power straps to VDD and VSS. I saw similar issues in the past but could not find a good solution to my problem. It seems to be related to resolution but when I turn down the resolution it places vias off grid. 

### Expected Behavior

I expect vias to be inserted in all power strap locations that are on grid.

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.

It is recommened to pull the latest changes.

If problem persists, file a github issue with the re-producible test case.

kernel: Linux 6.5.0-44-generic

os: Ubuntu 22.04.4 LTS (Jammy Jellyfish)

cmake version 3.24.2

-- The CXX compiler identification is GNU 11.4.0

-- Detecting CXX compiler ABI info

-- Detecting CXX compiler ABI info - done

-- Check for working CXX compiler: /usr/bin/c++ - skipped

-- Detecting CXX compile features

-- Detecting CXX compile features - done

-- OpenROAD version: v2.0-13893-gf0cc0ee1a

-- System name: Linux

-- Compiler: GNU 11.4.0

-- Build type: RELEASE

-- Install prefix: /usr/local

-- C++ Standard: 17

-- C++ Standard Required: ON

-- C++ Extensions: OFF

-- The C compiler identification is GNU 11.4.0

-- Detecting C compiler ABI info

-- Detecting C compiler ABI info - done

-- Check for working C compiler: /usr/bin/cc - skipped

-- Detecting C compile features

-- Detecting C compile features - done

-- Found Python: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter 

-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
[... 243 lines cut ...]
cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0")  

-- Found Eigen3: /usr/local/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.1") 

-- TCL readline enabled

-- Tcl Extended disabled

-- Python3 enabled

-- Configuring done

-- Generating done

-- Build files have been written to: /tmp/tmp.EPU85IzXOw
```

### To Reproduce

I am working with a proprietary technology and cannot provide technical files for reproduction.

### Screenshots

This image is demonstrating that vias are being inserted into neighboring power straps but not the two internal straps from the edge. This is happening for every power pad to core on every side for the same exact straps.
[image]

gadfort, 2024-08-17:
@nchiolino 
1. what do you mean by changing resolution?
2. Did this work prior to (https://github.com/The-OpenROAD-Project/OpenROAD/pull/5541) or has this always been an issue?

without a testcase it would be pretty hard to debug, but clarification on the above questions would help.

nchiolino, 2024-08-20:
@gadfort . I apologize for the delay in responding to you. I was dealing with a separate LEF issue that I have ironed out. 
Right now this technology is set for a certain manufacturing grid in the LEF file. If I decrease that grid setting by an order of magnitude, it will insert all the via arrays for the pad power straps to the internal power ring. However, this is not a valid solution because all the vias are off grid during DRC. So if I am to use the correct manufacturing grid setting, those power straps in the image above will not populate. So I am wondering if it is a pad position issue or am if I have set up the via generation incorrectly? What script controls the creation of those power strap vias?

nchiolino, 2024-08-20:
Normally I would try to repeat this condition with one of the technologies native to ORFS, but for this instance I am afraid the issue is dependent on this technology I am using and unfortunately I cannot share any technology specific information. 
I really do apologize for that headache.

nchiolino, 2024-08-20:
I cannot repeat the issue with sky130hd so it must be something with how the via arrays are being generated.

gadfort, 2024-08-20:
@nchiolino VIARULEs are used when available, but you would have to force PDN to use it if it's the only one you want to use.
I think the issue you might be running into is if the width of the strap is a odd number of manufacturing grid points and the vias are likely going to be an even number. The vias will be rejected by PDN if they cannot fit under the strap cleanly.
Would you be able to report the product of the lef units and manufacturing grid? If not, maybe just let know if it is >= 5. And can you look at your setup and see how wide the straps are in relation to the manufacturing grid?
Are you using any kind of snap to routing tracks? If so, we would likely need to double check some things there as well (the pitch and offset of the tracks in manufacturing grid points.

nchiolino, 2024-08-20:
Ok, that makes sense on the VIARULEs, thank you for explaining that. After thinking about this more, I have elected to fix the layout of the PAD cells to simply be whole numbers, instead of a width that does not work for the DRC rules required for the vias. This has benefits for lithographic processing as well. I was able to confirm this fix in ORFS. There is no issue with ORFS. 
Thank you very much for the support.

gadfort, 2024-08-21:
@nchiolino I know you decided to try something else, but I think you happened upon an actual issue in PDN. Would you be willing to to see if: https://github.com/gadfort/OpenROAD/tree/pdn-grid-checks work on your unmodified example? If so, this ensures that the widths of straps are an even multiple of the grid which I think should avoid the issue you ran into

nchiolino, 2024-08-21:
Excellent. Thank you so much for sending me that. Unfortunately right now I am under a tight deadline so I cant prioritize this until later. I will mark it for a later date.
Thank you