The-OpenROAD-Project/OpenROAD-flow-scripts #2273: make clean_all doesn't clean BLOCKS data

jeffng-or, 2024-08-21:
### Subject

[Flow] for any util, flow Makefile, or flow script issues.

### Describe the bug

make clean_all for any design that uses BLOCKS (e.g. mock-array) doesn't clean the results/logs/objects for any of the blocks.

### Expected Behavior

The rm commands that are part of clean_all should also clean up any results, objects, logs for any of the BLOCKS associated with the design.

### Environment

```shell
Git commit: 42ba30d730851a73bd2996c5673a05fd79520f87
kernel: Linux 6.5.0-1025-gcp
os: Ubuntu 22.04.4 LTS (Jammy Jellyfish)
cmake version 3.24.2
-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-15311-g42ba30d73
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function
[... 114 lines cut ...]
-- Generating done
-- Build files have been written to: /tmp/tmp.5WvnqTPYDg
```

### To Reproduce

> ls -l results/asap7/mock-array*/base

ls: cannot access 'results/asap7/mock-array*/base': No such file or directory (expected since nothing has been created yet)

> make DESIGN_CONFIG=designs/asap7/mock-array/config.mk

build successful

> ls -l results/asap7/mock-array*/base

lots of files under mock-array and mock-array-Element

> make DESIGN_CONFIG=designs/asap7/mock-array/config.mk clean_all

mock-array-Element missing from rm commands

> ls -l results/asap7/mock-array*/base

mock-array-Element directory and contents still there

maliberty, 2024-08-21:
I wonder if you can generate a clean rule in the same way we generate the build rules for the blocks.  We don't write new rules for them but instead use the `$(foreach block,$(BLOCKS),$(eval ...`

jeffng-or, 2024-08-21:
Yeah, that's probably better. So, just to make sure I'm in sync with you:

1. Define a clean_macros target that will clean the final LEF, Liberty, GDS, CDL and log files for the macro, as well as recursively call make clean_all on the macro to clean out all of the ODB, SDC data from the macro's build stage
2. Add clean_macros to the clean_all target

That way, if someone calls clean_synth, then only the synth data for the top level is cleaned (i.e. the macros aren't touched). If someone wants to clean the macros specifically, they can call make clean_macros

maliberty, 2024-08-21:
I think it good to separate them.  We could have a target to clean both.