The-OpenROAD-Project/OpenROAD-flow-scripts #2294: GUI crashes after make gui_final tries to load

nchiolino, 2024-08-27:
### Subject

[Stage]: Other. Please describe below.

### Describe the bug

After all steps have completed (> stage 6), and make gui_final is called the OpenROAD gui pulls up then crashes with signal 11. 

### Expected Behavior

I expect the gui to load the design with all timing information included.

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.

It is recommened to pull the latest changes.

If problem persists, file a github issue with the re-producible test case.

kernel: Linux 6.8.0-40-generic

os: Ubuntu 22.04.4 LTS (Jammy Jellyfish)

cmake version 3.24.2

-- The CXX compiler identification is GNU 11.4.0

-- Detecting CXX compiler ABI info

-- Detecting CXX compiler ABI info - done

-- Check for working CXX compiler: /usr/bin/c++ - skipped

-- Detecting CXX compile features

-- Detecting CXX compile features - done

-- OpenROAD version: v2.0-13893-gf0cc0ee1a

-- System name: Linux

-- Compiler: GNU 11.4.0

-- Build type: RELEASE

-- Install prefix: /usr/local

-- C++ Standard: 17

-- C++ Standard Required: ON

-- C++ Extensions: OFF

-- The C compiler identification is GNU 11.4.0

-- Detecting C compiler ABI info

-- Detecting C compiler ABI info - done

-- Check for working C compiler: /usr/bin/cc - skipped

-- Detecting C compile features

-- Detecting C compile features - done

-- Found Python: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter 

-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
[... 248 lines cut ...]

-- Tcl Extended disabled

-- Python3 enabled

-- Configuring done

-- Generating done

-- Build files have been written to: /tmp/tmp.bKvOJXGvfS
```

### To Reproduce

Unfortunately I am working with a proprietary technology so I cannot provide files for reproduction.

### Additional Context

From what I have gathered from debuging this issue, it looks like the crash is happening in gui.tcl at line 50: 
```
 } elseif {$design_stage >= 5} {
    log_cmd estimate_parasitics -global_routing
```
I am not creating a spef file so it skips that if statement and tried to estimate parasitics then crashes. This aligns with the message I am getting on the output after calling make gui_final.
[image]

nchiolino, 2024-08-27:
Thank you for the reply. Can you help me out by walking me through what this download is?

maliberty, 2024-08-27:
There is a general attack of posting fake/malicious downloads.  We have been deleting such comments as we find them.

nchiolino, 2024-08-27:
Ah yes. I had a suspicion. Thanks for confirming!

maliberty, 2024-08-27:
Do you get a stack trace?

nchiolino, 2024-08-27:
Yes! My apologies. Let me know if you need more info.
[image]

nchiolino, 2024-08-27:
Let me know if I need to paste the txt as well.

maliberty, 2024-08-27:
I would guess you are getting a null dbTechLayer somehow.  @eder-matheus any guesses?

nchiolino, 2024-08-27:
It does seem to be related to grt in someway. When I try to load a smaller design with the same technology it loads no problem. It has something to do with the size of the design. If that is of any help at all.

eder-matheus, 2024-08-27:
@nchiolino  It seems it's trying to estimate parasitics from grt in a script that you didn't run the global_route command.
We recently added a guard where the tool should throw an error message instead of crashing. Could you try the latest version of the OpenROAD master branch? You can do it by building the ORFS tools with the `--latest` flag

nchiolino, 2024-08-28:
@eder-matheus . Interesting, are you saying that I am possibly missing a script that is supposed to have a global_route command in it? 
Unfortunately this close to the deadline I would really prefer not to change tool versions. Its not a major issue for now, I was just wanting to load the timing information with the gui. Is there a way to get the gui to load the final design with the timing information?

maliberty, 2024-08-28:
`make gui_final`

nchiolino, 2024-08-28:
@maliberty . I apologize for the confusion but `make gui_final` is the command that I am using that is resulting in this crash. Right now I am loading the design database after running `openroad -gui`. How do I load the timing information in the gui after it opens?

maliberty, 2024-08-28:
The gui.tcl script shows the steps.  You'll want to load the Liberty files before the odb.

Why would you be using estimate_parasitics with a final db.  There is an extracted spef file to load.

closed 2024-08-28 by nchiolino