The-OpenROAD-Project/OpenROAD-flow-scripts #2298: Error: 'abc' is not configured as a git submodule.

jaejunllee, 2024-08-28:
### Subject

[Flow] for any util, flow Makefile, or flow script issues.

### Describe the bug

When executing build_openroad.sh in a centos7.4 Linux environment, the following error message occurs.
Error : 'abc' is not configured as a git submodule.
[image]

### Expected Behavior

I would like to know how to fix the error I shared.

### Environment

```shell
Unknown git commit, this is not a git repository.

please make sure that you have the latest code changed and add the commit hash in the description.

kernel: Linux 3.10.0-693.el7.x86_64
os : Red Hat Enterprise Linux Server 7.4 (Maipo)
cmake version 2.8.12.2
Cmake Error : The source directory ~~~
```

### To Reproduce

When executing build_openroad.sh in a centos7.4 Linux environment, the following error message occurs.
Error : 'abc' is not configured as a git submodule.

maliberty, 2024-08-28:
What steps did you do before running the build script.  This seems to be a git error and nothing platform specific.

jaejunllee, 2024-08-29:
> What steps did you do before running the build script. This seems to be a git error and nothing platform specific.

Before running the script, I performed the following steps
git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts
cd OpenROAD-flow-scripts
sudo ./setup.sh

jaejunllee, 2024-08-29:
> @jaejunllee You are likely updating your local clone from an old version of ORFS. A couple of weeks back, Yosys changed the way they handle `abc`. To fix this try:
> 
> ```
> cd tools/yosys
> git clean -xdf
> git submodule sync
> git submodule update --init --recursive
> ```

Thank you for your answer.
I tried the method you suggested, but I get the same error....

maliberty, 2024-08-29:
Can you just clone a new area?

jaejunllee, 2024-08-29:
I didn't understand what you said, could you explain it to me in more detail?

maliberty, 2024-08-29:
Just clone ORFS again in a new area as if for the first time.

jaejunllee, 2024-08-29:
I tried again after cloning ORFS to a new region from scratch. 
However, I keep getting the error I shared before...

Error : 'abc' is not configured as a git submodule.

vvbandeira, 2024-08-29:
> If you go to the path below, you will see that the "abc" file has been changed to "abc @ 28d955c"

This is just a GH notation to indicate that `abc` is a submodule and it is pointing to the commit hash `28d955c` of that repo.

Another thing that I found odd is the `Unknown git commit, this is not a git repository.` from your `Env.sh` run. Where did you run the command?

maliberty, 2024-08-29:
Please give the exact sequence of commands you are giving from nothing to failure.

jaejunllee, 2024-08-29:
Start at the following path : /home/jjunjun/ 
1. git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts
2. cd OpenROAD-flow-scripts
3. sudo ./setup.sh (dir: home/jjunjun/OpenROAD-flow-scripts
4. ./build_openroad.sh --local

Please let me know if you need any further information

maliberty, 2024-09-05:
I did these steps without error.  Please try it in a fresh directory as I think you may have an older result that is interfering.  Note that we don't build or test on Centos7 anymore as it is EOL.

vvbandeira, 2025-02-06:
Closing due to inactivity and since CentOS7 is EOL we cannot reproduce ourselves.

closed 2025-02-06 by vvbandeira