The-OpenROAD-Project/OpenROAD-flow-scripts #2305: Global Routing is failing due to High Congetion

Wiki1998-dev, 2024-08-28:
### Subject

[Stage]: Global Router.

### Describe the bug

hello, 
Firstly my design is showing congestion too high .I have tried different values for placement density ,Halo but it gives same issue.
how can i understand what to set .Can you kindly guide/suggest me.

export SYNTH_HIERARCHICAL = 1
export MAX_UNGROUP_SIZE = 100
# Adders degrade setup repair
export ADDER_MAP_FILE :=

export CORE_UTILIZATION = 20
export CORE_ASPECT_RATIO = 1
export CORE_MARGIN = 2

export PLACE_DENSITY_LB_ADDON = 0.2
export PLACE_DENSITY          = 0.2

export RTLMP_FLOW = True

export MACRO_PLACE_HALO = 70 70
export MACRO_PLACE_CHANNEL = 140 140

### Expected Behavior

It should keep Running

### Environment

```shell
I dont find this file in my system i have just installed the ORFS from prebuilt libraries with latest release
(base) syed@syed-Latitude-E5450:~/OpenROAD-flow-scripts/flow$ which openroad
/usr/bin/openroad
(base) syed@syed-Latitude-E5450:/usr/bin$ openroad
OpenROAD 42ba30d730851a73bd2996c5673a05fd79520f87 
Features included (+) or not (-): +Charts +GPU +GUI +Python : None
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
openroad>
```

### To Reproduce

https://drive.google.com/file/d/1KasNChJvnRrOwvnGMV3AjRBe5ORF0xrj/view?usp=sharing
it is global route issue   tar i am uploading but incase you need anything more please let me know

### Relevant log output

```shell
OpenROAD 42ba30d730851a73bd2996c5673a05fd79520f87 
Features included (+) or not (-): +Charts +GPU +GUI +Python : None
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ORD-0030] Using 4 thread(s).
global_route -congestion_report_file ./reports/sky130hd/core_v_mini_mcu/base/congestion.rpt -congestion_iterations 30 -congestion_report_iter_step 5 -verbose
[INFO GRT-0020] Min routing layer: met1
[INFO GRT-0021] Max routing layer: met5
[INFO GRT-0022] Global adjustment: 0%
[INFO GRT-0023] Grid origin: (0, 0)
[INFO GRT-0043] No OR_DEFAULT vias defined.
[INFO GRT-0088] Layer li1     Track-Pitch = 0.4600  line-2-Via Pitch: 0.3400
[INFO GRT-0088] Layer met1    Track-Pitch = 0.3400  line-2-Via Pitch: 0.3400
[INFO GRT-0088] Layer met2    Track-Pitch = 0.4600  line-2-Via Pitch: 0.3500
[INFO GRT-0088] Layer met3    Track-Pitch = 0.6800  line-2-Via Pitch: 0.6150
[INFO GRT-0088] Layer met4    Track-Pitch = 0.9200  line-2-Via Pitch: 1.0400
[INFO GRT-0088] Layer met5    Track-Pitch = 3.4000  line-2-Via Pitch: 3.1100
[INFO GRT-0019] Found 33791 clock nets.
[INFO GRT-0001] Minimum degree: 2
[INFO GRT-0002] Maximum degree: 241
[INFO GRT-0003] Macros: 16
[INFO GRT-0043] No OR_DEFAULT vias defined.
[INFO GRT-0004] Blockages: 372462

[INFO GRT-0053] Routing resources analysis:
          Routing      Original      Derated      Resource
Layer     Direction    Resources     Resources    Reduction (%)
---------------------------------------------------------------
li1        Vertical            0             0          0.00%
met1       Horizontal   12970160       6355806          51.00%
met2       Vertical      9726815       6544551          32.72%
met3       Horizontal    6485080       4328358          33.26%
met4       Vertical      3892175       1940756          50.14%
met5       Horizontal    1296855        552162          57.42%
---------------------------------------------------------------

[INFO GRT-0101] Running extra iterations to remove overflow.
[INFO GRT-0197] Via related to pin nodes: 1722957
[INFO GRT-0198] Via related Steiner nodes: 54119
[INFO GRT-0199] Via filling finished.
[INFO GRT-0111] Final number of vias: 2325451
[INFO GRT-0112] Final usage 3D: 10350417

[INFO GRT-0096] Final congestion report:
Layer         Resource        Demand        Usage (%)    Max H / Max V / Total Overflow
---------------------------------------------------------------------------------------
li1                  0             0            0.00%             0 /  0 /  0
met1           6355806       1206855           18.99%             0 /  0 /  0
met2           6544551       1350414           20.63%             0 /  0 /  0
met3           4328358        510560           11.80%             0 /  0 /  0
met4           1940756        267422           13.78%             0 /  0 /  0
met5            552162         38813            7.03%             0 /  0 /  0
---------------------------------------------------------------------------------------
Total         19721633       3374064           17.11%             0 /  0 /  0

[INFO GRT-0018] Total wirelength: 28806216 um
[INFO GRT-0014] Routed nets: 314920
Perform buffer insertion...
[INFO RSZ-0058] Using max wire length 2141um.
[ERROR GRT-0232] Routing congestion too high. Check the congestion heatmap in the GUI.
Error: global_route.tcl, 138 GRT-0232
Command exited with non-zero status 1
Elapsed time: 1:41:32[h:]min:sec. CPU time: user 5784.14 sys 305.97 (99%). Peak memory: 5691696KB.
```

### Additional Context

one thing to mention I have currently 8gb ram with which i  have made 4 gb swap 
I will be ordering more 8gb to upgrade if the issue is due to it.

maliberty, 2024-08-29:
I think the congestion is the symptom.  The problem starts in CTS with the insertion of many large buffers:
[image]

to fairly terrible effect:
[image]

@arthurjolo please investigate with help as needed from @precisionmoon

maliberty, 2024-08-29:
You can see the weird clumps of large buffers:
[image]

arthurjolo, 2024-08-29:
> You can see the weird clumps of large buffers: [...]

These clumps of buffers are caused by the level balancing of CTS. The level balancing is a week part of CTS and needs some inhencement, I am going to look into this.

Wiki1998-dev, 2024-08-29:
@arthurjolo Do you need any more files ?
export DESIGN_NICKNAME = core_v_mini_mcu
export DESIGN_NAME = core_v_mini_mcu
export PLATFORM    = sky130hd

export VERILOG_FILES = ./designs/src/$(DESIGN_NICKNAME)/*.sv

export ADDITIONAL_LIBS = $(wildcard ./lib/*.lib)
export ADDITIONAL_GDSS = $(wildcard ./gds/*.gds)
export ADDITIONAL_LEFS = $(wildcard ./lef/*.lef) 
export SDC_FILE      = ./designs/$(PLATFORM)/$(DESIGN_NICKNAME)/constraint.sdc

export ABC_AREA      = 1

export SYNTH_HIERARCHICAL = 1
export MAX_UNGROUP_SIZE = 100

export CORE_UTILIZATION = 20  # increase to provide more space for buffer distribution.lower the core utilisation to provide more space for cell placement reducing dense clustering
export CORE_ASPECT_RATIO = 1
export CORE_MARGIN = 2

export PLACE_DENSITY_LB_ADDON = 0.16
export PLACE_DENSITY          = 0.16#lower the placement density to encourage a more even distribution of cells.

export RTLMP_FLOW = True

export MACRO_PLACE_HALO = 70 70 # Increase to provide more space around macros for better buffer placement
export MACRO_PLACE_CHANNEL = 140 140  # Increasedto widen channels between macros

export FASTROUTE_TCL = $(PLATFORM_DIR)/fastroute.tcl

do these deduction is correct would these in any way improve it ?

arthurjolo, 2024-08-29:
These changes wouldn't prevent CTS from forming those clumps of buffers, the level balancing step is purposely forming them.

Wiki1998-dev, 2024-08-30:
Hi It is related to it
https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/issues/1286 
maybe you can get more of idea

Wiki1998-dev, 2024-09-02:
I used the older version of ORFS 
and it worked to give the final GDS 
But Can I use Autotuner for this to find correct parameters?

Wiki1998-dev, 2024-09-23:
Hello i wanted to ask kindly  is there any update for modifications  in cts regarding the issue?

maliberty, 2024-09-23:
@arthurjolo please update

Wiki1998-dev, 2024-09-24:
Ok sure Thanks

arthurjolo, 2024-09-27:
@Wiki1998-dev The Pull Request with the CTS update was created. I tested the new version of CTS for your design and GRT is still having overflow, now the overflow is not related to any CTS nets or buffers. It seems that it is related to the high number of connections between the macros:
[image]

Wiki1998-dev, 2024-09-30:
Thanks a lot for your help and effort .
Much appreciated