The-OpenROAD-Project/OpenROAD-flow-scripts #2325: error in installing

ripudamank2, 2024-09-05:
### Subject

[Build] for compilation-related issues.

### Describe the bug

 74%] Building CXX object third-party/abc/CMakeFiles/libabc.dir/src/bdd/llb/llb2Dump.c.o
[ 74%] Building CXX object third-party/abc/CMakeFiles/libabc.dir/src/bdd/llb/llb2Flow.c.o
[ 74%] Building CXX object third-party/abc/CMakeFiles/libabc.dir/src/bdd/llb/llb2Image.c.o
[ 74%] Building CXX object third-party/abc/CMakeFiles/libabc.dir/src/bdd/llb/llb3Image.c.o
[ 74%] Building CXX object third-party/abc/CMakeFiles/libabc.dir/src/bdd/llb/llb3Nonlin.c.o
[ 74%] Building CXX object third-party/abc/CMakeFiles/libabc.dir/src/bdd/llb/llb4Cex.c.o
[ 74%] Building CXX object third-party/abc/CMakeFiles/libabc.dir/src/bdd/llb/llb4Image.c.o
[ 74%] Building CXX object third-party/abc/CMakeFiles/libabc.dir/src/bdd/llb/llb4Nonlin.c.o
[ 74%] Building CXX object third-party/abc/CMakeFiles/libabc.dir/src/bdd/llb/llb4Sweep.c.o
[ 74%] Linking CXX static library libabc.a
[ 74%] Built target libabc
gmake: *** [Makefile:146: all] Error 2

please let me know what should i do ?

### Expected Behavior

it should have installed the tools 

### Environment

```shell
ubunut 22
```

### To Reproduce

git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts
cd OpenROAD-flow-scripts
sudo ./setup.sh
./build_openroad.sh --local   --- in this setup

rovinski, 2024-09-06:
The root cause is not shown here. Please attach the full log (`build_openroad.log`)

ripudamank2, 2024-09-07:
[build_openroad.log](https://github.com/user-attachments/files/16917036/build_openroad.log)

Please have a look at  attached log.

ripudamank2, 2024-09-07:
it solved i found the problem in log file , we can close . Thank you

closed 2024-09-07 by ripudamank2

MeowOhMeow, 2024-09-07:
I encountered the same issue during the build process, and it was caused by libgmock-dev and libgtest-dev not being installed on my system. To resolve this, I had to manually install these dependencies. I believe it would be helpful if these two packages were automatically installed when running ./setup.sh to avoid this issue for future users.

steveniscoming, 2024-09-07:
Can anyone tell me how to resolve this problem