The-OpenROAD-Project/OpenROAD-flow-scripts #2330: failed on "./build_openroad.sh --local"

weiii1121, 2024-09-07:
### Subject

[Build] for compilation-related issues.

### Describe the bug

[ 94%] Building CXX object src/CMakeFiles/openroad.dir/Design.cc.o
[ 94%] Building CXX object src/CMakeFiles/openroad.dir/Timing.cc.o
[ 94%] Building CXX object src/CMakeFiles/openroad.dir/Tech.cc.o
[ 94%] Building CXX object src/CMakeFiles/openroad.dir/OpenRoad.cc.o
[ 94%] Building CXX object src/CMakeFiles/openroad.dir/Main.cc.o
[ 94%] Linking CXX executable openroad
/usr/bin/ld: gpl/gpl.a(mbff.cpp.o): in function `gpl::MBFF::RunILP(std::vector<gpl::Flop, std::allocator<gpl::Flop> > const&, std::vector<gpl::Tray, std::allocator<gpl::Tray> > const&, std::vector<std::pair<int, int>, std::allocator<std::pair<int, int> > >&, float, float, std::vector<int, std::allocator<int> >)':
mbff.cpp:(.text+0x1ac5c): undefined reference to `operations_research::sat::CpModelProto::CpModelProto(google::protobuf::Arena*)'
/usr/bin/ld: mbff.cpp:(.text+0x1ac63): undefined reference to `absl::lts_20240116::container_internal::kEmptyGroup'
/usr/bin/ld: mbff.cpp:(.text+0x1f379): undefined reference to `absl::lts_20240116::container_internal::kEmptyGroup'
/usr/bin/ld: mbff.cpp:(.text+0x1f3c9): undefined reference to `operations_research::sat::SatParameters::SatParameters(google::protobuf::Arena*)'
collect2: error: ld returned 1 exit status
gmake[2]: *** [src/CMakeFiles/openroad.dir/build.make:517: src/openroad] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:2262: src/CMakeFiles/openroad.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2

this error comes out when i running the command "./build_openroad.sh --local --latest --clean --threads 1".
please let me know how to fix it. thanks.

### Expected Behavior

build successfully

### Environment

```shell
Ubuntu 22.04
```

### To Reproduce

git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts
cd OpenROAD-flow-scripts
sudo ./setup.sh
./build_openroad.sh --local --latest --clean --threads 1

### Screenshots

[image]

### Additional Context

[build_openroad.log](https://github.com/user-attachments/files/16918054/build_openroad.log)

kevinnnnn0131, 2024-09-08:
it's your problem

maliberty, 2024-09-08:
@kevinnnnn0131 not a helpful comment.

I would guess you have conflicting versions of or_tools and abseil.  Perhaps you already have abseil installed and cmake is finding the wrong version.  Please check.

oharboe, 2024-09-10:
@maliberty Looks like the problem I ran into.

@weiii1121 Try these two pull requests together:

https://github.com/The-OpenROAD-Project/OpenROAD/pull/5660
https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/2295

vijayank88, 2024-10-17:
I'm also facing issue with local installation, same as above error message.

vvbandeira, 2025-08-22:
Close due to inactivity. We are building on Ubuntu 22.04 on our CI, so it should be fixed. Please reopen if the problem is still happening on the latest version.

closed 2025-08-22 by vvbandeira