The-OpenROAD-Project/OpenROAD-flow-scripts #2354: make gui_grt reads in 5_route.sdc instead of 4_cts.sdc leading to a ton of errors and slow loads

oharboe, 2024-09-18:
### Subject

[Stage]: Other. Please describe below.

### Describe the bug

This will load 5_route.sdc insead of 4_cts.sdc, this leads to problems with slow loads for large designs and lots of error messages because 5_route.sdc refers to registers that don't exist in 4_cts.sdc.

```
make grt
make gui_grt
```

### Expected Behavior

No errors/warnings and load 4_cts.sdc and not 5_route.sdc when running make gui_grt

### Environment

```shell
$ git log -1 origin/master 
commit 74df8de6bf80af9197ed51285489a088b29af841 (HEAD, origin/master, origin/HEAD)
Merge: b4ab3c1a 8142fc4d
Author: Matt Liberty <mliberty@precisioninno.com>
Date:   Mon Sep 16 14:30:51 2024 +0000

    Merge pull request #2295 from antmicro/64503-choose-installed-dependencies
    
    Make sure OpenROAD is built with installed dependencies
```

### To Reproduce

See above

### Relevant log output

```shell
Takes forever to load and lots of warnings.

Fast if I deleted 5_route.sdc in which case 4_cts.sdc is loaded.

$ build/make gui_grt
ODB_FILE=.//results/asap7/BoomTile/base/5_1_grt.odb /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad -gui   /home/oyvind/OpenROAD-flow-scripts/flow/scripts/gui.tcl
QSocketNotifier: Can only be used with threads started with QThread
OpenROAD v2.0-15655-gdc5f404d7 
Features included (+) or not (-): +Charts +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[WARNING STA-0349] instance 'core/FpPipeline/REG[7]$_DFF_P_' not found.
[WARNING STA-0349] instance 'core/FpPipeline/fp_issue_unit/io_dis_uops_0_ready_REG$_DFF_P_' not found.
[WARNING STA-0349] instance 'core/FpPipeline/fp_issue_unit/io_dis_uops_1_ready_REG$_DFF_P_' not found.
[WARNING STA-0349] instance 'core/FpPipeline/fp_issue_unit/io_dis_uops_2_ready_REG$_DFF_P_' not found.
[WARNING STA-0349] instance 'core/FpPipeline/fp_issue_unit/io_dis_uops_3_ready_REG$_DFF_P_' not found.
[WARNING STA-0349] instance 'core/FpPipeline/fp_issue_unit/slots_0/p1$_SDFF_PP0_' not found.
```

maliberty, 2024-09-18:
Registers should never be created at this point in the flow so I don't understand "5_route.sdc refers to registers that don't exist in 4_cts.sdc.".

oharboe, 2024-09-18:
See warnings above

maliberty, 2024-09-18:
I see them but your explanation makes no sense to me and there is no test case to look at.

oharboe, 2024-09-18:
make gui_grt loads 5_route.sdc if present, whereas it should load 4_cts.sdc.

PR #2355 (2024-09-20): Ensure find_sdc_file works for substeps
fixes #2354
commit 0c2294a3 (2024-09-20): Ensure find_sdc_file works for substeps

fixes #2354
closed 2024-09-20 by maliberty