The-OpenROAD-Project/OpenROAD-flow-scripts #2424: Write out a nice $(REPORTS_DIR)/index.html file to view results

oharboe, 2024-10-05:
### Description

It would be nice if `make final` wrote out one or more .html files to view the results.

This would avoid having to open the OpenROAD GUI (which can be slow for large designs). 

My thinking is that over time, this could cover cover 95% of the use-cases for viewing results of a build.

I added this to the end of save_images.tcl and it worked, but immediately it is clear that making this nice and user-friendly is a fair amount of work...

```
proc generate_image_html {html_filename image_path} {
  set shared_path [file dirname [file normalize $html_filename]]
  set normalized_image_path [file normalize $image_path]
  set image [string range $normalized_image_path [expr [string length $shared_path]+1] end]
  return "<img src=\"$image\" alt=\"$image\"><br>\n"
}

set html_filename "$::env(REPORTS_DIR)/index.html"
set html_file [open $html_filename "w"]
puts $html_file "<!DOCTYPE html>\n<html>\n<head>\n<title>Image Report</title>\n</head>\n<body>\n<h1>Image Report</h1>\n"
foreach image [glob -nocomplain -directory $::env(REPORTS_DIR) *.webp] {
    puts $html_file [generate_image_html $html_filename $image]
}
puts $html_file "</body>\n</html>"
close $html_file
```

### Suggested Solution

Collect information on common use-cases and figure out what static .html files to write out to the $(REPORTS_DIR)

maliberty, 2024-10-05:
What are your common use cases?  The images written cover the ones I need 😉

oharboe, 2024-10-05:
> What are your common use cases? The images written cover the ones I need 😉

:-)

Good question.... Ideally, OpenROAD would just be so fast that I wasn't trying to avoid having to open it and drill into whatever I'm interested for a design. For MegaBoom it takes ~10 minutes waiting for paint to dry(STA loading to view timing reports, heatmaps, loading .odb, etc.).

For MegaBoom I created `bazel build sram_report`, which saves me opening OpenROAD 15 times(once per SRAM) and inspecting 5 datapoints and creating a table. So that is a fantastic report from an automation user-time saving perspective and at the same time a good example of a report that is probably highly specific to a design. Ideally there were lots of large designs that needs to mock SRAMs with ORFS :-) so perhaps one day, a general feature.

maliberty, 2024-10-05:
You'll find no tool so fast that large designs open instantly.   I'm sure it could be faster but there will always be a cost.  I haven't heard a specific request here other than what your experiment already shows.

Note https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/flow/util/genReportTable.py is used by the CI to generate html.

oharboe, 2024-10-05:
So you're saying that OpenROAD could be better than everything else? :-)

More serioulsy: making such reports is a way to mitigate the inherent time it takes to gather and analyze data.

oharboe, 2024-10-06:
For my part, https://github.com/The-OpenROAD-Project/OpenROAD/issues/5885 would be far superior...

closed 2024-10-06 by oharboe