The-OpenROAD-Project/OpenROAD-flow-scripts #2425: make synth stuck at synth -top $::env(DESIGN_NAME) -run fine: {*}$synth_args

janboeye, 2024-10-05:
### Subject

[Stage]: Synthesis.

### Describe the bug

when I synthesis Xiangshan which is a riscv cpu, it stuck at the second round yosys synth command
synth -top $::env(DESIGN_NAME) -run fine: {*}$synth_args
at stage
5.3.1. Executing OPT_EXPR pass (perform const folding).

### Expected Behavior

synthesis could be finished.

### Environment

```shell
make synth
```

### To Reproduce

pull the my branch of  https://github.com/janboeye/OpenROAD-flow-scripts.

execute  make synth

maliberty, 2024-10-06:
You forgot to mention the changes are on the `my` branch.

maliberty, 2024-10-06:
Your design looks to be quite large (82Mb input Verilog with 1500 modules).  I expect this is unrealistic to do flat.  In any case it is a yosys issue and should be filed [there](https://github.com/YosysHQ/yosys)

closed 2024-10-06 by maliberty

janboeye, 2024-10-07:
Thanks @maliberty 
I'm curious why needs the second round synth 'synth -top $::env(DESIGN_NAME) -run fine:'?