The-OpenROAD-Project/OpenROAD-flow-scripts #2444: Run `flow/scripts/generate-variables-docs.py` and create doc update commit as part of OpenROAD bot

oharboe, 2024-10-12:
### Description

It is annoying to re-run flow/scripts/generate-variables-docs.py to update docs/user/FlowVariables.md as part of PRs, because it is likely to create conflicts. Especially if the format of docs/user/FlowVariables.md is amended.

### Suggested Solution

Run `flow/scripts/generate-variables-docs.py` and create doc update commit as part of OpenROAD bot

oharboe, 2024-10-12:
Or... alternatively, create manual occasional standalone PRs to update docs. This would avoid merge conflicts and keep the docs up to date in practice.

@maliberty Good 'nuf? Close?

luarss, 2024-10-12:
Agreed, this would be a good addition. It could also be nice to have a yaml linter in the CI to for formatting / checking missing yaml keys