The-OpenROAD-Project/OpenROAD-flow-scripts #2458: Detailed placement check failure from within `global_route.tcl`

povik, 2024-10-13:
### Subject

[Stage]: Global Router.

### Describe the bug

On asap7/mock-cpu with some modification to the synthesis scripts I get the following failure at detail_route time:

```
...
[INFO RSZ-0059] Removed 1 buffers.
[INFO RSZ-0045] Inserted 12 buffers, 1 to split loads.
[INFO RSZ-0041] Resized 13 instances.
[INFO RSZ-0043] Swapped pins on 15 instances.
[INFO RSZ-0049] Cloned 6 instances.
[WARNING RSZ-0062] Unable to repair all setup violations.
[INFO RSZ-0033] No hold violations found.
Placement Analysis
---------------------------------
total displacement         28.7 u
average displacement        0.0 u
max displacement            1.7 u
original HPWL           41727.9 u
legalized HPWL          41750.9 u
delta HPWL                    0 %

Repair antennas...
[WARNING GRT-0246] No diode with LEF class CORE ANTENNACELL found.
[WARNING DPL-0005] Overlap check failed (5).
 _39070_ overlaps _38820_
 fifo_in.fifomem.mem\[1\]\[16\]$_DFFE_PP_ overlaps _19706_
 mult_result\[2\]\[4\]$_DFF_P_ overlaps _24054_
 fifo_out.wptr_full.wbin\[3\]$_DFF_PN0__114 overlaps fifo_out.sync_w2r.rq1_wptr\[3\]$_DFF_PN0_
 rebuffer14 overlaps _19705_
[ERROR DPL-0033] detailed placement checks failed.
Error: global_route.tcl, 114 DPL-0033
       10.01 real         8.33 user         1.50 sys
make[1]: *** [Makefile:786: do-5_1_grt] Error 1
make: *** [Makefile:784: results/asap7/mock-cpu/baseline_gainbuf/5_1_grt.odb] Error 2
```

### Expected Behavior

No failure

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Darwin 23.1.0
os: macOS 14.1.2
cmake version 3.29.6
CMake Warning at src/CMakeLists.txt:267 (message):
  spdlog: SPDLOG_FMT_EXTERNAL=ON

CMake Warning (dev) at src/sta/CMakeLists.txt:32 (option):
  Policy CMP0077 is not set: option() honors normal variables.  Run "cmake
  --help-policy CMP0077" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

  For compatibility with older versions of CMake, option is clearing the
  normal variable 'USE_TCL_READLINE'.
This warning is for project developers.  Use -Wno-dev to suppress it.

Number of processor cores: 10
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
TCL_LIBRARY
[... 166 lines cut ...]
-- Found OpenMP: TRUE (found version "5.0")
-- GUI is enabled
-- Charts widget is enabled
-- Found Boost: /nix/store/pjmwmbng25l1s4z3m4dvkxlp53pj8ps1-boost-1.83.0-dev/lib/cmake/Boost-1.83.0/BoostConfig.cmake (found version "1.83.0") found components: serialization
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS) 
-- Found Boost: /nix/store/pjmwmbng25l1s4z3m4dvkxlp53pj8ps1-boost-1.83.0-dev/lib/cmake/Boost-1.83.0/BoostConfig.cmake (found suitable version "1.83.0", minimum required is "1.78")
-- TCL library: TCL_LIBRARY-NOTFOUND
-- TCL header: TCL_HEADER-NOTFOUND
-- Found Boost: /nix/store/pjmwmbng25l1s4z3m4dvkxlp53pj8ps1-boost-1.83.0-dev/lib/cmake/Boost-1.83.0/BoostConfig.cmake (found version "1.83.0") found components: serialization system thread
-- Found Boost: /nix/store/pjmwmbng25l1s4z3m4dvkxlp53pj8ps1-boost-1.83.0-dev/lib/cmake/Boost-1.83.0/BoostConfig.cmake (found version "1.83.0")
-- Found Eigen3: /nix/store/cjs08d4r8id8h9dg29v6fcha6vq1mdgd-eigen-3.4.0/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.0")
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done (22.9s)
-- Generating done (0.9s)
```

### To Reproduce

Please see this test case extracted with `make global_route_issue`.

https://drive.google.com/file/d/1_HnpYS9mCNMZWu_FLQ5p7_XALSTJ7Ugj/view?usp=sharing

Note this reproduces with both f9cfd938 of openroad (the submodule revision) and 3a75eb5 (the current head)

maliberty, 2024-10-14:
The placement density is very high - do you know why?
[image]
The instance placement also have an odd 'stringy' character
[image]

maliberty, 2024-10-14:
The pins being off towards the side might be related to it.  I also see a fair number of gain* buffers in the area near the pins.  The density is pretty high at 71% so you could try reducing it a bit.  You might also investigate the buffering happening in gpl.

I don't think this is fundamentally a dpl issue, I think that is the symptom.

maliberty, 2024-10-14:
That's a fair point.  I'll take a look.

oharboe, 2024-10-14:
I dont know what that error means.

It should just be like any garden variety logic.

povik, 2024-10-14:
Ah, the design is called mock-cpu but I see its sources are placed at `./designs/src/fifo/*.v` -- stringy features are not unexpected with a FIFO, are they... :)

maliberty, 2024-10-22:
It looks like there is something wrong even in the incoming odb.  How can I run this full flow (or least post-yosys onwards)?

povik, 2024-10-22:
Should I create a post floorplan checkpoint?

maliberty, 2024-10-22:
Yes please

povik, 2024-10-22:
It looks like the original archive by `make global_route_issue` contains odbs for the preceding stages of the flow too. In any case I was able to reproduce again with OR f9cfd938 and below is the export from `make floorplan_issue`. There are no changes to the scripts post floorplan, and the changed `floorplan.tcl` is part of the archive.

https://drive.google.com/file/d/1jD6J6BIq1RH0jw9MZ6GnjS4-LOPYQhe9/view?usp=sharing

@maliberty Let me know if this is usable by you as-is.

closed 2024-10-24 by maliberty