The-OpenROAD-Project/OpenROAD-flow-scripts #2461: invalid command name "design_is_routed"

mi-and-core, 2024-10-14:
### Subject

[Flow] for any util, flow Makefile, or flow script issues.

### Describe the bug

regardless of the design [sky130hd, asap, nangate45, ..] / [gcd, ibex, aes, ...] always stops after successful DRT with:
Error: detail_route.tcl, 75 invalid command name "design_is_routed"

### Expected Behavior

design_is_routed  in detail_route.tcl should be defined

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 6.8.0-45-generic
os: Ubuntu 22.04.5 LTS (Jammy Jellyfish)
cmake version 3.24.2
-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-15833-ge3b4fc0c9
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function - Success
[... 118 lines cut ...]
```

### To Reproduce

cd flow
make DESIGN_CONFIG=./designs/nangate45/gcd/config.mk

### Relevant log output

```shell
...
[INFO DRT-0267] cpu time = 00:00:37, elapsed time = 00:00:30, memory = 648.43 (MB), peak = 729.98 (MB)

[INFO DRT-0180] Post processing.
[WARNING GRT-0246] No diode with LEF class CORE ANTENNACELL found.
[INFO ANT-0002] Found 0 net violations.
[INFO ANT-0001] Found 0 pin violations.
[INFO ANT-0002] Found 0 net violations.
[INFO ANT-0001] Found 0 pin violations.
Error: detail_route.tcl, 75 invalid command name "design_is_routed"
Command exited with non-zero status 1
Elapsed time: 0:43.78[h:]min:sec. CPU time: user 65.32 sys 2.24 (154%). Peak memory: 747504KB.
make[1]: *** [Makefile:785: do-5_2_route] Fehler 1
make: *** [Makefile:783: results/nangate45/gcd/base/5_2_route.odb] Fehler 2
```

maliberty, 2024-10-14:
You must have an older version of OpenROAD as this is a new command.  Try re-building with the latest.

mi-and-core, 2024-10-15:
@maliberty 
Thanks. After re-building it works as expected. ( It looks like OR/ORFS are a "fast moving target" )

maliberty, 2024-10-15:
Definitely, we are hitting 200 PRs/month now.

closed 2024-10-15 by maliberty