The-OpenROAD-Project/OpenROAD-flow-scripts #2473: make command fails inside the docker during detail routing

vijayank88, 2024-10-17:
### Subject

[Stage]: Detail Router.

### Describe the bug

Installed latest ORFS with docker based.
While running with `make` command with default design, it fails with below error
```
[INFO DRT-0267] cpu time = 00:00:31, elapsed time = 00:00:11, memory = 859.24 (MB), peak = 859.24 (MB)

[INFO DRT-0180] Post processing.
[WARNING GRT-0246] No diode with LEF class CORE ANTENNACELL found.
Error: detail_route.tcl, 65 expected boolean value but got ""
Command exited with non-zero status 1
Elapsed time: 0:13.18[h:]min:sec. CPU time: user 48.55 sys 0.94 (375%). Peak memory: 880628KB.
make[1]: *** [Makefile:785: do-5_2_route] Error 1
make: *** [Makefile:783: results/nangate45/gcd/base/5_2_route.odb] Error 2
I have no name!@bf6aca521866:/OpenROAD-flow-scripts/flow$ 
```

### Expected Behavior

successfully complete `make` command inside docker

### Environment

```shell
I have no name!@bf6aca521866:/OpenROAD-flow-scripts/tools/OpenROAD$ ./etc/Env.sh 
tee: build/openroad-env.log: Permission denied
Unknown git commit, this is not a git repository.

Please make sure that you have the latest code changes and add the commit
hash in the description.

kernel: Linux 6.8.0-1015-gcp
os: Ubuntu 22.04.4 LTS (Jammy Jellyfish)
cmake version 3.24.2
CMake Warning at CMakeLists.txt:98 (message):
  OpenROAD git describe failed, using sha1 instead

CMake Error: Could not open file for write in copy operation /OpenROAD-flow-scripts/tools/OpenROAD/include/ord/Version.hh.tmp
CMake Error: : System Error: Inappropriate ioctl for device
CMake Error at CMakeLists.txt:135 (configure_file):
  configure_file Problem configuring file

-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: HEAD-HASH-NOTFOUND
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
[... 112 lines cut ...]
-- Found OpenMP: TRUE (found version "4.5")  
-- GPU is not enabled
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- GUI is enabled
-- Charts widget is enabled
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization 
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS) 
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found suitable version "1.80.0", minimum required is "1.78")  
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization system thread 
-- Found Boost: /usr/localNumber of processor cores: 32
/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0")  
-- Found Eigen3: /usr/local/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.1") 
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring incomplete, errors occurred!
```

### To Reproduce

Install ORFS with docker based installation.
Inside docker run `make`

maliberty, 2024-10-17:
This looks like a recent bug.  Please try with the head.

vijayank88, 2024-10-18:
I build again by pulling latest code, but stops at same error

ChienKaiMa, 2024-10-19:
I have the same issue, I followed the instructions in https://openroad-flow-scripts.readthedocs.io/en/latest/user/BuildWithDocker.html

ChienKaiMa, 2024-10-19:
This design_is_routed only appears once in the .tcl file.

https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/7df3c9c3802c4d98f2fa5aa34664184cfd7050a6/flow/scripts/detail_route.tcl#L71

After commenting this section, I could the GUI part, which also failed in pre-built binaries.

ChienKaiMa, 2024-10-19:
For the envvar issue, add `set ::env(SKIP_ANTENNA_REPAIR_POST_DRT) 1` to the beginning of `detail_route.tcl` (in my case, line 22) should be an easy workaround.

simi1505, 2024-11-29:
I have added `set ::env(SKIP_ANTENNA_REPAIR_POST_DRT) 1` to `detail_route.tcl` and I also commented out `if {![design_is_routed]} { error "Design has unrouted nets." }`.

Now, I have the following error:

```
==========================================================================
finish report_design_area
--------------------------------------------------------------------------
Design area 27644 u^2 98% utilization.
[ERROR GUI-0070] Error: save_images.tcl, 9 Invalid method. Must be one of: configure cget -acquire -disown -delete getName getConstName getBBox getChip getDataBase getTech getParent getParentInst getTopModule getChildren findChild getBTerms findBTerm getITerms getInsts getModules getModInsts getModNets getModBTerms getPowerDomains getLogicPorts getPowerSwitches getIsolations getLevelShifters getGroups getAccessPoints getGlobalConnects globalConnect addGlobalConnect reportGlobalConnect clearGlobalConnect getComponentMaskShift setComponentMaskShift findInst findModule findModInst findPowerDomain findLogicPort findPowerSwitch findIsolation findLevelShifter findGroup findSomeInst findSomeMaster findITerm getObstructions getBlockages getNets getCapNodes getRSegs findNet findSomeNet writeDb writeGuides findVia getVias getGCellGrid getDefUnits setDefUnits getDbUnitsPerMicron dbuToMicrons dbuAreaToMicrons micronsToDbu micronsAreaToDbu getHierarchyDelimeter setBusDelimeters getBusDelimeters getExtCount copyExtDb adjustRC groundCC adjustCC undoAdjustedCC getCornerCount extCornersAreIndependent getCornersPerBlock getExtDbCount getExtCornerName getExtCornerIndex setCornerNameList getCornerNameList setCornerCount setCornersPerBlock initParasiticsValueTables createExtCornerBlock findExtCornerBlock getExtCornerBlock getTrackGrids findTrackGrid getRows getFills getMasters setDieArea getDieArea getCoreArea addBlockedRegionForPins getBlockedRegionsForPins setExtmi getExtmi getExtControl getDft getMinRoutingLayer setMinRoutingLayer getMaxRoutingLayer setMaxRoutingLayer getMinLayerForClock setMinLayerForClock getMaxLayerForClock setMaxLayerForClock getExtCornerNames getCCSegs getSearchDb destroyCCs destroyRSegs destroyCNs destroyParasitics destroyCornerParasitics getCcHaloNets destroyOldParasitics destroyOldCornerParasitics restoreOldCornerParasitics replaceOldParasitics restoreOldParasitics keepOldParasitics keepOldCornerParasitics preExttreeMergeRC clear getWireUpdatedNets getRegions findRegion findNonDefaultRule getNonDefaultRules levelize levelizeFromPrimaryInputs levelizeFromSequential markBackwardsUser2 setDrivingItermsforNets clearUserInstFlags
Error: final_report.tcl, 71 GUI-0070
Command exited with non-zero status 1
Elapsed time: 0:02.93[h:]min:sec. CPU time: user 1.46 sys 0.29 (60%). Peak memory: 178756KB.
make[1]: *** [Makefile:850: do-6_report] Error 1
make: *** [Makefile:848: logs/ihp-sg13g2/gcd/base/6_report.log] Error 2
```

Any suggestions would be much appreciated. Thanks.

simi1505, 2024-11-29:
@vijayank88 I tried to use ORFS with IIC-OSIC-TOOLS since Yosys, Openroad, etc. are already installed in there. I opened an issue in the repo of IIC-OSIC-TOOLS (https://github.com/iic-jku/IIC-OSIC-TOOLS/issues/80), since this might be related to it. If not, I will open a new issue here.
Thanks.

vvbandeira, 2024-12-16:
@vijayank88 @simi1505 does this error still happen with the latest code on the master branch?

simi1505, 2024-12-16:
@vvbandeira The latest update of the IIC-OSIC-Tools fixed it for me.

If you are interested, here is the issue history: https://github.com/iic-jku/IIC-OSIC-TOOLS/issues/80

I also created a GitHub repo template (https://github.com/simi1505/SG13G2_ASIC-Design-Template) for analog-digital mixed-signal simulation for SG13G2 PDK and IIC-OSIC-Tools. This works just fine for me now. Thanks!

vvbandeira, 2025-08-22:
Closing as it looks to be resolved. Please reopen if the problem is still happening on the latest version.

closed 2025-08-22 by vvbandeira