The-OpenROAD-Project/OpenROAD-flow-scripts #2484: Add synth.json, floorplan.json, place.json, grt.json, route.json, final.json

oharboe, 2024-10-20:
### Description

Some design is required to create an initial version of this that is going to be broadly useful for reporting.

The idea is to have machine readable information across stages.

Something like:

- a dictionary that has substeps as keys and a sub-dictionary with information per substep
- running times
- wns
- crpr (a tiny crpr compared to the clock period tells me that two flip flops talking to eachother are separated high up in the tree, making them vulnerable to large )
- there are some existing .json files, these still need to be written out per substep, but could be rolled into the per stage .json file as a convenience: a key for substep and the substep .json file available as a value.

Other?

Also some information that is already known before runtimes that could be useful in a reporting context without further effort than reading the json file:

- a dictionary keyed on clocks and clock period

### Suggested Solution

Some thoughts above on what information is useful for reporting

maliberty, 2024-10-21:
The data is separate by stage already.  Wns is already recorded.  Crpr is path specific and not a general metric so I don't see how to report this.

maliberty, 2024-10-21:
The format is already established by https://vlsicad.ucsd.edu/Publications/Conferences/388/c388.pdf - I don't think we should modify it unless there is a compelling need.

oharboe, 2024-10-21:
Sounds good.

Will look into this as time permits and reopen specific feature requests, if I have any

closed 2024-10-21 by oharboe