The-OpenROAD-Project/OpenROAD-flow-scripts #2485: [WARNING STA-0349] instance 'core/FpPipeline/core/FpPipeline/REG[7]$_DFF_P_' not found.

oharboe, 2024-10-20:
### Subject

[Stage]: Floorplaning.

### Describe the bug

untar https://drive.google.com/file/d/1jE2atKt8WTVgmKjxovWRRICYNemUPKDZ/view?usp=sharing

```
$ ODB_FILE=results/asap7/BoomTile/2/2_2_floorplan_io.odb ./run-me-BoomTile-asap7-2.sh 
OpenROAD v2.0-16664-g9ba039016 
Features included (+) or not (-): +Charts +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
GUI_TIMING=1 reading timing, takes a little while for large designs...
[WARNING STA-0349] instance 'core/FpPipeline/core/FpPipeline/REG[7]$_DFF_P_' not found.
[WARNING STA-0349] instance 'core/FpPipeline/core/FpPipeline/fregfile_io_write_ports_0_REG_bits_addr[0]$_DFF_P_' not found.
[WARNING STA-0349] instance 'core/FpPipeline/core/FpPipeline/fregfile_io_write_ports_0_REG_bits_addr[1]$_DFF_P_' not found.
[WARNING STA-0349] instance 'core/FpPipeline/core/FpPipeline/fregfile_io_write_ports_0_REG_bits_addr[2]$_DFF_P_' not found.
[runs for a very long time... didn't wait]
```

### Expected Behavior

Works with 2_1_floorplan.odb:

```
ODB_FILE=results/asap7/BoomTile/2/2_1_floorplan.odb ./run-me-BoomTile-asap7-2.sh
```

```
OpenROAD v2.0-16664-g9ba039016 
Features included (+) or not (-): +Charts +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
GUI_TIMING=1 reading timing, takes a little while for large designs...
Populating timing paths...
openroad> 
```

### Environment

```shell
OpenROAD v2.0-16664-g9ba039016
```

### To Reproduce

See above

oharboe, 2024-10-21:
Stack trace from `make open_floorplan` (which runs for hours, at least, didn't wait for it to complete)

```
std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char>>::_S_copy(char*, char const*, unsigned long) (/usr/include/c++/13/bits/basic_string.h:427)
std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char>>::_S_copy_chars(char*, char const*, char const*) (/usr/include/c++/13/bits/basic_string.h:484)
void std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char>>::_M_construct<char const*>(char const*, char const*, std::forward_iterator_tag) (/usr/include/c++/13/bits/basic_string.tcc:247)
std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char>>::basic_string<std::allocator<char>>(char const*, std::allocator<char> const&) (/usr/include/c++/13/bits/basic_string.h:649)
odb::dbInst::getName[abi:cxx11]() (/home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/src/odb/src/db/dbInst.cpp:477)
sta::dbNetwork::name(sta::Instance const*) const (/home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/src/dbSta/src/dbNetwork.cc:717)
sta::SdcNetwork::name(sta::Instance const*) const (/home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/src/sta/network/SdcNetwork.cc:727)
sta::dbSdcNetwork::findInstancesMatching1(sta::PatternMatch const*, sta::Vector<sta::Instance const*>&) const (/home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/src/dbSta/src/dbSdcNetwork.cc:97)
sta::dbSdcNetwork::findInstancesMatching(sta::Instance const*, sta::PatternMatch const*) const (/home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/src/dbSta/src/dbSdcNetwork.cc:83)
find_instances_matching(char const*, bool, bool) (/home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/build/src/dbSta/src/CMakeFiles/dbSta.dir/dbStaTCL_wrap.cxx:3709)
::_wrap_find_instances_matching(ClientData, Tcl_Interp *, int, Tcl_Obj *const *) (/home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/build/src/dbSta/src/CMakeFiles/dbSta.dir/dbStaTCL_wrap.cxx:18578)
TclNRRunCallbacks (@TclNRRunCallbacks:38)
___lldb_unnamed_symbol1507 (@___lldb_unnamed_symbol1507:348)
Tcl_EvalEx (@Tcl_EvalEx:11)
Tcl_Eval (@Tcl_Eval:12)
sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) (/home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/src/sta/app/StaMain.cc:97)
::tclAppInit(int &, char **, const char *, Tcl_Interp *) (/home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/src/Main.cc:495)
ord::tclAppInit(Tcl_Interp*) (/home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/src/Main.cc:523)
Tcl_MainEx (@Tcl_MainEx:113)
main (/home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/src/Main.cc:314)
```

oharboe, 2024-10-21:
get_cells takes close to a second to fail, so with the 2_floorplan.sdc not finding any of the signals below, waiting for hours is to be expected.

```
group_path -name reg2reg\
    -from [list [get_cells {core/FpPipeline/core/FpPipeline/REG[7]$_DFF_P_}]\
           [get_cells {core/FpPipeline/core/FpPipeline/fregfile_io_write_ports_0_REG_bits_addr[0]$_DFF_P_}]\
           [get_cells {core/FpPipeline/core/FpPipeline/fregfile_io_write_ports_0_REG_bits_addr[1]$_DFF_P_}]\
           [get_cells {core/FpPipeline/core/FpPipeline/fregfile_io_write_ports_0_REG_bits_addr[2]$_DFF_P_}]\
           [get_cells {core/FpPipeline/core/FpPipeline/fregfile_io_write_ports_0_REG_bits_addr[3]$_DFF_P_}]\
           [get_cells {core/FpPipeline/core/FpPipeline/fregfile_io_write_ports_0_REG_bits_addr[4]$_DFF_P_}]\
           [get_cells {core/FpPipeline/core/FpPipeline/fregfile_io_write_ports_0_REG_bits_addr[5]$_DFF_P_}]\
           [get_cells {core/FpPipeline/core/FpPipeline/fregfile_io_write_ports_0_REG_bits_addr[6]$_DFF_P_}]\
```

oharboe, 2024-10-21:
There is something more going on here that I don't understand. #2486 doesn't fix it.

Also: is it expected that open.tcl takes forever in the case that .sdc is corrupt?

maliberty, 2024-10-21:
@AcKoucher please look at the performance not the actual error source.

maliberty, 2024-10-21:
I got `Error: open.tcl, 20 can't read "::env(ODB_FILE)": no such variable`

maliberty, 2024-10-21:
How was the SDC generated?  The cell names are wrong:
```
get_cells {core/FpPipeline/core/FpPipeline/REG[7]$_DFF_P_}
```
should be
```
get_cells {core/FpPipeline/REG[7]$_DFF_P_}
```
The fully expanded SDC is 145Mb so I guess it will take a while to read regardless of errors.

oharboe, 2024-10-21:
floorplan.tcl wrote it out in megaboom

oharboe, 2024-10-21:
To generate a corrupt 2_floorplan.sdc, I think this should do it:

untar https://drive.google.com/file/d/1ljzEKErlYMBQlE40vHWhjPrOWpVfq8Zk/view?usp=sharing

./run*.sh

wait 4 hours or so...?

oharboe, 2024-10-21:
@maliberty Aha! Modify floorplan.tcl to write out xxx.sdc as soon as the design is loaded, which only takes a few minutes. The xxx.sdc has the corrupt get_cells names:

```
utl::set_metrics_stage "floorplan__{}"
source $::env(SCRIPTS_DIR)/load.tcl
erase_non_stage_variables floorplan
puts loading...
load_design 1_synth.v 1_synth.sdc

# writes out corupt xxx.sdc file
write_sdc -no_timestamp xxx.sdc
puts "Inspect xxx.sdc, it is corrupt"
exit 1
```

It has the corrupt syntax:

```
$ head -n 15 xxx.sdc 
###############################################################################
# Created by write_sdc
###############################################################################
current_design BoomTile
###############################################################################
# Timing Constraints
###############################################################################
create_clock -name clock -period 1200.0000 [get_ports {clock}]
group_path -name reg2reg\
    -from [list [get_cells {core/FpPipeline/core/FpPipeline/REG[7]$_DFF_P_}]\
           [get_cells {core/FpPipeline/core/FpPipeline/fregfile_io_write_ports_0_REG_bits_addr[0]$_DFF_P_}]\
           [get_cells {core/FpPipeline/core/FpPipeline/fregfile_io_write_ports_0_REG_bits_addr[1]$_DFF_P_}]\
           [get_cells {core/FpPipeline/core/FpPipeline/fregfile_io_write_ports_0_REG_bits_addr[2]$_DFF_P_}]\
           [get_cells {core/FpPipeline/core/FpPipeline/fregfile_io_write_ports_0_REG_bits_addr[3]$_DFF_P_}]\
           [get_cells {core/FpPipeline/core/FpPipeline/fregfile_io_write_ports_0_REG_bits_addr[4]$_DFF_P_}]\
```

maliberty, 2024-10-21:
I've got a fix for the bad sdc.  I appears that when a non-matching name is given sta tries hard to find a way to make it match (names are horrible mess with (non-)escaping of hierarchy and bus delimiters).  I think that's where the extra time goes and there isn't too much to about it.

PR #2486 (2024-10-21): floorplan: use 2_floorplan.sdc in floorplan, written by floorplan.tcl
Since floorplan.tcl can do retiming, it is important to use post retiming .sdc file in subsequent steps to avoid incorrect timing information.

Note that loading 1_synth.sdc would appear to work in the case that the SDC_FILE is using patterns, rather than specific names written out longhand in 2_floorplan.sdc.

fixes #2485
commit 3d807918 (2024-10-21): floorplan: use 2_floorplan.sdc in floorplan, written by floorplan.tcl

Since floorplan.tcl can do retiming, it is important to use post
retiming .sdc file in subsequent steps to avoid incorrect
timing information

fixes #2485
closed 2024-10-21 by maliberty