The-OpenROAD-Project/OpenROAD-flow-scripts #2505: setup.sh unable to write to openroad_deps_prefixes

jeffng-or, 2024-10-24:
### Subject

[Flow] for any util, flow Makefile, or flow script issues.

### Describe the bug

When running within a docker container (or presumably a new machine), the logging of the openroad_deps_prefixes.txt file within setup.sh fails with a permission issue.

My best guess that lines 32 & 33 call the ORFS and OR DependencyInstaller.sh scripts as the SUDO_USER (e.g. normal user who called setup.sh with sudo). But, the same DependencyInstaller.sh scripts were called as root on lines 29 & 30 and the file was written by one of those calls.

So, the end result is that the calls as root created the file and then we try to write to the file as the SUDO_USER, who doesn't have permission.

### Expected Behavior

setup.sh doesn't fail on a new machine or new docker container

### Environment

```shell
Running with the current head of ORFS, which would be commit 7e8d10b5a595cd772f01fe0b4b7ebd7669315a22
```

### To Reproduce

Use the following Dockerfile:

```
# syntax=docker/dockerfile:1

FROM ubuntu:22.04
RUN apt update && apt upgrade -y
RUN apt install python3
RUN apt install python3-pip -y
RUN apt install git -y
RUN apt install sudo -y

RUN adduser --disabled-password --gecos '' --home /home/localuser --shell /bin/bash localuser
RUN adduser localuser sudo
RUN echo '%sudo ALL=(ALL) NOPASSWD:ALL' >> /etc/sudoers

USER localuser
WORKDIR /home/localuser

RUN git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts.git
WORKDIR OpenROAD-flow-scripts
```

1. docker build -t Dockerfile -t test:1.0 .
2. docker run -it test:1.0 bash
3. sudo ./setup.sh

### Relevant log output

```shell
./tools/OpenROAD/etc/DependencyInstaller.sh: line 981: /home/jeffng/OpenROAD-flow-scripts/tools/OpenROAD/etc/openroad_deps_prefixes.txt: Permission denied
```

jeffng-or, 2024-11-20:
Working around by:

```
sudo install -m 666 /dev/null OpenROAD-flow-scripts/tools/OpenROAD/etc/openroad_deps_prefixes.txt
```

to create the file owned by root, but open to all for writing before calling setup.sh

mguthaus, 2025-01-31:
I also had this error on an Ubuntu 22.04 system with a local install:

```
git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts
cd OpenROAD-flow-scripts
sudo ./setup.sh
```

> ./tools/OpenROAD/etc/DependencyInstaller.sh: line 922: /home/mrg/OpenROAD-flow-scripts/tools/OpenROAD/etc/openroad_deps_prefixes.txt: Permission denied