The-OpenROAD-Project/OpenROAD-flow-scripts #2506: Add running times and skew to .json files

oharboe, 2024-10-24:
### Description

- Add skew for worst path alongside the worst slack measurements. Example: `cts__timing__setup__skew` alongside `cts__timing__setup__ws`. Of course there is no clock tree before the cts, stage so this would be from the cts stage and on.
- Add running time for .json file that the step corresponds to

I don't think that the worst skew makes sense to add. Minimizing skew is not optimal for CTS as it can overcompensate for something that might not be a problem. A large skew for paths that have a comfortable positive slack to help paths with a negative or little positive slack is an example of useful skew.

Currently SKIP_REPORT_METRICS=1 in megaboom, but if the running time for reporting metrics is not prohibitive, I would like to switch to the .json files to pick up statistics and present in the build report, but I need the two numbers above to get rid of all the cruft in megaboom and use .json files only.

### Suggested Solution

Add two measurements

oharboe, 2024-10-24:
@jeffng-or FYI

maliberty, 2024-10-24:
Runtimes are fine to add (though I wouldn't set any metric limit on them).

Just because you have small skew/crpr on the worst path doesn't really tell you a lot.  You might have terrible values on the second-worst path.  What decision will you make based on these metrics?

oharboe, 2024-10-28:
@maliberty Add running times for now and close this issue?

As you say, skew/CRPR is more of an in depth investigation than a simple metric issue.

maliberty, 2024-10-28:
> Add running times for now and close this issue?

Yes.  Are you planning to make a PR or requesting PII do it?

oharboe, 2024-10-28:
> > Add running times for now and close this issue?
> 
> Yes. Are you planning to make a PR or requesting PII do it?

It is a feature request to the community. It is not urgent. You asked why I'm not using the .json files in megaboom and I filed a few github issues w.r.t. issues with using the .json files.

oharboe, 2024-10-28:
make elapsed could be rolled into .json if generation of .json files works SKIP_REPORT_METRICS=0

closed 2026-03-03 by maliberty