The-OpenROAD-Project/OpenROAD-flow-scripts #2510: Can not generate GDS and save images (save_images.tcl)

b224hisl, 2024-10-26:
### Subject

[Stage]: GDS generation.

### Describe the bug

It seems to have a problem with `save_images.tcl` because it will report
```
[ERROR GUI-0070] Error: save_images.tcl, 9 Invalid method. Must be one of: configure cget -acquire -disown -delete getName getConstName getBBox getChip getDataBase getTech getParent getParentInst getTopModule getChildren findChild getBTerms findBTerm getITerms getInsts getModules getModInsts getModNets getModBTerms getPowerDomains getLogicPorts getPowerSwitches getIsolations getLevelShifters getGroups getAccessPoints getGlobalConnects globalConnect addGlobalConnect reportGlobalConnect clearGlobalConnect getComponentMaskShift setComponentMaskShift findInst findModule findModInst findPowerDomain findLogicPort findPowerSwitch findIsolation findLevelShifter findGroup findSomeInst findSomeMaster findITerm getObstructions getBlockages getNets getCapNodes getRSegs findNet findSomeNet writeDb writeGuides findVia getVias getGCellGrid getDefUnits setDefUnits getDbUnitsPerMicron dbuToMicrons dbuAreaToMicrons micronsToDbu micronsAreaToDbu getHierarchyDelimeter setBusDelimeters getBusDelimeters getExtCount copyExtDb adjustRC groundCC adjustCC undoAdjustedCC getCornerCount extCornersAreIndependent getCornersPerBlock getExtDbCount getExtCornerName getExtCornerIndex setCornerNameList getCornerNameList setCornerCount setCornersPerBlock initParasiticsValueTables createExtCornerBlock findExtCornerBlock getExtCornerBlock getTrackGrids findTrackGrid getRows getFills getMasters setDieArea getDieArea getCoreArea addBlockedRegionForPins getBlockedRegionsForPins setExtmi getExtmi getExtControl getDft getMinRoutingLayer setMinRoutingLayer getMaxRoutingLayer setMaxRoutingLayer getMinLayerForClock setMinLayerForClock getMaxLayerForClock setMaxLayerForClock getExtCornerNames getCCSegs getSearchDb destroyCCs destroyRSegs destroyCNs destroyParasitics destroyCornerParasitics getCcHaloNets destroyOldParasitics destroyOldCornerParasitics restoreOldCornerParasitics replaceOldParasitics restoreOldParasitics keepOldParasitics keepOldCornerParasitics preExttreeMergeRC designIsRouted clear getWireUpdatedNets getRegions findRegion findNonDefaultRule getNonDefaultRules levelize levelizeFromPrimaryInputs levelizeFromSequential markBackwardsUser2 setDrivingItermsforNets clearUserInstFlags
Error: final_report.tcl, 75 GUI-0070
```

### Expected Behavior

Finish the whole process

### Environment

```shell
Unknown git commit, this is not a git repository.

Please make sure that you have the latest code changes and add the commit
hash in the description.

kernel: Linux 3.10.0-1160.76.1.el7.x86_64
os: Ubuntu 22.04 (Jammy Jellyfish)
cmake version 3.24.2
CMake Warning at CMakeLists.txt:106 (message):
  OpenROAD git describe failed, using sha1 instead

-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: 76c4043faaceb03e28fdbaf63fdb29db91d13210
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
[... 145 lines cut ...]

  Some of these libraries may not be found correctly.

BRARIES VTune_INCLUDE_DIRS) 
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found suitable version "1.80.0", minimum required is "1.78")  
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization system thread 
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0")  
-- Found Eigen3: /usr/local/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.1") 
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.SDTW7emqWk
```

### To Reproduce

https://drive.google.com/file/d/1sGo7f41bEXr6v5_fiIbxw0EV-2EJEIZP/view?usp=sharing

### Screenshots

[image]

b224hisl, 2024-10-27:
I have found what's wrong. This error arises from `set markerdb [[ord::get_db_block] findMarkerCategory DRC]`

maliberty, 2024-10-27:
Then you have an out of date OpenRoad sub-module.  Try build.sh --latest