The-OpenROAD-Project/OpenROAD-flow-scripts #2545: homebrew disabled pyyaml, this breaks "make" in OpenROAD-flow-scripts/flow

stefanottili, 2024-11-02:
### Subject

[Flow] for any util, flow Makefile, or flow script issues.

### Describe the bug

MacOS M1. I know: not officially supported, but it worked for the last 4 years.
homebrew "disabled" pyyaml on 2024-10-06.
```
flow % make
Traceback (most recent call last):
  File "OpenROAD-flow-scripts/flow/scripts/defaults.py", line 4, in <module>
    import yaml
ModuleNotFoundError: No module named 'yaml'
...
make: *** [results/nangate45/gcd/base/1_synth.v] Error 2

flow % brew install pyyaml
==> Downloading https://formulae.brew.sh/api/formula.jws.json
########################################################################################################################################################################### 100.0%
==> Downloading https://formulae.brew.sh/api/cask.jws.json
########################################################################################################################################################################### 100.0%
Error: pyyaml has been disabled because it does not meet homebrew/core's requirements for Python library formulae! It was disabled on 2024-10-06.
```

### Expected Behavior

Any hint on how to install yaml in a different way ?

### Environment

```shell
MacOS Sequoia, M1, using homebrew
```

### To Reproduce

cd flow
make

stefanottili, 2024-11-03:
It seems that brew alone is not enough any more, one have to set up using a "virtual env".

https://discuss.python.org/t/on-macos-14-pip-install-throws-error-externally-managed-environment/50352/4

maliberty, 2024-11-03:
No clue but if you get it to work please make a PR.

oharboe, 2024-11-05:
ubuntu 24 removed "sudo pip3 install". Could it be that brew did something similiar for the same reasons? ("security" is the usual suspect....)

stefanottili, 2024-11-05:
Not security, too many users messed up their os with different versions.
https://github.com/orgs/Homebrew/discussions/5707#discussioncomment-11150840

oharboe, 2024-11-05:
Makes sense. Same argument goes for Ubuntu 24, I think.

closed 2025-01-09 by maliberty

stefanottili, 2025-01-09:
Why would you close an issue that still is an issue ?
How was this solved when using Ubuntu 24 ?

oharboe, 2025-01-09:
> Why would you close an issue that still is an issue ?
> How was this solved when using Ubuntu 24 ?

This is the how Python works in 2024... 

Unbuntu, use apt install python-pyyaml or venv.

stefanottili, 2025-01-09:
there doesn't seem to be a homebrew equivalent to pythong-pyyaml, so venv it must be.
Would be nice if this piece of knowledge would be documented somewhere ...

oharboe, 2025-01-09:
This is a generic os/package/python thing in 2024... I dont think OpenROAD should document anything, it will very quickly be moot and it would be highly os/package manager specific.