The-OpenROAD-Project/OpenROAD-flow-scripts #2548: Detail route is exiting with error in default flow

socclosure, 2024-11-05:
### Subject

[Stage]: Detail Router.

### Describe the bug

I am trying to generate GDS view of a design by running all steps of default ORFS (i.e. make commnad) on my Windows machine ,
But the flow is exiting with below error from detail route steps
"
[WARNING GRT-0246] No diode with LEF class CORE ANTENNACELL found.
Error: detail_route.tcl, 63 expected boolean value but got ""
Command exited with non-zero status 1
Elapsed time: 0:05.95[h:]min:sec. CPU time: user 15.09 sys 0.47 (261%). Peak memory: 793444KB.
make[1]: *** [Makefile:794: do-5_2_route] Error 1
make: *** [Makefile:792: results/nangate45/gcd/base/5_2_route.odb] Error 2
"

I have looked in "logs/nangate45/gcd/base/5_2_route.log" file for further debugging and found below error message in it.

"
[WARNING GRT-0246] No diode with LEF class CORE ANTENNACELL found.
Error: detail_route.tcl, 63 expected boolean value but got ""
Command exited with non-zero status 1
Elapsed time: 0:05.95[h:]min:sec. CPU time: user 15.09 sys 0.47 (261%). Peak memory: 793444KB.
"

I donot understand how to fix this error, hence filing this issue.

### Expected Behavior

Suggestion to resolve the error which i am seeing while running default ORFS.

### Environment

```shell
root@docker-desktop:/OpenROAD-flow-scripts/tools/OpenROAD# ./etc/Env.sh
Unknown git commit, this is not a git repository.

Please make sure that you have the latest code changes and add the commit
hash in the description.

kernel: Linux 5.15.153.1-microsoft-standard-WSL2
os: Ubuntu 22.04.4 LTS (Jammy Jellyfish)
cmake version 3.24.2
CMake Warning at CMakeLists.txt:98 (message):
  OpenROAD git describe failed, using sha1 instead

-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: HEAD-HASH-NOTFOUND
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
[... 112 lines cut ...]
-- GUI is enabled
-- Charts widget is enabled
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS)
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found suitable version "1.80.0", minimum required is "1.78")
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization system thread
-- Found Boost: /usr/localNumber of processor cores: 12
/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0")
-- Found Eigen3: /usr/local/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.1")
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.Hx1Jfw1Cb2
root@docker-desktop:/OpenROAD-flow-scripts/tools/OpenROAD#
```

### To Reproduce

[detail_route_gcd_nangate45_base_2024-11-05_06-52.tar.gz](https://github.com/user-attachments/files/17628288/detail_route_gcd_nangate45_base_2024-11-05_06-52.tar.gz)

Command: "make"

maliberty, 2024-11-05:
`Unknown git commit, this is not a git repository.` - what version of OR & ORFS are you using?

socclosure, 2024-11-06:
@maliberty 

I am checking out latest version of ORFS as per the instruction in 
https://openroad-flow-scripts.readthedocs.io/en/latest/contrib/GitGuide.html#../../index.md#build-or-installing-orfs-dependencies

Master branch of my local repository (i.e. fork) of ORFS is at
https://github.com/socclosure/OpenROAD-flow-scripts/tree/master

in that OpenRoad tool is linked to one of the git commit version.

I am running the flow using

"./util/docker_shell  make" command under

flows directory.

closed 2026-04-16 by github-actions[bot]