The-OpenROAD-Project/OpenROAD-flow-scripts #2570: Document mapping from OpenROAD commands and options to variables and vice versa

oharboe, 2024-11-17:
### Description

Curretnly, the only way to figure out what ORFS variables affect which OpenROAD commands and options and vice versa is to carefully read scripts/*.tcl files.

There's generally little help from the naming convention in ORFS variables. Even with something like `PLACE_DENSITY` it is very hard to connect the dots from this variable name to how they are actually used by OpenROAD commands and options.

### Suggested Solution

Unsure what to suggest...

maliberty, 2024-11-17:
I don't know what naming convention would be reasonable as variables may be used in multiple places.  I don't know that the average user needs to know which stages they are used in.  It seems to be more of a flow maintainer question and I expect such people will have a lot more familiarity.

maliberty, 2024-11-17:
Please reopen if there is something actionable.

closed 2024-11-17 by maliberty