The-OpenROAD-Project/OpenROAD-flow-scripts #2599: make command fails to complete finish stage

vijayank88, 2024-11-30:
### Subject

[Flow] for any util, flow Makefile, or flow script issues.

### Describe the bug

Currently using WSL based installation Ubuntu22.04 and installed using prebuilt binareis.
It fails during finish stage.
Error message attached as screenshot
[image]

### Expected Behavior

Complete flow

### Environment

```shell
Latest commit
```

### To Reproduce

WSL installation with prebuilt binaries, Run
make

### Screenshots

[image]

maliberty, 2024-11-30:
Which pre-built binaries did you use? "Latest commit" is unclear.

vijayank88, 2024-12-04:
"Latest commit" means ORFS latest commit cloned and using it.

Pre-built binaries from here: https://github.com/Precision-Innovations/OpenROAD/releases/tag/2.0-17198-g8396d0866

vijayank88, 2024-12-04:
@vvbandeira  @luarss fyi

vvbandeira, 2024-12-16:
This has been fixed on the latest release.

closed 2024-12-16 by vvbandeira