The-OpenROAD-Project/OpenROAD-flow-scripts #2616: [ERROR] LEF Cell 'spi_DEF_FILL' has no matching GDS/OAS cell. Cell will be empty when running ihp-sg13g2 examples

hpretl, 2024-12-08:
### Subject

[Flow] for any util, flow Makefile, or flow script issues.

### Describe the bug

When running the stock `ihp-sg13g2` examples (e.g., the `spi` block), the following error pops up. Likely something is wrong with the fill procedure, as the resulting GDS cell is empty; in other words, there is no fill.

```
finish report_design_area
--------------------------------------------------------------------------
Design area 9076 u^2 98% utilization.
Elapsed time: 0:01.41[h:]min:sec. CPU time: user 1.31 sys 0.09 (99%). Peak memory: 209748KB.
cp ./results/ihp-sg13g2/spi/base/5_route.sdc ./results/ihp-sg13g2/spi/base/6_final.sdc
cp /tmp/orfs/flow/platforms/ihp-sg13g2/lef/sg13g2_tech.lef ./objects/ihp-sg13g2/spi/base/klayout_tech.lef
SC_LEF_RELATIVE_PATH="$\(env('FLOW_HOME')\)/platforms/ihp-sg13g2/lef/sg13g2_stdcell.lef"; \
OTHER_LEFS_RELATIVE_PATHS=$(echo "<lef-files>$(realpath --relative-to=./results/ihp-sg13g2/spi/base ./objects/ihp-sg13g2/spi/base/klayout_tech.lef)</lef-files>"); \
sed 's,<lef-files>.*</lef-files>,<lef-files>'"$SC_LEF_RELATIVE_PATH"'</lef-files>'"$OTHER_LEFS_RELATIVE_PATHS"',g' /tmp/orfs/flow/platforms/ihp-sg13g2/sg13g2.lyt > ./objects/ihp-sg13g2/spi/base/klayout.lyt
sed -i 's,<map-file>.*</map-file>,<map-file>/tmp/orfs/flow/platforms/ihp-sg13g2/sg13g2.map</map-file>,g' ./objects/ihp-sg13g2/spi/base/klayout.lyt
(env time -f 'Elapsed time: %E[h:]min:sec. CPU time: user %U sys %S (%P). Peak memory: %MKB.' stdbuf -o L /foss/tools/klayout/klayout -zz -rd design_name=spi \
        -rd in_def=./results/ihp-sg13g2/spi/base/6_final.def \
        -rd in_files="/tmp/orfs/flow/platforms/ihp-sg13g2/gds/sg13g2_stdcell.gds  " \
        -rd seal_file="" \
        -rd out_file=./results/ihp-sg13g2/spi/base/6_1_merged.gds \
        -rd tech_file=./objects/ihp-sg13g2/spi/base/klayout.lyt \
        -rd layer_map= \
        -r /tmp/orfs/flow/util/def2stream.py) 2>&1 | tee ./logs/ihp-sg13g2/spi/base/6_1_merge.log
[INFO] Reporting cells prior to loading DEF ...
[INFO] Reading DEF ...
[INFO] Clearing cells...
[INFO] Merging GDS/OAS files...
	/tmp/orfs/flow/platforms/ihp-sg13g2/gds/sg13g2_stdcell.gds
[INFO] Copying toplevel cell 'spi'
[INFO] Checking for missing cell from GDS/OAS...
[ERROR] LEF Cell 'spi_DEF_FILL' has no matching GDS/OAS cell. Cell will be empty.
[INFO] Checking for orphan cell in the final layout...
[INFO] No orphan cells
[INFO] Writing out GDS/OAS './results/ihp-sg13g2/spi/base/6_1_merged.gds'
Command exited with non-zero status 1
Elapsed time: 0:00.22[h:]min:sec. CPU time: user 0.19 sys 0.02 (98%). Peak memory: 115600KB.
make[1]: *** [Makefile:892: do-gds-merged] Error 1
make: *** [Makefile:887: results/ihp-sg13g2/spi/base/6_1_merged.gds] Error 2
```

### Expected Behavior

Fill should be in the GDS, and the Makefile runs w/o errors.

### Environment

```shell
Not feasible as using self-compiled OpenROAD, but this likely has nothing to do with the issue above.
```

### To Reproduce

export DESIGN_CONFIG=./designs/ihp-sg13g2/spi/config.mk
make

hpretl, 2024-12-08:
@KrzysztofHerman Maybe this has to do with the `fill.json` for ihp-sg13g2? It looks a bit drafty, when I compare it to the sky130 ones.

maliberty, 2024-12-17:
I don't see that error in the nightly run and I do see fill, eg 30/22
[image]

You'll have to give more info on reproducing it.

hpretl, 2024-12-19:
This is interesting. I was building OR from latest commit, and run on IHP PDK, and I do get the error and I am also not having fill...

How can we find out the difference between the nightly run and my installation?

maliberty, 2024-12-19:
Are you using the IHP PDK from ORFS?  What commit ids of OR & ORFS are you using?  What version of KLayout?  What OS?

hpretl, 2024-12-19:
1) I am using the stock Makefile, I just activate an IHP design. I assume it is using ORFS PDK then, right?
2) OpenROAD from yesterday, that is d629eed8fb387a19c0920f8a5cad4b50b0ba0465
3) ORFS from just now, that is bec5c26bd1bba106a511d265028627a1a890253a
4) KLayout 0.29.10
5) Ubuntu 22.04

I have the feeling, looking at the log, that no fill generation is even run. Stage 6 is finished, and the comes this, but I see no fill action:

```
finish report_design_area
--------------------------------------------------------------------------
Design area 9076 u^2 98% utilization.
Elapsed time: 0:00.77[h:]min:sec. CPU time: user 0.71 sys 0.05 (100%). Peak memory: 217860KB.
cp ./results/ihp-sg13g2/spi/base/5_route.sdc ./results/ihp-sg13g2/spi/base/6_final.sdc
cp /tmp/orfs/flow/platforms/ihp-sg13g2/lef/sg13g2_tech.lef ./objects/ihp-sg13g2/spi/base/klayout_tech.lef
SC_LEF_RELATIVE_PATH="$\(env('FLOW_HOME')\)/platforms/ihp-sg13g2/lef/sg13g2_stdcell.lef"; \
OTHER_LEFS_RELATIVE_PATHS=$(echo "<lef-files>$(realpath --relative-to=./results/ihp-sg13g2/spi/base ./objects/ihp-sg13g2/spi/base/klayout_tech.lef)</lef-files>"); \
sed 's,<lef-files>.*</lef-files>,<lef-files>'"$SC_LEF_RELATIVE_PATH"'</lef-files>'"$OTHER_LEFS_RELATIVE_PATHS"',g' /tmp/orfs/flow/platforms/ihp-sg13g2/sg13g2.lyt > ./objects/ihp-sg13g2/spi/base/klayout.lyt
sed -i 's,<map-file>.*</map-file>,<map-file>/tmp/orfs/flow/platforms/ihp-sg13g2/sg13g2.map</map-file>,g' ./objects/ihp-sg13g2/spi/base/klayout.lyt
(env time -f 'Elapsed time: %E[h:]min:sec. CPU time: user %U sys %S (%P). Peak memory: %MKB.' stdbuf -o L /foss/tools/klayout/klayout -zz -rd design_name=spi \
        -rd in_def=./results/ihp-sg13g2/spi/base/6_final.def \
        -rd in_files="/tmp/orfs/flow/platforms/ihp-sg13g2/gds/sg13g2_stdcell.gds  " \
        -rd seal_file="" \
        -rd out_file=./results/ihp-sg13g2/spi/base/6_1_merged.gds \
        -rd tech_file=./objects/ihp-sg13g2/spi/base/klayout.lyt \
        -rd layer_map= \
        -r /tmp/orfs/flow/util/def2stream.py) 2>&1 | tee /tmp/orfs/flow/logs/ihp-sg13g2/spi/base/6_1_merge.log
[INFO] Reporting cells prior to loading DEF ...
[INFO] Reading DEF ...
[INFO] Clearing cells...
[INFO] Merging GDS/OAS files...
	/tmp/orfs/flow/platforms/ihp-sg13g2/gds/sg13g2_stdcell.gds
[INFO] Copying toplevel cell 'spi'
[INFO] Checking for missing cell from GDS/OAS...
[ERROR] LEF Cell 'spi_DEF_FILL' has no matching GDS/OAS cell. Cell will be empty.
[INFO] Checking for orphan cell in the final layout...
[INFO] No orphan cells
[INFO] Writing out GDS/OAS './results/ihp-sg13g2/spi/base/6_1_merged.gds'
Command exited with non-zero status 1
Elapsed time: 0:00.15[h:]min:sec. CPU time: user 0.14 sys 0.01 (100%). Peak memory: 114516KB.
make[1]: *** [Makefile:892: do-gds-merged] Error 1
make: *** [Makefile:887: results/ihp-sg13g2/spi/base/6_1_merged.gds] Error 2
```

KrzysztofHerman, 2024-12-19:
@hpretl @maliberty I have faced the same issues on my installation. It seems there are two issues here:

1. At some moment referencing  our `sg13g2.map` layer map file in the technology file `sg13g2.lyt` stopped to work. There is a solution for this I mean setting `GDS_LAYER_MAP` in platform's  `config.mk` to `sg13g2.map`  
2. As far as I know USE_FILL drives openroad to generate filler elements on using metal stack. We should not do this using openroad because we have a separate set of scripts for klayout to generate the fillers however it is still WiP. Once we finish we can add this step to the flow. 

I will rework the scripts and the IHP examples to correctly  map the layers and do not use fillers.

hpretl, 2024-12-19:
@KrzysztofHerman Excellent, thank you!

hpretl, 2024-12-19:
@KrzysztofHerman @maliberty A side comment: In our IIC-OSIC-TOOLS I now install two versions of OpenROAD starting with tag `2024.12`: The one version OpenLane2 wants to see, and the latest one at compile time. In this way users can either run OL2, or use ORFS.

vijayank88, 2024-12-19:
ORFS default setup used KLayout 0.28.8 had no issues and complete successfully. 
If we try to use KLayout 0.29.10 flow will fails as per description.

PR #2861 (2025-02-18): def2stream: Don't clear DEF fill created by newer klayout
Fixes #2616

Required for DEF fill to work without error (or being silently dropped if `GDS_ALLOW_EMPTY` is used) on newer klayout versions
closed 2025-02-18 by maliberty